RIOT-OS/RIOT #13015: Quickest start does not build with vagrant

PeterFromSweden, 2019-12-30:
#### Description
Build of [https://doc.riot-os.org/index.html#the-quickest-start](url)
fails with message
```
Building application "default" for "native" with MCU "native".

gcc: error: unrecognized command line option '-Wformat-overflow'
gcc: error: unrecognized command line option '-Wformat-truncation'
```
#### Steps to reproduce the issue
```
vagrant up
vagrant ssh
git clone https://github.com/RIOT-OS/RIOT RIOT.cloned
cd RIOT.cloned
git checkout 2019.10-branch
cd examples/default/
make all
```

MrKevinWeiss, 2020-01-02:
Confirmed.
ping @cgundogan or @kaspar030

cgundogan, 2020-01-02:
related to #12674. IIRC, @aabadie and @fjmolinas have good experience with RIOT and virtual box (for their tutorials). Can you find a simple way to update the vagrant image? Maybe there's a better way than building the image from scratch, which is quite tedious?

aabadie, 2020-01-02:
> Can you find a simple way to update the vagrant image?

Maybe the configuration used to build the vagrant image used by the RIOT course could be useful, see https://github.com/RIOT-OS/riot-course/tree/master/packer for details.

muhammadadeelinfo, 2020-01-03:
I think gcc version is not understanding "-Wformat-overflow" and "-Wformat-truncation". Maybe you can try with an old version i.e gcc-4.2. or maybe in general gcc community need  to exclude "-Wformat-overflow" and "-Wformat-truncation" from gcc next version.

muhammadadeelinfo, 2020-01-07:
I have tried with gcc version 9.2.1 and it builds without any error.
[image]

fjmolinas, 2020-01-10:
So if I understand correctly this is an issue with the gcc version used in the current vagrant image. I have no idea about the status of the current image used for vagrant, what did it have? @cgundogan can you point me to the location?

@aabadie from your knowledge what is the difference between your image and the one used by vagrant in master?

fjmolinas, 2020-01-10:
@aabadie i have the impresion it only has `arm-gcc`, `openocd` and edbg, right?

aabadie, 2020-01-10:
The image used in the course uses Ubuntu18.04, provide arm gcc 7 and many other flashing tools. To summarize it's more up-to-date with recent tools.

aabadie, 2020-01-10:
I can port the packer/vagrant configuration to RIOT.

fjmolinas, 2020-01-10:
> I can port the packer/vagrant configuration to RIOT

That would be awesome, would be much easier to maintain. Maybe you can add all the toolchains that are already in the docker image?

aabadie, 2020-01-10:
> would be much easier to maintain

For easy maintenance, we would need to setup some CI for the vagrant image. I did that for IoT-LAB so it's possible.

muhammadadeelinfo, 2020-01-16:
If I am not mistaken, this issue will be resolved in master by updating the current vagrant image from ubuntu 16.04 version to 18.04 (+toolchains). i.e it gives error https://github.com/RIOT-OS/RIOT/issues/13015#issue-543783812  with gcc 5.4.0 and builds without error with gcc 7.4.0 or higher version together with corresponding gcc-multilib https://github.com/RIOT-OS/RIOT/issues/13015#issuecomment-571749067. (as i tested with gcc-9.2.0, gcc-7.4.0, gcc-8.3.0)

fjmolinas, 2020-01-16:
> If I am not mistaken, this issue will be resolved in master by updating the current vagrant image from ubuntu 16.04 version to 18.04 (+toolchains). i.e it gives error #13015 (comment) with gcc 5.4.0 and builds without error with gcc 7.4.0 or higher version together with corresponding gcc-multilib #13015 (comment). (as i tested with gcc-9.2.0, gcc-7.4.0, gcc-8.3.0)

Yes we will work on generating a new image. @aabadie will address this to have it in before this release.

closed 2020-02-19 by cgundogan