RIOT-OS/RIOT #13016: boards/atmega328p: linker/memory issue?

wosym, 2019-12-30:
#### Description
I noticed that there are (linker?) memory issues with the atmega328p. Several innocuous lines of code cause the same errors when compiling:

```
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x8008b9 of /home/wouter/Programming/cv_controller/bin/atmega328p/cv_controller.elf section `.bss' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x8008bb of /home/wouter/Programming/cv_controller/bin/atmega328p/cv_controller.elf section `.noinit' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x8008b9 of /home/wouter/Programming/cv_controller/bin/atmega328p/cv_controller.elf section `.bss' is not within region `data'
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: address 0x8008bb of /home/wouter/Programming/cv_controller/bin/atmega328p/cv_controller.elf section `.noinit' is not within region `data'
```
I've encountered this behavior several times in the last couple of weeks, always in different scenario's. The last one, only after adding the conditional in the following pseudocode:
```
val = functioncall()
    if(val == -1) {
    puts("Error!");
    return -1;
}
printf("Val = %d", val);
```
When removing the inner puts(), the bug is solved. Strangely enough. When removing the printf outside of the conditional, the bug also disappears! 

#### Steps to reproduce the issue
I was able to create a minimal code sample which triggers the errors shown above:
```
static int test(int argc, char **argv)
{
        (void) argc;
        (void) argv;
//        puts("test aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa"); 
        printf("%d test  %d %lf\n ", 5, 10, 10.1);
        return 0;
}

static const shell_command_t shell_commands[] = {
    { "test", "test", test},
    { NULL, NULL, NULL }
};

int main(void)
{
    puts("Hello World!");

    printf("You are running RIOT on a(n) %s board.\n", RIOT_BOARD);
    printf("This board features a(n) %s MCU.\n", RIOT_MCU);

    char line_buf[SHELL_DEFAULT_BUFSIZE]; 
    shell_run(shell_commands, line_buf, SHELL_DEFAULT_BUFSIZE);

    //(void) shell_commands;
    //test(1, NULL);

    return 0;
}
```
The above code will compile and run just fine. However, when uncommenting the long puts() in test() the errors will show up. 
It can be fixed by commenting out the functions that enable the shell, and then manually call the test() function (uncomment last two lines). This will work fine. However, if you would increase the string length of the puts() in test() even more, it will break again.

### Related PR? 
In #11122 @roberthartung reports a similar issue while testing. However, since it was merged I'm assuming this issue got fixed? I'm also not sure if it's related at all... 

#### Sidenote
I'm not even sure this is an actual bug. Maybe the m328p is just too weak to run RIOT. 
Of course, inserting extremely large strings like this is something you shouldn't do, but: 
1. This was only to trigger the errors. It also occurred several times while doing very normal and basic things (see the pseudocode above). 
2. Imo it should warn you that your variables are too large. (even though I'm convinced this isn't the actual issue)

fjmolinas, 2019-12-31:
This is not a bug in iteself although it can indeed be improved. When you compile `examples/hello-world` you get roughly 1200kb in RAM (.bss + .data), this gives you only 800kb of wigle room. `shell` pulls in a lot a of modules among them `stdin`, `isr_pipe`, etc. as well as many `printf` and `puts`. In harvard architecture all data goes into RAM, so every constant string in a puts go into `RAM` which is why it can very easily overflow. A workaround for this is https://www.nongnu.org/avr-libc/user-manual/group__avr__pgmspace.html. I started exploring this in https://github.com/RIOT-OS/RIOT/pull/12940 but haven't found time to get to this, If you want to explore this or @maribu suggestions in that PR, it would be welcomed.

fjmolinas, 2019-12-31:
> This is not a bug in iteself although it can indeed be improved. 

I would say this isn't a bug, but it is an issue.

wosym, 2019-12-31:
Yes, was already thinking in this direction, and I agree that it's not really a bug. However, I feel like it is presented as something that appears to be a bug, and imo this shouldn't be happening.

I tested the PSTR-fix in my code, and indeed it seems to solve the issue (there are several ways of doing it, I noticed). 

I'm not entirely sure if I'm getting what you're trying to do in that PR.
Why not do something similar as [this](https://github.com/RIOT-OS/RIOT/blob/master/sys/test_utils/interactive_sync/interactive_sync.c)?

fjmolinas, 2019-12-31:
> I'm not entirely sure if I'm getting what you're trying to do in that PR.
> Why not do something similar as this?

I was trying to do exactly as that change but applied generally, so that all puts (and eventually other stdio functions) would be using there `pgmspace` version by default, but that only works if a string literal is an argument.

In that PR, a different approach was suggested writing our own `puts_somename` that basically implements:

```
#if defined(__AVR__)
#include <avr/pgmspace.h>
/* For small AVR targets the extra strings generated by test interactive
   can make the application overflow the .data section (RAM), we use puts_P()
   to write those constant string to .txt section (FLASH)*/
#define PUTS(_s)                       puts_P(PSTR(_s))
#else
#define PUTS(_s)                       puts(_s)
#endif
```

and migrate all possible puts to that.

Axelandre42, 2020-01-27:
Hey, fell into this issue too. Same chip, trying to compile the _example/saul_ application...

wosym, 2020-01-27:
@Axelandre42 : It basicly means that you're out of memory (usually RAM). Try moving some string constants to Flash by prefixing them with PROGMEM and using puts_p() and printf_p() if you want to print them in the shell. This might give you just enough wiggle-room to flash your controller.

maribu, 2020-05-11:
RAM usage on Harvard Architectures can be indeed improved quite significantly. Still, this is not a bug.

(And so could be the error message while linking. Something like "RAM overflown by foo bytes" would be less cryptic. Still, the message is technically correct.)

I'm closing this, as this is not really a bug. I have some improvements on the RAM usage situation on my todo-list - but sadly this work will not be done anytime soon.

closed 2020-05-11 by maribu