RIOT-OS/RIOT #13030: pkg: always rebuild when patches are applied

basilfx, 2020-01-05:
#### Description
Any external package that uses patches is recompiled every time a build is triggered. For some packages, this increases build times unnecessarily.

I believe the problem was introduced by [this change/line](https://github.com/RIOT-OS/RIOT/pull/11129/files#diff-3b02d4d58d2447e87eb9d6c74c03b1ccR26) in `pkg/pkg.mk`. Applying a patch will result in an additional commit, so `$(shell git -C $(PKG_BUILDDIR) rev-parse HEAD) != $(PKG_VERSION)` will always be true and the repository is always fetched.

A quick test of https://github.com/RIOT-OS/RIOT/pull/11553 showed that it fixes the problem, but I have little knowledge of Makefiles.

#### Steps to reproduce the issue
Any package that uses patches is affected. For example, `pkg/u8g2`.

```
QUIET=0 make -C tests/pkg_u8g2 clean
QUIET=0 make -C tests/pkg_u8g2
```

#### Expected results
After a rebuild, the files of `u8g2` are not rebuilt.

#### Actual results
After an initial compilation, files are still rebuild, because the repository is fetched again:

```
gmake[1]: Entering directory '/Users/basilfx/Desktop/RIOT_knx/RIOT/pkg/u8g2'
Removing .git-downloaded
Removing .git-patched
Removing Makefile
Removing csrc/Makefile
Removing csrc/u8g2_riotos.c
Removing sys/sdl/common/Makefile
Removing sys/utf8/common/Makefile
touch /Users/basilfx/Desktop/RIOT_knx/RIOT/tests/pkg_u8g2/bin/basilfx-taster/.test
From https://github.com/olikraus/u8g2
 * branch              f08ff974c03e5c848bc5d2ae3fddb6a97897881a -> FETCH_HEAD
Warning: you are leaving 1 commit behind, not connected to
any of your branches:

  03ef9955 add RIOT-OS interface.

If you want to keep it by creating a new branch, this may be a good time
to do so with:

 git branch <new-branch-name> 03ef9955

HEAD is now at f08ff974 issue #556
git -C /Users/basilfx/Desktop/RIOT_knx/RIOT/tests/pkg_u8g2/bin/pkg/basilfx-taster/u8g2 checkout  -f f08ff974c03e5c848bc5d2ae3fddb6a97897881a
HEAD is now at f08ff974 issue #556
git -c user.email=buildsystem@riot -c user.name="RIOT buildsystem" -C /Users/basilfx/Desktop/RIOT_knx/RIOT/tests/pkg_u8g2/bin/pkg/basilfx-taster/u8g2 am  --no-gpg-sign --ignore-whitespace --whitespace=nowarn "/Users/basilfx/Desktop/RIOT_knx/RIOT/pkg/u8g2"/patches/*.patch
Applying: add RIOT-OS interface.
touch /Users/basilfx/Desktop/RIOT_knx/RIOT/tests/pkg_u8g2/bin/pkg/basilfx-taster/u8g2/.git-patched
gmake[1]: Leaving directory '/Users/basilfx/Desktop/RIOT_knx/RIOT/pkg/u8g2'
```

#### Versions
```
Operating System Environment
-----------------------------
       Operating System: Mac OS X 10.15.2
                 Kernel: Darwin 19.2.0 x86_64 i386

Installed compiler toolchains
-----------------------------
             native gcc: Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
      arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: Apple clang version 11.0.0 (clang-1100.0.33.8)

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.1.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                 ccache: missing
                  cmake: missing
               cppcheck: missing
                doxygen: missing
                    git: git version 2.24.1
                   make: GNU Make 4.2.1
                openocd: missing
                 python: Python 2.7.16
                python2: Python 2.7.16
                python3: Python 3.7.5
                 flake8: error: /usr/local/opt/python/bin/python3.7: No module named flake8
             coccinelle: missing
```

fjmolinas, 2020-01-06:
@basilfx maybe the labels are misplaced?

basilfx, 2020-01-06:
@fjmolinas yes, but I don't know what happened. I corrected it.

fjmolinas, 2020-01-08:
@basilfx this is fixed now right?

basilfx, 2020-01-08:
Yes, this issue is now fixed.

closed 2020-01-08 by basilfx