RIOT-OS/RIOT #13054: NRF52811 support

vvzvlad, 2020-01-08:
#### Description
I have a developed device with NRF52811 that I would like to run under RIOT OS. As far as I can see, there is no support for 52811 in RIOT now. 
What do I have to do to get it to appear? What prevents it from appearing? 
I plan to use only 802.15.4 radio without BLE AoA, AoD and LR.

benpicco, 2020-02-27:
Might be as simple as adding

```make
ifneq (,$(filter nrf52811xxaa,$(CPU_MODEL)))
  ROM_LEN ?= 0x30000
  RAM_LEN ?= 0x6000
endif
```

to `cpu/nrf52/Makefile.include` - did you try that aleady?

Citrullin, 2020-05-03:
@benpicco Do I get this right? The `CPU_IRQ_NUMOF` is the number of NVICs?

benpicco, 2020-05-03:
Yes, you probably also have to adapt `cpu/nrf52/vectors.c`

benpicco, 2020-05-10:
Fixed by #14008

closed 2020-05-10 by benpicco