RIOT-OS/RIOT #13055: PKG_SOURCE_LOCAL override is broken

haukepetersen, 2020-01-09:
#### Description
using a local copy for a package with `PKG_SOURCE_LOCAL` is broken since #13036 was merged. Now the packages source is copied from the local path, but it is not build anymore.

This should be fixed ASAP, as this is stalling my ongoing experiments :-)

#### Steps to reproduce the issue
Take any package (e.g. `nimble`) and try to build RIOT with a local copy of that package
```
PKG_SOURCE_LOCAL_NIMBLE=/local/path/foo make clean all
```
You will see in the build output, that the package is actually copied:
```
Building application "nimble_gatt" for "nrf52dk" with MCU "nrf52".

rm -Rf /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/pkg/nrf52dk/nimble
mkdir -p $(dirname /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/pkg/nrf52dk/nimble)
cp -a /home/hauke/dev/ble/mynewt-nimble /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/pkg/nrf52dk/nimble
touch /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/pkg/nrf52dk/nimble/.prepared
"make" -C /home/hauke/dev/riot/RIOT/pkg/nimble
"make" -C /home/hauke/dev/riot/RIOT/boards/nrf52dk
"make" -C /home/hauke/dev/riot/RIOT/boards/common/nrf52xxxdk
"make" -C /home/hauke/dev/riot/RIOT/core
...
```
but it is not compiled
```
...
"make" -C /home/hauke/dev/riot/RIOT/sys/xtimer
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_controller.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_drivers_nrf5x.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_host.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_host_store_ram.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_host_util.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_npl_riot.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_porting_nimble.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_riot_contrib.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_svc_gap.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_svc_gatt.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_tinycrypt.a: No such file or directory
arm-none-eabi-gcc: error: /home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_transport_ram.a: No such file or directory
/home/hauke/dev/riot/RIOT/examples/nimble_gatt/../../Makefile.include:537: recipe for target '/home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_gatt.elf' failed
make: *** [/home/hauke/dev/riot/RIOT/examples/nimble_gatt/bin/nrf52dk/nimble_gatt.elf] Error 1
```

#### Expected results
Should be build using the local copy

#### Actual results
does not

fjmolinas, 2020-01-09:
Ok, this slipped in, we did not test this. Will work on fix.

fjmolinas, 2020-01-09:
This fixes compile

```
diff --git a/pkg/local.mk b/pkg/local.mk
index 4f5755da2e..b55aba3c8c 100644
--- a/pkg/local.mk
+++ b/pkg/local.mk
@@ -8,6 +8,8 @@
 
 .PHONY: prepare clean
 
+all: $(PKG_BUILDDIR)/.prepared
+
 prepare: $(PKG_BUILDDIR)/.prepared
        @true
 
```

Trying to test how it clean.s

haukepetersen, 2020-01-09:
I can verify, that works. 

Are you going to open the according PR?

PR #13056 (2020-01-09): pkg/local.mk: fix local pkg build

### Contribution description

In #13036 a test case was missed and with the change in wasn't building local pkg. Still not sure what broke it but this at least fixes the issue. See #13055

Re-added phony since the ifdef prevents reaching the declaration in `pkg/pkg.mk`

### Testing procedure

See #13055

### Issues/PRs references

Introduced by #13036
Fixes #13055
closed 2020-01-09 by basilfx