RIOT-OS/RIOT #13069: BrokenPipeError when flashing bluepill with BMP

Citrullin, 2020-01-10:
#### Description
Flashing the bluepill with the BMP results in BrokenPipeError.

#### Steps to reproduce the issue
1. [Flash one bluepill with the BMP firmware.](https://buger.dread.cz/black-magic-probe-bmp-on-bluepill-stm32f103c8-minimum-system-development-board.html)
2. Connect the BMP to your bluepill (SWD <-> SWD, SWCLK <-> SWCLK, 3V3 <-> 3V3, GND <-> GND)
3. Try to flash the hello world example with
```
BOARD=bluepill PROGRAMMER=bmp make flash term -C examples/hello-world/
```

#### Expected results
A flashed bluepill with the hello world example and an open serial terminal connection.

#### Actual results
BrokenPipeError:
```
/home/citrullin/git/RIOT/dist/tools/bmp/bmp.py  flash /home/citrullin/git/RIOT/examples/hello-world/bin/bluepill/hello-world.elf
found following Black Magic GDB servers:
	[/dev/ttyACM0] Serial: E2C88FD1 <- default 
connecting to [/dev/ttyACM0]...
GDB EXECUTABLE NOT FOUND! FALLING BACK TO /opt/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb
Traceback (most recent call last):
  File "/home/citrullin/git/RIOT/dist/tools/bmp/bmp.py", line 265, in <module>
    gdbmi = connect_to_target(port)
  File "/home/citrullin/git/RIOT/dist/tools/bmp/bmp.py", line 224, in connect_to_target
    expected_result='connected')
  File "/home/citrullin/git/RIOT/dist/tools/bmp/bmp.py", line 99, in gdb_write_and_wait_for_result
    res = gdbmi.write(cmd, timeout_sec=TIMEOUT)
  File "/home/citrullin/.local/lib/python3.7/site-packages/pygdbmi/gdbcontroller.py", line 233, in write
    self.gdb_process.stdin.write(mi_cmd_to_write_nl.encode())
BrokenPipeError: [Errno 32] Broken pipe
make: *** [/home/citrullin/git/RIOT/examples/hello-world/../../Makefile.include:669: flash] Error 1
```

#### Versions

```
Operating System Environment
-----------------------------
         Operating System: "Ubuntu" "19.10 (Eoan Ermine)"
                   Kernel: Linux 5.3.0-22-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: missing
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: missing (missing)

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.13.4
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.24.0
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01000-gdb23c13d (2020-01-10-01:06)
                   python: Python 2.7.17rc1
                  python2: Python 2.7.17rc1
                  python3: Python 3.7.5rc1
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

Maybe @maz3max or @maribu are able to help.

maribu, 2020-01-10:
@Citrullin: Thanks for reporting this. The `BrokenPipeError` seems to be the same as `EPIPE`, which indicates an I/O error (e.g. writing to a pipe with no reader left).

Can you try manually to connect via GDB to the BMP?

E.g. start GDB using:

```
$ /opt/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gdb /home/citrullin/git/RIOT/examples/hello-world/bin/bluepill/hello-world.elf
```

Then enter:

```
tar ext /dev/ttyACM0
mon swdp_scan
attach 1
load
run
```

That should connect (`tar ext /dev/ttyACM0`) with the BMP, scan for targets via SWD (`mon swdp_scan`), attach to the first found target (`attach 1`), flash the program (`load`) and start it (`run`).

If this works, we can rule out that there is some issue with the setup and it definitely is related to the BMP integration in RIOT.

Citrullin, 2020-01-10:
Thanks @maribu
The gdb command gave me the actual error. `libncurses5` was missing. The command is helpful for troubleshooting.

fjmolinas, 2020-01-10:
@Citrullin does this mean we can close this one?

closed 2020-01-10 by Citrullin