RIOT-OS/RIOT #13083: cpu/atmegaxxxx: i2c driver not fully implemented?

wosym, 2020-01-10:
#### Description
When using I2C on an atmega1284p ENXIO (-63) is always returned after a write or any other bus-action. According to the docs, this is when no devices answer the message. However when checking with a logic analyzer it becomes clear that it's impossible for a device to answer, because no message is sent at all. The plot shows a single pulse on the SDA-line after the write (or other bus-operation), nothing more. 
This is behavior is shown both in `tests/periph_i2c`, and in basic test-code written based on the docs:
```
    i2c_init(I2C_DEV(0));
    ret = i2c_acquire(I2C_DEV(0));
    if(ret < 0) {
        _print_i2c_error(ret);
        return ret;
    } else {
        puts("I2C acquired");
    }
    puts("sending... ");
    ret = i2c_write_byte(I2C_DEV(0), 0x07, 0x00, (I2C_NOSTOP));
    if(ret < 0) {
        _print_i2c_error(ret);
    } else {
        printf("Wrote %d bytes \n", ret);
    }
    i2c_release(0);
```
I also tested the same code (and test) on an atmega328p, but the behavior was the same.

My assumption was that the I2C driver was not fully implemented. However, this is done in `cpu/atmega_common/periph/i2c.c` , so it should be fine. But ... is this all that needs to be done for it to work? I noticed other boards also needed configuration in `board/xxxx/include/periph_conf.h`. I noticed this is not present in any of the atmega-boards. Is this configuration not needed for these boards? I tried adding a config based on the STM32-ones, but no success so far. 

#### Steps to reproduce the issue
- Run `tests/periph_i2c` on an atmega1284p or atmega328p (even though my guess is that other atmega boards would also suffer from the same issue).
- Write some bytes on the bus.

#### Expected results
- Proper I2C message placed on bus, visible with logic analyzer and receivable on other bus devices
- No errors in the `tests/periph_i2c`

#### Actual results
- Non-functioning test app: errors
- No data sent to bus

MrKevinWeiss, 2020-01-13:
I can see two potential problems, either the configuration is not defined/correct or the bus needs pull-up resistors.

I would assume it is more something to do with the configuration since the atmega2560 is working correctly without additional pullups (though maybe they were included in the board, I haven't verified).
Though looking through it, I can't see anything. The registers for the atmega2560 i2c have the same function as the atmega1284p.

I don't have any boards that can test that (only cortex or atmega2560 boards).  Maybe @fjmolinas  has some laying around?

wosym, 2020-01-13:
Thanks to a colleague, I was able to deduce the issue to a combination of a faulty logic analyzer and a design flaw on the customer's board. 
So I think this issue can be closed. There is nothing wrong with RIOT-i2c. My apologies.

[image]

closed 2020-01-13 by wosym