RIOT-OS/RIOT #13086: Failing tests on FE310 (Hifive1b)

aabadie, 2020-01-10:
#### Description

Opening this issue for tracking the problems with tests found in #12934. Detailed issues can be found [here](https://github.com/fjmolinas/ci-results/tree/master/results/hifive1b).

See https://github.com/RIOT-OS/RIOT/pull/12934#issuecomment-572908529 for some insight.

#### Steps to reproduce the issue

```
./dist/tools/compile_and_test_for_board/compile_and_test_for_board.py . hifive1b --jobs=4
```

#### Expected results

No failure

#### Actual results

```
Failures during test:
- [tests/cpp11_condition_variable](tests/cpp11_condition_variable/test.failed)
- [tests/gnrc_ipv6_fwd_w_sub](tests/gnrc_ipv6_fwd_w_sub/test.failed)
- [tests/pkg_utensor](tests/pkg_utensor/test.failed)
- [tests/ps_schedstatistics](tests/ps_schedstatistics/test.failed)
- [tests/pthread_barrier](tests/pthread_barrier/test.failed)
- [tests/xtimer_periodic_wakeup](tests/xtimer_periodic_wakeup/test.failed)
- [tests/xtimer_usleep](tests/xtimer_usleep/test.failed)
```

#### Versions

current master

fjmolinas, 2020-01-22:
With #13182 I got these results:

```
ERROR:hifive1b:Tests failed: 6
Failures during test:
- [tests/cpp11_mutex](tests/cpp11_mutex/test.failed)
- [tests/gnrc_ipv6_fwd_w_sub](tests/gnrc_ipv6_fwd_w_sub/test.failed)
- [tests/pkg_utensor](tests/pkg_utensor/test.failed)
- [tests/ps_schedstatistics](tests/ps_schedstatistics/test.failed)
- [tests/xtimer_periodic_wakeup](tests/xtimer_periodic_wakeup/test.failed)
- [tests/xtimer_usleep](tests/xtimer_usleep/test.failed)
```

aabadie, 2020-07-02:
I have new results:

```
ERROR:hifive1b:Tests failed: 13
Failures during test:
- [examples/lorawan](examples/lorawan/test.failed)
- [tests/gnrc_ipv6_ext_opt](tests/gnrc_ipv6_ext_opt/test.failed)
- [tests/gnrc_ipv6_fwd_w_sub](tests/gnrc_ipv6_fwd_w_sub/test.failed)
- [tests/gnrc_ipv6_nib_dns](tests/gnrc_ipv6_nib_dns/test.failed)
- [tests/gnrc_sixlowpan_frag](tests/gnrc_sixlowpan_frag/test.failed)
- [tests/periph_wdt](tests/periph_wdt/test.failed)
- [tests/pkg_heatshrink](tests/pkg_heatshrink/test.failed)
- [tests/pkg_semtech-loramac](tests/pkg_semtech-loramac/test.failed)
- [tests/pkg_utensor](tests/pkg_utensor/test.failed)
- [tests/ps_schedstatistics](tests/ps_schedstatistics/test.failed)
- [tests/xtimer_now_irq](tests/xtimer_now_irq/test.failed)
- [tests/xtimer_periodic_wakeup](tests/xtimer_periodic_wakeup/test.failed)
- [tests/xtimer_usleep](tests/xtimer_usleep/test.failed)
```

The cpp tests are working now.

fjmolinas, 2020-07-21:
During release testing I found that `tests/periph_wft` is failing as well:

```
PASSED target time: 128[ms], actual_time: 125.153[ms]
Timeout in expect script at "child.expect(r"start time: (\d+) us", timeout=1)" (tests/periph_wdt/tests/01-run.py:55)
```

aabadie, 2021-02-08:
New tests are failing in master: tests/event_wait_timeout, tests/thread_flags and tests/thread_flags_xtimer. As reported in https://github.com/RIOT-OS/RIOT/pull/15718#issuecomment-774248134.

The current situation is:

```
ERROR:hifive1b:Tests failed: 10
Failures during test:
- [tests/event_wait_timeout](tests/event_wait_timeout/test.failed)
- [tests/gnrc_ipv6_fwd_w_sub](tests/gnrc_ipv6_fwd_w_sub/test.failed)
- [tests/periph_wdt](tests/periph_wdt/test.failed)
- [tests/pkg_utensor](tests/pkg_utensor/test.failed)
- [tests/ps_schedstatistics](tests/ps_schedstatistics/test.failed)
- [tests/sys_architecture](tests/sys_architecture/test.failed)
- [tests/thread_flags](tests/thread_flags/test.failed)
- [tests/thread_flags_xtimer](tests/thread_flags_xtimer/test.failed)
- [tests/xtimer_periodic_wakeup](tests/xtimer_periodic_wakeup/test.failed)
- [tests/xtimer_usleep](tests/xtimer_usleep/test.failed)
```

`tests/sys_architecture` is also a new one.

aabadie, 2021-02-25:
After #15958 got merged, I retested the failing tests mentioned above and here is the new situation:

```
ERROR:hifive1b:Tests failed: 5
Failures during test:
- [tests/gnrc_ipv6_fwd_w_sub](tests/gnrc_ipv6_fwd_w_sub/test.failed)
- [tests/periph_wdt](tests/periph_wdt/test.failed)
- [tests/ps_schedstatistics](tests/ps_schedstatistics/test.failed)
- [tests/xtimer_periodic_wakeup](tests/xtimer_periodic_wakeup/test.failed)
- [tests/xtimer_usleep](tests/xtimer_usleep/test.failed)
```

aabadie, 2023-01-09:
With this change:

```diff
git diff
diff --git a/cpu/fe310/include/periph_cpu.h b/cpu/fe310/include/periph_cpu.h
index ee9c02d892..e022319032 100644
--- a/cpu/fe310/include/periph_cpu.h
+++ b/cpu/fe310/include/periph_cpu.h
@@ -167,7 +167,7 @@ typedef struct {
 #define RTT_MIN_FREQUENCY   (1U)                    /* in Hz */
 
 #ifndef RTT_FREQUENCY
-#define RTT_FREQUENCY       (RTT_MAX_FREQUENCY)     /* in Hz */
+#define RTT_FREQUENCY       (RTT_MIN_FREQUENCY)     /* in Hz */
 #endif
 
 /**
```

The failing tests falls down to:

```
Failures during test:
- [tests/gnrc_ipv6_fwd_w_sub](tests/gnrc_ipv6_fwd_w_sub/test.failed)
- [tests/gnrc_netif_ipv6_wait_for_global_address](tests/gnrc_netif_ipv6_wait_for_global_address/test.failed)
- [tests/malloc_thread_safety](tests/malloc_thread_safety/test.failed)
- [tests/periph_wdt](tests/periph_wdt/test.failed)
- [tests/pkg_qdsa](tests/pkg_qdsa/test.failed)
- [tests/ps_schedstatistics](tests/ps_schedstatistics/test.failed)
- [tests/sys_sema_inv](tests/sys_sema_inv/test.failed)
- [tests/xtimer_periodic_wakeup](tests/xtimer_periodic_wakeup/test.failed)
- [tests/xtimer_usleep](tests/xtimer_usleep/test.failed)
```

Note that the diff above corresponds to the situation before #14741 was merged and according to https://github.com/RIOT-OS/RIOT/pull/14878#pullrequestreview-496729649, `tests/periph_rtc` is broken for hifive1b in master since #14741.

maribu, 2023-05-19:
Would you mind given this another spin?