RIOT-OS/RIOT #13088: Riot-os freezes with lwip + enc28j60 + stm32L4

Ciusss89, 2020-01-10:
#### Description

I'm developing a project that uses an STM32L4 on Nucleo-64 (nucleo-l476rg) board + ENC28J60 transceiver. The system freezes completely in a few seconds when I use networking.

#### Steps to reproduce the issue

###### System setup
1. ENC28J60 transceiver is connected to Nucleo-64 as reported [here](https://github.com/Ciusss89/gtip-riotos/tree/master/test_01)
2. I add the bootstrap support for the transceiver to lwip pkg
```
--- a/pkg/lwip/contrib/lwip.c
+++ b/pkg/lwip/contrib/lwip.c
@@ -28,6 +28,11 @@
 #include "at86rf2xx_params.h"
 #endif
 
+#ifdef MODULE_ENC28J60
+#include "enc28j60.h"
+#include "enc28j60_params.h"
+#endif
+
 #ifdef MODULE_MRF24J40
 #include "mrf24j40.h"
 #include "mrf24j40_params.h"
@@ -52,7 +57,7 @@
 
 #include "lwip.h"
 
-#define ENABLE_DEBUG    (0)
+#define ENABLE_DEBUG    1
 #include "debug.h"
 
 #ifdef MODULE_NETDEV_TAP
@@ -63,6 +68,10 @@
 #define LWIP_NETIF_NUMOF        ARRAY_SIZE(at86rf2xx_params)
 #endif
 
+#ifdef MODULE_ENC28J60    /* is mutual exclusive with above ifdef */
+#define LWIP_NETIF_NUMOF        ARRAY_SIZE(enc28j60_params)
+#endif
+
 #ifdef MODULE_MRF24J40     /* is mutual exclusive with above ifdef */
 #define LWIP_NETIF_NUMOF        ARRAY_SIZE(mrf24j40_params)
 #endif
@@ -95,6 +104,10 @@ static netdev_tap_t netdev_taps[LWIP_NETIF_NUMOF];
 static at86rf2xx_t at86rf2xx_devs[LWIP_NETIF_NUMOF];
 #endif
 
+#ifdef MODULE_ENC28J60
+static enc28j60_t enc28j60_devs[LWIP_NETIF_NUMOF];
+#endif
+
 #ifdef MODULE_MRF24J40
 static mrf24j40_t mrf24j40_devs[LWIP_NETIF_NUMOF];
 #endif
@@ -149,6 +162,27 @@ void lwip_bootstrap(void)
             return;
         }
     }
+#elif defined(MODULE_ENC28J60)
+    DEBUG("[###] lwip->bootstrap[enc28j60]\n");
+    for (unsigned i = 0; i < LWIP_NETIF_NUMOF; i++) {
+        enc28j60_setup(&enc28j60_devs[i], &enc28j60_params[i]);
+       DEBUG("[###] lwip->bootstrap[enc28j60_setup]\n");
+#ifdef MODULE_LWIP_IPV4
+       if (netif_add(&netif[0], IP4_ADDR_ANY, IP4_ADDR_ANY, IP4_ADDR_ANY,
+           &enc28j60_devs[i], lwip_netdev_init, tcpip_input) == NULL) {
+               DEBUG("Could not add enc28j60 device\n");
+               return;
+       }
+       DEBUG("[###] lwip->bootstrap[MODULE_LWIP_IPV4]\n");
+#else /* MODULE_LWIP_IPV4 */
+       if (netif_add(&netif[0], &enc28j60_devs[i], lwip_netdev_init,
+           tcpip_input) == NULL) {
+               DEBUG("Could not add enc28j60 device\n");
+               return;
+       }
+       DEBUG("[###] lwip->bootstrap[ not MODULE_LWIP_IPV4]\n");
+#endif /* MODULE_LWIP_IPV4 */
+    }
 #elif defined(MODULE_SOCKET_ZEP)
     for (unsigned i = 0; i < LWIP_NETIF_NUMOF; i++) {
         socket_zep_setup(&socket_zep_devs[i], &socket_zep_params[i]);
```
3. I enbled debug macro for `enc28j60`  `lwip`
4. Makefile:
```
## App name
#
APPLICATION = gtip-test-01

## Target
#
BOARD ?= nucleo-l476rg

## This has to be the absolute path to the RIOT base directory:
#
RIOTBASE ?= $(CURDIR)/../../RIOT/

## Generic (GNRC) network stack:
#
#USEMODULE += gnrc_netdev_default
#USEMODULE += auto_init_gnrc_netif

## Choose the debug level: LOG_NONE, LOG_WARNING, LOG_ERROR, LOG_DEBUG, LOG_ALL
CFLAGS += -DLOG_LEVEL=LOG_ALL

# Comment this out to disable code in RIOT that does safety checking
# which is not needed in a production environment but helps in the
# development process:
CFLAGS += -DDEVELHELP

LWIP_IPV4 ?= 1
LWIP_IPV6 ?= 0
USEMODULE += ipv4_addr
USEMODULE += lwip_arp
USEMODULE += lwip_ipv4
USEMODULE += lwip_dhcp_auto
CFLAGS += -DETHARP_SUPPORT_STATIC_ENTRIES=1

USEMODULE += lwip lwip_sock_ip lwip_netdev
USEMODULE += lwip_udp
USEMODULE += lwip_tcp
USEMODULE += od
USEMODULE += netdev_default

USEMODULE += lwip_ethernet

# Change this to 0 show compiler invocation lines by default:
QUIET ?= 1

# Modules to include:
USEMODULE += shell
USEMODULE += shell_commands
USEMODULE += ps

# make sure we read the local enc28j60 params file
CFLAGS += -I$(CURDIR)

## Hardware setup
#
include $(CURDIR)/Makefile.hw.setup

include $(RIOTBASE)/Makefile.include

```
5. Main:
```
#include <unistd.h>
#include <stdio.h>

#include "shell.h"

int main(void)
{
▸       printf("RIOT-OS, MCU=%s Board=%s\n\r", RIOT_MCU, RIOT_BOARD);

▸       /* start shell */
▸       char line_buf[SHELL_DEFAULT_BUFSIZE];

▸       shell_run(NULL, line_buf, SHELL_DEFAULT_BUFSIZE);
                                                                                                                                                                                                                                              
▸       /* should be never reached */
▸       return 0;
}
```

#### Expected results

#### Actual results

Output:

```
> [###] lwip->bootstrap[enc28j60]
[###] lwip->bootstrap[enc28j60_setup]
[###] lwip->bootstrap[MODULE_LWIP_IPV4]
main(): This is RIOT! (Version: 2020.01-devel-1649-g69f6c)
RIOT-OS, MCU=stm32l4 Board=nucleo-l476rg
> [enc28j60] isr: link up!
[enc28j60] isr: packet transmitted
[enc28j60] isr: packet received
[enc28j60] recv: size=108 next=118 buf=0x20002df8 len=1518
[enc28j60] isr: packet received
[enc28j60] recv: size=60 next=188 buf=0x20002df8 len=1518
[enc28j60] isr: packet transmitted
[enc28j60] isr: packet received
[enc28j60] recv: size=342 next=540 buf=0x20002df8 len=1518
[enc28j60] isr: packet transmitted
[enc28j60] isr: packet received
[enc28j60] recv: size=342 next=892 buf=0x20002df8 len=1518
[enc28j60] isr: packet transmitted
```
system freezes on the last line 
#### Versions
I've the same issue with
1. current master.

miri64, 2020-01-10:
Looks like a duplicate of https://github.com/RIOT-OS/RIOT/issues/11860.

miri64, 2020-01-10:
(not sure though, maybe @Clusss89 can confirm)

Ciusss89, 2020-01-10:
> (not sure though, maybe @Clusss89 can confirm)

it looks like a duplicate but I'm sure though. 
#11860 reports a problem when He uses the transceiver with GNRC stack, in this case I'm using the lwip whiteout GNRC stack.

Btw I compiled this [demo](https://github.com/Ciusss89/gtip-riotos/tree/master/test_01) plus ipv6 support It seems to work fine
```
~/mcu/gtip-riotos/test_01 (master) $ git diff
diff --git a/test_01/Makefile b/test_01/Makefile
index 683fc92090a4..04674ab9d65b 100644
--- a/test_01/Makefile
+++ b/test_01/Makefile
@@ -15,6 +15,10 @@ RIOTBASE ?= $(CURDIR)/../../RIOT/
 USEMODULE += gnrc_netdev_default
 USEMODULE += auto_init_gnrc_netif
 
+USEMODULE += gnrc_ipv6_default
+USEMODULE += gnrc_ipv6
+
+
 # Change this to 0 show compiler invocation lines by default:
 QUIET ?= 1
```

Take into account that my main file runs only the shell, while #11860 runs a custom task

I can do other tests/experiments if you need it

miri64, 2020-01-10:
Well, if you face a similar / the same problem, and we could at least beyond reasonable doubt show, that it *is* the same / a similar problem, it might be shown that this is neither a problem in GNRC or lwIP, but with the device driver or the MCU.

Ciusss89, 2020-01-11:
@miri64 

I found the cause of the problem, system freezes when `CFLAGS += -DDEVELHELP` is enabled.

miri64, 2020-01-13:
`DEVELHELP` enables a number of features that could cause it. Let's try to isolate:
- Try compiling with `FORCE_ASSERT=0` (this will cause `-DNDEBUG` to be set, deactivating the `asserts()`. https://github.com/RIOT-OS/RIOT/blob/4524fb1f1ed08455bc922ecd77bc0c981c2b16a8/makefiles/cflags.inc.mk#L56-L60
- If it isn't an assert the node might crash for other reasons, but trying to print out the thread table causes another crash. Try to compile without the module `ps`. https://github.com/RIOT-OS/RIOT/blob/4524fb1f1ed08455bc922ecd77bc0c981c2b16a8/core/panic.c#L63-L66
- Maybe it crashes on `LOG_ERROR`: Try setting `-DLOG_LEVEL=LOG_NONE`.

These are at least some ideas, I had to isolate the problem a bit.

Ciusss89, 2020-01-13:
1.  System freezes in few seconds. (uptime < 10s)
```
CFLAGS += -DLOG_LEVEL=LOG_NONE                                                                                                                                                                                                                

CFLAGS += -DDEVELHELP

#FORCE_ASSERT=0
```

2. System works
```
## Chose the debug level: LOG_NONE, LOG_WARNING, LOG_ERROR, LOG_DEBUG, LOG_ALL
CFLAGS += -DLOG_LEVEL=LOG_ALL                                                                                                                                                                                                                 

#CFLAGS += -DDEVELHELP

#FORCE_ASSERT=0
```

3. System freezes in few seconds.  (uptime < 10s)
```
## Chose the debug level: LOG_NONE, LOG_WARNING, LOG_ERROR, LOG_DEBUG, LOG_ALL
CFLAGS += -DLOG_LEVEL=LOG_NONE

CFLAGS += -DDEVELHELP                                                                                                                                                                                                                         

#FORCE_ASSERT=0

# Change this to 0 show compiler invocation lines by default:
QUIET ?= 1

# Modules to include:
USEMODULE += shell
USEMODULE += shell_commands
#USEMODULE += ps
```

4. System freezes in few seconds. (uptime < 10s)
```
## Chose the debug level: LOG_NONE, LOG_WARNING, LOG_ERROR, LOG_DEBUG, LOG_ALL
CFLAGS += -DLOG_LEVEL=LOG_NONE

CFLAGS += -DDEVELHELP                                                                                                                                                                                                                         

#FORCE_ASSERT=0

# Change this to 0 show compiler invocation lines by default:
QUIET ?= 1

# Modules to include:
USEMODULE += shell
USEMODULE += shell_commands
#USEMODULE += ps
```

5. System works
```
## Chose the debug level: LOG_NONE, LOG_WARNING, LOG_ERROR, LOG_DEBUG, LOG_ALL
CFLAGS += -DLOG_LEVEL=LOG_NONE

#CFLAGS += -DDEVELHELP

FORCE_ASSERT=0

# Change this to 0 show compiler invocation lines by default:
QUIET ?= 1

# Modules to include:
USEMODULE += shell
USEMODULE += shell_commands
USEMODULE += ps                                                                                                                                                                                                                               
USEMODULE += xtimer
```

looks like it matches the first case? 

PS:
I added an uptime func at main.c report above, to monitor if it was still alive. Has riot-os an uptime applet?

miri64, 2020-01-14:
Sorry I should have been clearer... In all cases, DEVELHELP should be kept activated of course.

miri64, 2020-01-14:
(from what I can tell the system always only works if `DEVELHELP` is deactivated, which we already knew before https://github.com/RIOT-OS/RIOT/issues/13088#issuecomment-573320334. The point of this exercise was to find out which of the features activated by `DEVELHELP` might be causing the problem)

Ciusss89, 2020-01-17:
> (from what I can tell the system always only works if `DEVELHELP` is deactivated, which we already knew before [#13088 (comment)](https://github.com/RIOT-OS/RIOT/issues/13088#issuecomment-573320334). The point of this exercise was to find out which of the features activated by `DEVELHELP` might be causing the problem)

you're right, sorry me. 

I'm compiling it with `DEVELHELP` on,  `FORCE_ASSERT=0 `, without `ps` and with `LOG_NONE` but it freezes again.
btw I noted that the system works fine if the link is down (case of unplugged ethernet cable) while it freezes in a few seconds when the link goes up.  

How we continue now?

Ciusss89, 2022-06-01:
@miri64 @MrKevinWeiss is not reproducible anymore on the current master. I just compile the same project and it works fine.

maribu, 2023-05-18:
> @miri64 @MrKevinWeiss is not reproducible anymore on the current master. I just compile the same project and it works fine.

Perfect. There have been a couple of bugfixes merged a few months ago. Looks like the did the trick :tada:

closed 2023-05-18 by maribu