RIOT-OS/RIOT #13104: boards/hifive1: flashing issue

kaspar030, 2020-01-13:
#### Description

@fjmolinas realizied that the hifive1 shows a flasing issue.
Somehow, the boards gets into a state where openocd cannot flash it anymore.

There's a "safe bootloader mode" that can be entered by flashing the reset button twice (second time when the led blinks green, after which the led should pulse in red, indicating safe mode). In this the board can be flashed again. Unfortunately that's a manual procedure and thus makes the board unusable in CI or automated testing...

I've bisected the issue to this commit: `547ebd1b2719df88984b325d7e16f640ecd26be9`

#### Steps to reproduce the issue

1. `BOARD=hifive1 make -Cexamples/hello-world make flash`
2. if it succeeds (depends on previous state), try a second time, should fail

#### Expected results

Flashing succeeds.

#### Actual results

Flashing fails.

#### Versions

Current master, self compiled openocd, on arch linux.

kaspar030, 2020-01-13:
@fjmolinas this seems to be the issue for our hifive1 problems...

kaspar030, 2020-01-13:
@kenrabold any clue what could cause this? The change seems OK.

aabadie, 2020-01-13:
is it possible to add something like:
```
reset_config srst_only connect_assert_srst
```
?

kaspar030, 2020-01-13:
> reset_config srst_only connect_assert_srst

Doesn't help, but shows `Warn : 'srst_nogate' reset_config option is required`.

maribu, 2023-05-19:
Ping?