RIOT-OS/RIOT #13120: tests: broken with stdio_rtt if auto_init is disabled

kaspar030, 2020-01-14:
#### Description

Some tests disable auto_init (and thus xtimer's initialization).
xtimer is a hard dependency when using stdio_rtt, thus all tests disabling auto_init fail, unless they manually initialize xtimer.

Probably affected:

```
$ git grep DIS.*auto
bloom_bytes/Makefile:DISABLE_MODULE += auto_init
cpp_ctors/Makefile:DISABLE_MODULE += auto_init
driver_at86rf2xx/Makefile:DISABLE_MODULE += auto_init
float/Makefile:#DISABLE_MODULE += auto_init
gnrc_sixlowpan_frag/Makefile:DISABLE_MODULE += auto_init
lwip_sock_ip/Makefile:DISABLE_MODULE += auto_init
lwip_sock_tcp/Makefile:DISABLE_MODULE += auto_init
lwip_sock_udp/Makefile:DISABLE_MODULE += auto_init
minimal/Makefile:DISABLE_MODULE += auto_init
msg_avail/Makefile:DISABLE_MODULE += auto_init
mutex_unlock_and_sleep/Makefile:DISABLE_MODULE += auto_init
netdev_test/Makefile:DISABLE_MODULE = auto_init
periph_rtt/Makefile:DISABLE_MODULE += auto_init
posix_semaphore/Makefile:DISABLE_MODULE += auto_init
shell/Makefile:DISABLE_MODULE += auto_init
socket_zep/Makefile:DISABLE_MODULE += auto_init
struct_tm_utility/Makefile:DISABLE_MODULE += auto_init
thread_basic/Makefile:DISABLE_MODULE += auto_init
thread_cooperation/Makefile:DISABLE_MODULE += auto_init
thread_exit/Makefile:DISABLE_MODULE += auto_init
thread_flood/Makefile:DISABLE_MODULE += auto_init
thread_msg/Makefile:DISABLE_MODULE += auto_init
thread_msg_block_race/Makefile:DISABLE_MODULE += auto_init
thread_msg_block_w_queue/Makefile:DISABLE_MODULE += auto_init
thread_msg_block_wo_queue/Makefile:DISABLE_MODULE += auto_init
thread_msg_seq/Makefile:DISABLE_MODULE += auto_init
thread_race/Makefile:DISABLE_MODULE += auto_init
thread_zombie/Makefile:DISABLE_MODULE += auto_init
unittests/Makefile:DISABLE_MODULE += auto_init
```

Probably minus these that manually initialize xtimer:

```
$ for i in $(git grep -l DIS.*auto | xargs dirname); do git grep "xtimer_init" $i/*.c; done
bloom_bytes/main.c:    xtimer_init();
driver_at86rf2xx/main.c:    xtimer_init();
gnrc_sixlowpan_frag/main.c:    xtimer_init();
lwip_sock_ip/stack.c:    xtimer_init();
lwip_sock_tcp/main.c:    xtimer_init();
lwip_sock_tcp/stack.c:    xtimer_init();
lwip_sock_udp/main.c:    xtimer_init();
lwip_sock_udp/stack.c:    xtimer_init();
posix_semaphore/main.c:    xtimer_init();
shell/main.c:    xtimer_init();
unittests/main.c:    xtimer_init();
```

#### Steps to reproduce the issue

Run any of the above tests with a board configured to use `stdio_rtt`.

#### Expected results

Test succeeds.

#### Actual results

Varying, as apparently the stdio_rtt code only falls back to periodic polling on read or write after empting either input or output queue. But the tests I tried just get stuck, mostly after printing the interactive_sync info line.

#### Versions

https://gist.github.com/6a04e06f0f9ec34836d4c22df7d7011c

kaspar030, 2020-01-14:
IMO we should check all of them if disabling auto_init makes sense at all. For some that is done without reason (e.g., `tests/float`).

Others seem to really need it (e.g., some gnrc tests need to avoid the standard initialization).

#13089 could be a way out.

fjmolinas, 2020-02-12:
This list should now be reduced to `tests/periph_rtt` `minimal` and `unittests.`

fjmolinas, 2020-04-14:
> This list should now be reduced to tests/periph_rtt minimal and unittests.

This list is now reduced to `minimal` and `unittests`. Should we consider it closed?