RIOT-OS/RIOT #13133: tests/pkg_tensorflow-lite: tests randomly failing on nrf52dk and esp32-wroom-32

kaspar030, 2020-01-15:
#### Description

captured output (from murdock):

nrf52: https://gist.github.com/c35dd538a85c7a52ff987da7214d20dc
esp32-wroom-32: https://gist.github.com/0a3ac0d0362d2d8a52e289c54c2eee3b

#### Steps to reproduce the issue

#### Expected results

#### Actual results

#### Versions

fjmolinas, 2020-01-15:
espr32-wroom-32 seems more serious since its a hardfault.

kaspar030, 2020-01-15:
> espr32-wroom-32 seems more serious since its a hardfault.

I guess you mean nrf52dk. But probably the cause is the same.

fjmolinas, 2020-01-15:
Failed again on PR #12848, hardfaults on nrf52 as well:

esp32-wroom-32: https://gist.github.com/fjmolinas/77016d18c258cf8d815f10a72b11fddb
nrf52dk: https://gist.github.com/fjmolinas/fb8bcab2a9baddeaf6f5797715ce3a69

fjmolinas, 2020-01-15:
Can't reproduce on `nrf52dk`:

```TOOLACHAIN=llvm RIOT_CI_BUILD=1 BUILD_IN_DOCKER=1 TRIBE_CI=1 BOARD=nrf52dk make -C tests/pkg_tensorflow-lite/ flash term -j3```

aabadie, 2020-01-15:
I tested on esp32-wroom-32 and nrf52832-mdk (same CPU as nrf52dk)

on esp32, it indeed doesn't work: the firmware doesn't start at all.

<details>

```
BUILD_IN_DOCKER=1 make BOARD=esp32-wroom-32 -C tests/pkg_tensorflow-lite/ flash test
make: Entering directory '/work/riot/RIOT/tests/pkg_tensorflow-lite'
ESP32_SDK_DIR should be defined as /path/to/esp-idf directory
ESP32_SDK_DIR is set by default to /opt/esp/esp-idf
Launching build container using image "riot/riotbuild:latest".
docker run --rm -t -u "$(id -u)" \
    -v '/usr/share/zoneinfo/Europe/Paris:/etc/localtime:ro' -v '/work/riot/RIOT:/data/riotbuild/riotbase:delegated' -e 'RIOTBASE=/data/riotbuild/riotbase' -e 'CCACHE_BASEDIR=/data/riotbuild/riotbase' -e 'BUILD_DIR=/data/riotbuild/riotbase/build' -e 'RIOTPROJECT=/data/riotbuild/riotbase' -e 'RIOTCPU=/data/riotbuild/riotbase/cpu' -e 'RIOTBOARD=/data/riotbuild/riotbase/boards' -e 'RIOTMAKE=/data/riotbuild/riotbase/makefiles'      \
    -e 'BOARD=esp32-wroom-32' \
    -w '/data/riotbuild/riotbase/tests/pkg_tensorflow-lite/' \
    'riot/riotbuild:latest' make  'BOARD=esp32-wroom-32' 'EXTERNAL_MODULE_DIRS=/data/riotbuild/riotbase/tests/pkg_tensorflow-lite/mnist' 
ESP32_SDK_DIR should be defined as /path/to/esp-idf directory
ESP32_SDK_DIR is set by default to /opt/esp/esp-idf
Building application "tests_pkg_tensorflow-lite" for "esp32-wroom-32" with MCU "esp32".

make[1]: Nothing to be done for 'prepare'.
make[1]: Nothing to be done for 'prepare'.
make[1]: Nothing to be done for 'prepare'.
"make" -C /data/riotbuild/riotbase/pkg/flatbuffers
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/pkg/esp32-wroom-32/flatbuffers/src -f /data/riotbuild/riotbase/pkg/flatbuffers/Makefile.flatbuffers
"make" -C /data/riotbuild/riotbase/pkg/gemmlowp
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/pkg/esp32-wroom-32/gemmlowp -f /data/riotbuild/riotbase/pkg/gemmlowp/Makefile.gemmlowp
"make" -C /data/riotbuild/riotbase/pkg/tensorflow-lite
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/pkg/esp32-wroom-32/tensorflow-lite/tensorflow/lite/c -f /data/riotbuild/riotbase/pkg/tensorflow-lite/Makefile.tensorflow-lite-c
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/pkg/esp32-wroom-32/tensorflow-lite/tensorflow/lite/core/api -f /data/riotbuild/riotbase/pkg/tensorflow-lite/Makefile.tensorflow-lite-core
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/pkg/esp32-wroom-32/tensorflow-lite/tensorflow/lite/kernels -f /data/riotbuild/riotbase/pkg/tensorflow-lite/Makefile.tensorflow-lite-kernels
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/pkg/esp32-wroom-32/tensorflow-lite/tensorflow/lite/kernels/internal -f /data/riotbuild/riotbase/pkg/tensorflow-lite/Makefile.tensorflow-lite-kernels-internal
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/pkg/esp32-wroom-32/tensorflow-lite/tensorflow/lite/micro/kernels -f /data/riotbuild/riotbase/pkg/tensorflow-lite/Makefile.tensorflow-lite-micro-kernels
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/pkg/esp32-wroom-32/tensorflow-lite/tensorflow/lite/micro/memory_planner -f /data/riotbuild/riotbase/pkg/tensorflow-lite/Makefile.tensorflow-lite-memory
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/pkg/esp32-wroom-32/tensorflow-lite/tensorflow/lite/micro -f /data/riotbuild/riotbase/pkg/tensorflow-lite/Makefile.tensorflow-lite
"make" -C /data/riotbuild/riotbase/boards/esp32-wroom-32
"make" -C /data/riotbuild/riotbase/boards/common/esp32
"make" -C /data/riotbuild/riotbase/core
"make" -C /data/riotbuild/riotbase/cpu/esp32
"make" -C /data/riotbuild/riotbase/cpu/esp32/freertos
"make" -C /data/riotbuild/riotbase/cpu/esp32/periph
"make" -C /data/riotbuild/riotbase/cpu/esp32/vendor
"make" -C /data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf
"make" -C /data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf/driver
"make" -C /data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf/esp32
"make" -C /data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf/soc
"make" -C /data/riotbuild/riotbase/cpu/esp32/vendor/esp-idf/spi_flash
"make" -C /data/riotbuild/riotbase/cpu/esp_common
"make" -C /data/riotbuild/riotbase/cpu/esp_common/vendor
"make" -C /data/riotbuild/riotbase/cpu/esp_common/vendor/xtensa
"make" -C /data/riotbuild/riotbase/drivers
"make" -C /data/riotbuild/riotbase/drivers/periph_common
"make" -C /data/riotbuild/riotbase/sys
"make" -C /data/riotbuild/riotbase/sys/auto_init
"make" -C /data/riotbuild/riotbase/sys/cpp11-compat
"make" -C /data/riotbuild/riotbase/sys/div
"make" -C /data/riotbuild/riotbase/sys/isrpipe
"make" -C /data/riotbuild/riotbase/sys/log
"make" -C /data/riotbuild/riotbase/sys/luid
"make" -C /data/riotbuild/riotbase/sys/newlib_syscalls_default
"make" -C /data/riotbuild/riotbase/sys/posix/pthread
"make" -C /data/riotbuild/riotbase/sys/random
"make" -C /data/riotbuild/riotbase/sys/random/tinymt32
"make" -C /data/riotbuild/riotbase/sys/stdio_uart
"make" -C /data/riotbuild/riotbase/sys/test_utils/interactive_sync
"make" -C /data/riotbuild/riotbase/sys/timex
"make" -C /data/riotbuild/riotbase/sys/tsrb
"make" -C /data/riotbuild/riotbase/sys/xtimer
"make" -C /data/riotbuild/riotbase/tests/pkg_tensorflow-lite/mnist
[... 145 lines cut ...]
Welcome to pyterm!
Type '/exit' to exit.
READY
s
START
main(): This is RIOT! (Version: 2020.01-devel-1843-g367250)

Digit prediction: 7

make: Leaving directory '/work/riot/RIOT/tests/pkg_tensorflow-lite'
```

</details>

[... cut ...]  the generated firmware has a very large `data` section, this doesn't seem normal:
```
   text	   data	    bss	    dec	    hex	filename
  92453	  76520	  19128	 188101	  2dec5	/data/riotbuild/riotbase/tests/pkg_tensorflow-lite/bin/esp32-wroom-32/tests_pkg_tensorflow-lite.elf
```

Compared to nrf52, it's only 280:
```
   text	   data	    bss	    dec	    hex	filename
  98940	    280	  13124	 112344	  1b6d8	/work/riot/RIOT/tests/pkg_tensorflow-lite/bin/nrf52832-mdk/tests_pkg_tensorflow-lite.elf
```

aabadie, 2020-01-15:
Maybe @gschorcht could have an idea on what is wrong with esp32 ?

aabadie, 2020-01-15:
For esp32, I suspect something when building cpp files. The generated firmware is always insanely bigger than with other ARM platforms. Example with `tests/cpp11_thread`:

- nrf52:
```
   text	   data	    bss	    dec	    hex	filename
  29776	    152	   2700	  32628	   7f74	/data/riotbuild/riotbase/tests/cpp11_thread/bin/nrf52832-mdk/tests_cpp11_thread.elf
```

- esp32:
```
   text	   data	    bss	    dec	    hex	filename
  81041	  69260	   6944	 157245	  2663d	/data/riotbuild/riotbase/tests/cpp11_thread/bin/esp32-wroom-32/tests_cpp11_thread.elf
```

gschorcht, 2020-01-15:
> For esp32, I suspect something when building cpp files. The generated firmware is always insanely bigger than with other ARM platforms.

ESP32 is not an ARM :thinking:

gschorcht, 2020-01-15:
I will check when I'm back at my desk.

aabadie, 2020-01-15:
> ESP32 is not an ARM

Sure, it was just to give some numbers to compare with. Maybe this doesn't make sense.

gschorcht, 2020-01-15:
> > ESP32 is not an ARM
> 
> Sure, it was just to give some numbers to compare with. Maybe this doesn't make sense.

I was just a bit confused because of the statement: *bigger than with other ARM platforms* :wink:

gschorcht, 2020-01-15:
I got `tests/pkg_tensorflow-lite` running on `esp32-wroom-32` :smile:
```
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
START
main(): This is RIOT! (Version: 2020.01-devel-1843-g36725)
Digit prediction: 7

```
but the problem is indeed serious and a more general problem of C++ on ESP32 :worried: We have to decide on how to deal with it.

**The Problem**

With `tests/pkg_tensorflow-lite` I got the following backtrace in GDB:
```
#0  abort () at /home/gs/esp/crosstool-NG/.build/src/newlib-2.2.0/newlib/libc/stdlib/abort.c:62
#1  0x400e0992 in __cxxabiv1::__terminate (handler=0x400d62c0 <abort>) at /home/gs/esp/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_terminate.cc:47
#3  0x400e0d91 in __cxxabiv1::__cxa_get_globals () at /home/gs/esp/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_globals.cc:133
#4  0x400e08fa in __cxxabiv1::__cxa_throw (obj=0x3ffb57c0, tinfo=0x3f4105f8 <typeinfo for __gnu_cxx::__concurrence_lock_error>, dest=0x400dfea0 <__gnu_cxx::__concurrence_lock_error::~__concurrence_lock_error()>) at /home/gs/esp/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/eh_throw.cc:65
#5  0x400dff71 in __gnu_cxx::__throw_concurrence_lock_error () at /home/gs/esp/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/ext/concurrence.h:102
#6  __gnu_cxx::__recursive_mutex::lock (this=<optimized out>) at /home/gs/esp/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/ext/concurrence.h:208
#7  (anonymous namespace)::mutex_wrapper::mutex_wrapper (this=0x3ffb2940 <main_stack+2788>) at /home/gs/esp/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/guard.cc:74
#8  0x400e0031 in __cxxabiv1::__cxa_guard_acquire (g=0x3ffb3c28 <guard variable for setup()::micro_error_reporter>) at /home/gs/esp/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/guard.cc:296
#9  0x400d98d9 in setup () at /home/gs/src/RIOT-Xtensa-ESP.working/tests/pkg_tensorflow-lite/mnist/main_functions.cc:50
#10 0x400d97d6 in main (argc=0, argv=0x0) at /home/gs/src/RIOT-Xtensa-ESP.working/tests/pkg_tensorflow-lite/main.cpp:28
```
That is, the system lock was simply an `abort`due to an exception in `__cxa_guard_acquire` which is called on static object initialization in the `setup` function. This led me to the problem.

When I was porting RIOT to ESP32, I started with the existing ESP32 tool chain from the vendor. Unfortunately, this toolchain was generated with `crosstool-NG` and thread support using option `--enable-threads=posix`. That in turn led to the problem that the `__cxa_guard_*` functions used the `pthread_once` function from module `pthread` where the parameter `once` was of incompatible type.

When I was trying to enable C++ on ESP32, I therefore replaced the `__cxa_guard_*` functions in `libstdc++.a` with an own version, see PR #12060. This workaround was working for all our C++ applications. Obviously, it doesn't work with `tests/pgk_tensorflow_lite`.

**How to solve**

The problem can be solved with a recompiled tool chain version in which the thread support is disabled. I have already such a version on my local system. 

That is, we have to roll out an updated toolchain. Afterwards, we can remove the workaround introduced with PR #12060.

**What are the options**

I suggest the following approach:

1. Blacklist ESP32 boards for the moment in ``tests/pgk_tensorflow_lite` as a quick fix of this issue.
2. Roll out an updated toolchain in docker
3. Remove the workaround implementation of `__cxa_guard_*` functions.

I could provide everything as different PRs within one day.

What do you think?

aabadie, 2020-01-15:
Thanks for the detailed and clear analysis @gschorcht.

> What do you think?

Your plan sounds good.

fjmolinas, 2020-01-15:
Besides `RIOT_VERSION` anything that I might be doing different that might be happening in murdock compared to `TOOLACHAIN=llvm RIOT_CI_BUILD=1 BUILD_IN_DOCKER=1 BOARD=nrf52dk make -C tests/pkg_tensorflow-lite/ flash term -j3`?

aabadie, 2020-01-16:
@gschorcht @fjmolinas, I opened #13140 in order to blacklist ESP32 architecture for tensorflow-lite until the toolchain is updated in Docker.

aabadie, 2020-01-16:
Regarding the nrf52dk, I can reproduce the hard fault with LLVM and not only on nrf52dk, it affects at least all Cortex-M cpus (M3, M4 and probably M7). I'm suspecting something with the FPU.

Maybe I suggest that we disable the LLVM toolchain on Cortex-M ?

aabadie, 2020-01-16:
@fjmolinas, there's a typo in your command (`TOOLACHAIN` instead of `TOOLCHAIN`) and I'm wondering the TOOLCHAIN variable is passed to Docker when it's used like this.

aabadie, 2020-01-16:
@fjmolinas, I opened https://github.com/tensorflow/tensorflow/issues/35939 on TensorFlow ;) because I don't really understand what the problem is (it happens in TensorFlow itself).
To not block other PRs with unrelated failures, I suggest to blacklist the LLVM toolchain for Cortex-M. Do you agree ? Note that I think this PR would also need a backport.

aabadie, 2023-01-04:
I think this issue can be closed since tensorflow-lite was replaced by tflite-micro and in #17908, the PR description mentioned it is working on esp32. There were also toolchain updates for esp32 during the last 2 years.

I don't have hardware to confirm that it also works on nrf52 or esp32. Maybe someone can confirm ?

gschorcht, 2023-01-06:
> I don't have hardware to confirm that it also works on nrf52 or esp32. Maybe someone can confirm ?

I can confirm that it works on esp32 as well as nrf52.

aabadie, 2023-01-06:
Thanks for testing! Then let's close this.

closed 2023-01-06 by aabadie