RIOT-OS/RIOT #13154: gcoap: features required to wrap SAUL

miri64, 2020-01-17:
While trying to expose a boards actuators and sensors via gCoAP, I noticed, that it is not yet as easy and  as it could be using SAUL. There are several features missing for that:

- [ ] **Adding a dynamic list of resources to a gCoAP listener**: While SAUL manages its resources in a linked list to allow for extensions from basically anywhere (heterogenous drivers), gCoAP has uses fixed-sized arrays to manage the resources of a listener. A way to deal with this could be to have "wildcard resources" that can be expanded by the resource handler. A way to add those `.well-known/core` would be to have a special handler for those kind of resources.
- [ ] **`phydat` de-/serializer** so that the data read from a device can be converted into JSON, CBOR, etc. and vice-versa
- [ ] **"reactive" SAUL** to implement events that trigger a CoAP OBSERVE notification

kb2ma, 2020-01-17:
Does the _Server path matching_ section in the [nanocoap doc](http://doc.riot-os.org/group__net__nanocoap.html) provide the wildcard you're after?

miri64, 2020-01-17:
> Does the _Server path matching_ section in the [nanocoap doc](http://doc.riot-os.org/group__net__nanocoap.html) provide the wildcard you're after?

If that sub-tree is then also advertised via `.well-known/core`, yes.

kb2ma, 2020-01-17:
I agree that a phydat de-/serializer is a good idea. It's not specific to gcoap though, is it? I'm actually implementing a [serializer](https://github.com/kb2ma/riot-apps/blob/wakaama-temp/wakaama-temp/temp_reader.c) for temperature for Wakaama in the temperature app in RIOT-OS/applications#66. Maybe this could be generalized further.

miri64, 2020-01-17:
> I agree that a phydat de-/serializer is a good idea. It's not specific to gcoap though, is it? I'm actually implementing a [...]

No, as this is about interfacing, I see this tracking issue as both for SAUL and gCoAP. In this case, this is more specific to saul / phydat.

maribu, 2020-01-17:
> maribu closed this 42 seconds ago

Sorry, big fingers on small screen.

If I remeber correctly, the handler for .well-known/core in nanocoap is just a regular callback, with a default implementation for that for you to pick. So you could provide your own handler that would construct the discovery response from the SAUL endpoints. Both the handler for the SAUL resources and the extended .well-known/core handlers would be good candidates for a utility module to nanocoap, as this seems a common use case.

kb2ma, 2020-01-17:
I also agree that it would be useful for an event in some other module to propagate through gcoap to emit a notification. I don't know how much of this mechanism belongs in gcoap itself though.

It seems like you're describing one use for an application protocol like LwM2M. Ironically, I'm wondering myself what it would take to build a more resource constrained implementation of LwM2M based on gcoap. :-)

miri64, 2020-01-17:
> […]  I don't know how much of this mechanism belongs in gcoap itself though.

Again: this issue is not solely about gcoap, but also SAUL.

miri64, 2020-01-17:
What I envisioned as an end product would be an optional module that maps SAUL resources to gcoap resources (let's call it `gcoap_saul`) and provides a gcoap listener for them. The list above points out the features missing in both gcoap and SAUL to make this kind of possible.

kb2ma, 2020-01-17:
> If I remeber correctly, the handler for .well-known/core in nanocoap is just a regular callback, with a default implementation for that for you to pick.

Right you are, @maribu! See _Resource list creation_ in the [gcoap doc](http://doc.riot-os.org/group__net__gcoap.html).

kb2ma, 2020-01-17:
> What I envisioned as an end product would be an optional module that maps SAUL resources to gcoap resources (let's call it gcoap_saul) and provides a gcoap listener for them.

Totally makes sense and timely, @miri64.

miri64, 2020-01-17:
> > If I remeber correctly, the handler for .well-known/core in nanocoap is just a regular callback, with a default implementation for that for you to pick.
> 
> Right you are, @maribu! See _Resource list creation_ in the [gcoap doc](http://doc.riot-os.org/group__net__gcoap.html).

Ok, I guess gcoap then provides everything I need and I was just too lazy to read the doc :-D.

miri64, 2020-01-17:
Mh, but wait... I thought that handler only encodes the link-format of a particular resource (at least that's how I interpret its usage in `examples/gcoap`). Or am I missing something? Can I just append more resources there? What I want is that the output of `.well-known/core` is *extended* by resources "hiding" behind a matched prefix. E.g. if I have a resource `saul/` that is matching to `saul/sense/light` and `saul/sense/mag`, I want both the links for `saul/sense/light` and `saul/sense/mag` to show up at `.well-known/core`.

kb2ma, 2020-01-17:
Yes, you could add more resources to the list as you describe. The gcoap example uses the default implementation `gcoap_encode_link()` to write the link and then only appends attributes. However, you could ignore the default implementation if you don't want the top level link (`<saul>`) on its own, and then write whatever you want.

miri64, 2020-01-17:
Ok, then for now all we need to kick-off a `gcoap_saul` module is a `phydat` serializer :-). Maybe, if I find some time this weekend I write one up for JSON.

Citrullin, 2020-05-12:
Great, this sounds like the feature I need for the wot implementation I want to create.

jue89, 2020-05-24:
> * **"reactive" SAUL** to implement events that trigger a CoAP OBSERVE notification

I think #14121 could be a solution for this requirement.

miri64, 2020-05-24:
Mhhh, if I understand #14121 correctly, it is more about providing a reactive alternative to SAUL. In my mind I was more thinking about extending the SAUL API for this.

jue89, 2020-05-24:
> Mhhh, if I understand #14121 correctly, it is more about providing a reactive alternative to SAUL. In my mind I was more thinking about extending the SAUL API for this.

I intend to extend SAUL - so I called it Extend Sensor Actuator Layer ;) Of course, this requires additional code by the SAUL sensor / actuator since it knows best when its state has changed significantly.

The user of the SAUL library doesn't have to know whether its a SAUL device or a eSAUL device. It just won't receive change notifications by the SAUL devices and, thus, won't send OBSERVE notifications.

Have thought about how to extend SAUL to meet the stated requirement? I'd love to hear about it to come up with a sustainable solution for this problem.

miri64, 2020-05-25:
> I intend to extend SAUL - so I called it Extended Sensor Actuator Uber Layer ;)

Ok, then maybe the fact that you gave it a new name (eSAUL) confused me. I would just stick with extended SAUL ;-) (mind the capitalization).