RIOT-OS/RIOT #13155: Can't enable GCLK Generators for samr30-xplained.

Alturon, 2020-01-17:
I am trying to get a gclk generator to connect to a timer/counter, and hooking up the generator to a GCLK_IO (as clock source). Only the generator 0 is active and makes the timers count, but other generators aren't starting, even though I try to enable them. Here is my [code](https://github.com/Alturon/counting_external_source):

dylad, 2020-01-20:
Hello @Alturon 
I am not sure to quite understand your issue and I am not sure this is related to RIOT.

By default, GCLK are disabled at reset and RIOT only enables what it needs in `cpu/saml21/cpu.c ` (since SAMR30 is a SAML21 flavor). There is no need to enable all GCLK since this is power-hungry.

In the current state of your code, you setup PA22 to be a input clock source connected to GCLK6 input. Did you connect an oscillator or something on PA22 ? Since you configure it as input, the pin expects to receive a clock from somewhere so it can feed GCLK6.

Alturon, 2020-01-20:
Thanks for responding. Sorry if I wasn't clear.

My end goal is to count the pulses (a signal coming from an external source) on PA22 using the Timer/Counter.

The signal doesn't come from an oscillator, but from a motor controller that basically outputs a pulse each time a step is traveled.