RIOT-OS/RIOT #13204: xtimer problem with nucleo32 boards

jack6000, 2020-01-26:
#### Description
Trying this little code:

    void main(void)
    {
        LED0_OFF;
        while(1) { 
            LED0_TOGGLE;
            xtimer_sleep(1);
        }
    }

On board nucleo-f091rc, works well, the onboard led blink every second. But if i replace
`xtimer_sleep(1)`
by
`xtimer_usleep(500)`

it doesn't work anymore. The led doesn't blink.

Tested also on nucleo-f031k6, but nothing works at all.

#### Versions
Operating System Environment
-----------------------------
         Operating System: "KDE neon" "5.17"
                   Kernel: Linux 4.15.0-74-generic x86_64 x86_64
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.0.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.4.1
                    cmake: cmake version 3.16.3
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.17.1
                     make: GNU Make 4.1
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 2.7.17
                  python2: Python 2.7.17
                  python3: Python 3.6.9
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing

aabadie, 2020-01-26:
> The led doesn't blink.

I think it does, but your eyes are not able to see it: xtimer_usleep delay parameter is in microseconds, not milliseconds, so in your code it blinks every half milliseconds.
Use 500 * US_PER_MS, and it should be good.

jack6000, 2020-01-27:
Yes, i tought it was in milliseconds...so it works now on nucleo091rc.
But i confirm that nothing works on nucleo-f031k6

aabadie, 2020-01-27:
> i confirm that nothing works on nucleo-f031k6

Do you mean a simple hello-world doesn't work ? If I remember correctly, there's a very small jumper at the back of the board that should be plugged.

jack6000, 2020-01-27:
Sorry i wasn't clear enough. 
So for now, i will only focus on nucleo-f031k6 board.
This example works well:

    void main(void)
    {
        LED0_OFF;
        while(1) { 
            LED0_TOGGLE;
            for (uint32_t i=0; i<1000000; i++)
               __asm("nop");
        }
    }

The onboard led is blinking

The problem is when i use the xtimer module. So if i replace the delay loop with this:

    void main(void)
    {
        LED0_OFF;
        while(1) { 
            LED0_TOGGLE;
            xtimer_sleep(1);
        }
    }

The led lights up then nothing. So i really think that the xtimer_sleep() function is the cause of the problem.
On native platform it works, i can see the line "LED_RED_TOGGLE"  every second in the terminal.

aabadie, 2020-10-07:
@fjmolinas I think we can safely close this one. It was certainly fixed by #14974.

fjmolinas, 2020-10-07:
Yes lets close.

closed 2020-10-07 by fjmolinas