RIOT-OS/RIOT #13205: Support for CC1312R

amovitz, 2020-01-27:
There seems to be basic support for the CC1312R-launchpad (which I think is the old name, TI does not make clear why it goes by multiple names), but apparently no radio support? I've got the shell working, but `ifconfig` doesn't return anything and there's no radio running in `ps`. Upon inspection, it looks like there was an old issue opened to port this board, but it was closed from a lack of interest.

I'd like to help/lead the development effort for the CC1312R in general. So far, what I've seen is that there is basic CPU support, but almost no peripheral support past the LEDs that come on the LAUNCHXL-CC1312R1.

My basic question: since there's a lot of shared code for this chip, what needs to be done to get basic radio support? Do I need to do a full port and port the radio drivers from the TI SDK, or is there already a jumping off point for the simplelink radio?

aabadie, 2020-01-27:
I think @jeandudey is working on a driver implementation. See https://github.com/RIOT-OS/RIOT/pull/13134#issuecomment-575938801 about that.

jeandudey, 2020-01-27:
Hi @amovitz, as @aabadie said I'm working to get a functional and working radio driver, I'm still at very early stages because there isn't much documentation and example code on the driver, however any help is appreciated though.

I have been tried to use the TI CoreSDK directly in RIOT but some incompabilities prohibited me from doing that, as the RF driver uses a programming model based on *hardware and software interrupts* that doens't play very well with RIOT's context switching interrupts and such, and as a result every attempt to use it resulted in a crash. Mainly because they use their own vector table, and for software interrupts they use the `PendSV_IRQn` interrupt which is used by RIOT for thread context switching, this could be indeed changed in the TI's CoreSDK, but it means recompiling all the libraries of it (it's pretty undocumented the build instructions) and also means modifying it including some sorts of more complexity.

I've been researching the SDK code, and trying to get a minimal working implementation through the process, I haven't been documenting as much the code but reading the CoreSDK RF driver is a good head start, you can find it [here](https://github.com/contiki-ng/coresdk_cc13xx_cc26xx). Also a good resource is reading the RF implementation for `CC13x0` and `CC26x0` made by [Contiki-NG](https://github.com/contiki-ng/contiki-ng/blob/develop/arch/cpu/cc26x0-cc13x0/rf-core/rf-core.c), it's not exactly the same but provides the idea of how things work with the RF Core. Here is the for further documentation [CC13x2, CC26x2 Technical Reference Manual](https://www.ti.com/lit/ug/swcu185d/swcu185d.pdf).
___
In the btcven/RIOT#5 PR is where I have been putting all of the work, very little for now as I'm still learning.

Any help and/or review is welcome!

kaspar030, 2020-01-27:
> I'm working to get a functional and working radio driver,

I got a working radio driver (ported from OpenThred) [here](https://github.com/kaspar030/RIOT/tree/cc2650_driverlib), maybe that helps? Didn't get around to PR'ing it...

jeandudey, 2020-01-27:
@kaspar030 It helps a lot actually, thanks for sharing it, I'm going to look around it :+1:

amovitz, 2020-02-20:
Closing because a lot of great work is being done in #13295

closed 2020-02-20 by amovitz

jeandudey, 2020-02-20:
> Closing because a lot of great work is being done in #13295

Thanks! With some improvements on the code this driver has some future ;)