RIOT-OS/RIOT #13208: Best practices for adding SPI and I2C code to a driver?

wingunder, 2020-01-27:
#### Description
I have added working SPI support for the [lpsxxx](https://github.com/RIOT-OS/RIOT/tree/master/drivers/lpsxxx) driver and would like to share it. However, looking at the drivers that have SPI as well as I2C support at the moment, I found the following:

   * [bmx280](https://github.com/RIOT-OS/RIOT/tree/master/drivers/bmx280) (last relevant modification: 10.01.2019)
   * [lis2dh12](https://github.com/RIOT-OS/RIOT/tree/master/drivers/lisdh12) (last relevant modification: 23.01.2018)
   * [pn532](https://github.com/RIOT-OS/RIOT/tree/master/drivers/pn532) (last relevant modification: 29.08.2017)

The bmx280 and lis2dh12 drivers rely on defines, that put the driver into either SPI or I2C mode. (BMX280_USE_SPI and MODULE_LIS2DH12_SPI). The code that routes the calls to SPI or I2C is completely split and compiled in, depending on the defines.

The pn532 driver uses 2 defines (PN532_SUPPORT_I2C and PN532_SUPPORT_SPI) for routing the bus communication to I2C or SPI. The code that routes the calls to SPI or I2C is the same, but splits up at the point where a low-level I2C or SPI call gets done, using the defines. i.e. One implementation with may 'defined' checks.

The bmx280, lis2dh12 and pn532 drivers can only operate in one of the 2 modes, which is determined at compile time.

The way that I implemented I2C and/or SPI for the lpsxxx, relies on setting up function pointers for the routing the bus functions to either SPI or I2C, based on driver parameters, at startup (not compile time). This means that a board can have more than one of the same sensors, connected via SPI _or_ I2C. However, I can't think of any hardware, other than test setups, that might want this. A major advantage however, is that the I2C and SPI code can be re-used by different sensors. i.e. The SPI/I2C implementation is only done once, so much less code maintenance.

The function pointer solution however has a slight code space and minute, calling overhead. I have not got hard figures on this yet, as I first wanted to find out if this kind of implementation makes sense.

### Questions
1. Will a function pointer type solution for pivoting between I2C and SPI implementations, as described above, be useful for RIOT?
2. Are there any best-practice way of implementing SPI _and_ I2C for a single driver, in RIOT?

### Useful links
BTW, I happened to stumble accross the following. If this is related, it should also be keep in mind:
https://github.com/RIOT-OS/RIOT/blob/c5dfc7d16c8032dac8797d0c37bec7a10f1caf3e/drivers/include/net/netdev.h#L51-L62

kaspar030, 2020-01-27:
> * Will a function pointer type solution for pivoting between I2C and SPI implementations, as described above, be useful for RIOT?
> * Are there any best-practice way of implementing SPI _and_ I2C for a single driver, in RIOT?

As you've seen, both implementations and opinions on this vary.

If the function pointer selection is done at runtime, the driver cannot compile without both I2C and SPI being available, right? To me, this should be avoided. Personally I'd rather use a `switch (dev->bus_type) {
case DEV_I2C: return device_read_i2c(dev); case DEV_SPI: ... default: fail_hard() }`. With some thought, the not used version can be compiled out, and the logic is in .text (no RAM overhead).

kaspar030, 2020-01-27:
> However, I can't think of any hardware, other than test setups, that might want this.

Yup. But IMO, with proper code organization, there's little overhead if only one case is used, but we don't close opportunities for our test setups. ;)

> A major advantage however, is that the I2C and SPI code can be re-used by different sensors.

Could you give an example for this?

wingunder, 2020-01-27:
@kaspar030: Thanks for your answer.

> If the function pointer selection is done at runtime, the driver cannot compile without both I2C and SPI being available, right? To me, this should be avoided. Personally I'd rather use a `switch [...]

Well, actually I'm  doing more or less what you suggested. I can compile out the code that I don't use, with defines. However I have a 'setup' call that has a switch and sets up all the required function pointers at setup-time, not at call-time, as I think you're suggesting.

I think it'll be easier if I make a pull request with what I have at the moment. It's working for I2C and SPI, but at least the documentation is still missing. Discussing stuff like this will be easier. Is this OK with you?

wingunder, 2020-01-27:
> Yup. But IMO, with proper code organization, there's little overhead if only one case is used, but we don't close opportunities for our test setups. ;)

Thanks, I'm of the same opinion.

> > A major advantage however, is that the I2C and SPI code can be re-used by different sensors.
> 
> Could you give an example for this?

Sure. I don't have anything in hand at the moment, but I could try it with the lis2dh12 driver, as it already has an I2C/SPI implementation. This would also reveal some true stats on the amount of overhead that it will add. I however doubt that I have hardware for this, but for demonstration purposes, this shouldn't be a problem.

closed 2021-02-13 by stale[bot]