RIOT-OS/RIOT #13209: gnrc_border_router is crashing after some time

aabadie, 2020-01-27:
#### Description

After some time (~1h30), the `examples/gnrc_border_router` ends up in a failed assertion on iotlab-m3.

Note that the iotlab-m3 uses a 16bit width xtimer.

#### Steps to reproduce the issue

Run the following command with an iotlab-m3 plugged on your computer:
```
CFLAGS=-DDEBUG_ASSERT_VERBOSE make BOARD=iotlab-m3 -C examples/gnrc_border_router flash term
```
The `DEBUG_ASSERT_VERBOSE` is set in order to help finding out where the failed assertion occurs.

After about 1h30, the application crashes:
```
uhcp_client(): sending REQ...
got packet from fe80::dcb3:62ff:fe9e:5a59 port 53900
uhcp: push from fe80::dcb3:62ff:fe9e:5a59:53900 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): got same prefix again
uhcp_client(): sending REQ...
got packet from fe80::dcb3:62ff:fe9e:5a59 port 39408
uhcp: push from fe80::dcb3:62ff:fe9e:5a59:39408 prefix=2001:db8::/64
gnrc_uhcpc: uhcp_handle_prefix(): got same prefix again
sys/xtimer/xtimer_core.c:250 => 0x8000ff1
*** RIOT kernel panic:
FAILED ASSERTION.

ISR stack overflowed
Stack pointer corrupted, reset to top of stack
FSR/FAR:
 CFSR: 0x00009200
 HFSR: 0x40000000
 DFSR: 0x00000000
 AFSR: 0x00000000
 BFAR: 0x1ffffff8
Misc
EXC_RET: 0xfffffff1
```

Could this be related to #9530 ?

#### Expected results

The application runs fine endlessly

#### Actual results

A failed assertion is raised after some amount of time (1h30).

#### Versions

```
Operating System Environment
-----------------------------
         Operating System: "Ubuntu" "19.10 (Eoan Ermine)"
                   Kernel: Linux 5.3.0-26-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
                  avr-gcc: avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1759) 5.4.0
         mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
               msp430-gcc: missing
     riscv-none-embed-gcc: riscv-none-embed-gcc (GNU MCU Eclipse RISC-V Embedded GCC, 64-bit) 8.2.0
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 9.0.0-2 (tags/RELEASE_900/final)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: "2.1.0"
  riscv-none-embed-newlib: "3.0.0"
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.7.3
                    cmake: cmake version 3.13.4
                 cppcheck: Cppcheck 1.88
                  doxygen: 1.8.13
                      git: git version 2.20.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-00957-g9de7d9c8 (2019-11-11-22:36)
                   python: Python 2.7.17
                  python2: Python 2.7.17
                  python3: Python 3.7.5
                   flake8: 3.7.7 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.7.5 on Linux
               coccinelle: spatch version 1.0.4 with Python support and with PCRE suppor
```

fjmolinas, 2020-01-27:
@Hyungsin can you take a look as well?

fjmolinas, 2020-01-27:
@aabadie can you try to reproduce on 2019.10?

fjmolinas, 2020-01-27:
I'm going to run some long experiments using gnrc and see If I get somthing similar

Hyungsin, 2020-01-27:
I added this assert to catch any timer mis-ordering.
I looked into the code a bit and found a possible error, which is about comparing two absolute times instead of two offsets. I will create a PR to resolve this. Thanks!

Hyungsin, 2020-01-27:
@aabadie please test if #13212 solves the problem.

aabadie, 2020-01-27:
> please test if #13212 solves the problem.

Thanks for providing a fix @Hyungsin !
I'll give it a try tomorrow.

fjmolinas, 2020-01-28:
I was able to reproduce the issue between two nodes running `gnrc_networking`.

```
2020-01-27 19:29:47,680 # 108 bytes from fe80::806b:deec:5834:6579: icmp_seq=4194 ttl=64 rssi=-34 dBm time=18.000 ms
2020-01-27 19:29:48,688 # 108 bytes from fe80::806b:deec:5834:6579: icmp_seq=4195 ttl=64 rssi=-34 dBm time=20.556 ms
2020-01-27 19:29:49,677 # 0x8000b9f
2020-01-27 19:29:49,678 # *** RIOT kernel panic:
2020-01-27 19:29:49,678 # FAILED ASSERTION.
2020-01-27 19:29:49,678 #
2020-01-27 19:29:49,681 #       pid | name                 | state    Q | pri | stack  ( used) | base addr  | current
2020-01-27 19:29:49,683 #         - | isr_stack            | -        - |   - |    512 (  512) | 0x20000000 | 0x20000128
2020-01-27 19:29:49,684 #         1 | idle                 | running  Q |  15 |    256 (  128) | 0x2000041c | 0x2000049c
2020-01-27 19:29:49,685 #         2 | main                 | bl rx    _ |   7 |   1536 (  892) | 0x2000051c | 0x20000884
2020-01-27 19:29:49,686 #         3 | pktdump              | bl rx    _ |   6 |   1536 (  240) | 0x200031ac | 0x200036bc
2020-01-27 19:29:49,689 #         4 | 6lo                  | bl rx    _ |   3 |   1024 (  408) | 0x20003e0c | 0x2000410c
2020-01-27 19:29:49,690 #         5 | ipv6                 | bl rx    _ |   4 |   1024 (  392) | 0x20001020 | 0x2000131c
2020-01-27 19:29:49,691 #         6 | udp                  | bl rx    _ |   5 |   1024 (  252) | 0x200045f0 | 0x200048f4
2020-01-27 19:29:49,692 #         7 | at86rf2xx            | bl rx    _ |   2 |   1024 (  508) | 0x20000bdc | 0x20000ea4
2020-01-27 19:29:49,693 #         8 | RPL                  | bl rx    _ |   5 |   1024 (  196) | 0x20003800 | 0x20003b3c
2020-01-27 19:29:49,704 #           | SUM                  |            |     |   8960 ( 3528)
2020-01-27 19:29:49,704 #
2020-01-27 19:29:49,704 # *** halted.
2020-01-27 19:29:49,704 #
2020-01-28 08:46:42,996 # Exiting Pyterm
/home/francisco/workspace/RIOT/Makefile.include:683: recipe for target 'term' failed
make: *** [term] Interrupt
```

MichelRottleuthner, 2020-01-28:
With a small modification of `tests/xtimer_drift` I can reproduce the bug reliably within a few seconds on an `iotlab-m3`.
In `xtimer_core.c` set `_xtimer_current_time` to an initial value like this:
```c
volatile uint64_t _xtimer_current_time = 0x00000000FFB00000;
```
It doesn't have to be *exactly* that value but it should just be close to the 32bit wraparound and this value worked well for me.

On master it will crash as seen above close to the wraparound.

With #13212 applied I'm not able to reproduce it anymore.

kaspar030, 2020-01-28:
> With a small modification of `tests/xtimer_drift` I can reproduce the bug reliably within a few seconds on an `iotlab-m3`.

:+1:

> It doesn't have to be _exactly_ that value but it should just be close to the 32bit wraparound and this value worked well for me.
> 
> On master it will crash as seen above close to the wraparound.

That [shouldn't™](https://github.com/RIOT-OS/RIOT/pull/9530#issuecomment-572033911) have happened. :wink:

fjmolinas, 2020-01-28:
> With a small modification of `tests/xtimer_drift` I can reproduce the bug reliably within a few seconds on an `iotlab-m3`.
> In `xtimer_core.c` set `_xtimer_current_time` to an initial value like [...]

@MichelRottleuthner wha I don't undetstand is that wasn't 32b it wraparound  tested in #9365

>I did a run over the full 32 bit cycle (4k-something seconds) on iotlab-m3 and nrf51dk, neither showing high jitter and accumulating drift

In what way does this test differ from the one you did?

Also I think if that small change  you suggest tests the wraparound scenario it should be added to our tests, as well as 64 bit wraparround.

aabadie, 2020-01-28:
> it should be added to our tests, as well as 64 bit wraparround

+1

MichelRottleuthner, 2020-01-28:
@kaspar030  yes, and this case shows why I wrote it like that ;)

> In what way does this test differ from the one you did?

It was probably related to the fact that `xtimer_drift` didn't reproduce the problem and is only built for 32 bit. I think I also noted somewhere that It showed a problem with the wraparound *sometimes*.

> Also I think if that small change you suggest tests the wraparound scenario it should be added to our tests, as well as 64 bit wraparround.

+1

aabadie, 2020-07-01:
I think this was fixed by #13237 and should have been closed already.

closed 2020-07-01 by aabadie