RIOT-OS/RIOT #13265: tests: compilation of some test applications fail with NDEBUG

gschorcht, 2020-02-02:
#### Description

Some test applications use `assert` to check the test results for correctness. Applications that rely on `assert` should undefine `NDEBUG` at the beginning of their source code. Otherwise their compilation fail in
```
dist/tools/compile_and_test_for_board/compile_and_test_for_board.py
```
with defined environment variable `CLFAGS='-NDEBUG'`.

The following test application could already be identified:
```
- tests/gnrc_sock_ip
- tests/periph_eeprom
- tests/thread_msg_block_race
```

#### Steps to reproduce the issue

Compile these application for any board, for example:
```
CFLAGS='-DNDEBUG' make BOARD=arduino-mega2560 -C tests/gnrc_sock_ip
CFLAGS='-DNDEBUG' make BOARD=arduino-mega2560 -C tests/periph_eeprom
CFLAGS='-DNDEBUG' make BOARD=arduino-mega2560 -C tests/thread_msg_block_race
```

kaspar030, 2020-02-03:
> Applications that rely on `assert` should undefine `NDEBUG` at the beginning of their source code.

Good idea!

kaspar030, 2020-02-03:
> Some test applications use `assert` to check the test results for correctness.

BTW, IMO that should be avoided. While convenient, IMO we should be more interested in running tests that are compiled with different compilations, e.g., LTO enabled, or different optimizations, or with NDEBUG=1 (testing the rest of the codebase, not the test application itself).

Maybe we should work towards using something like `expect(condition)`, which would be like `assert()` but unaffected by NDEBUG.

kaspar030, 2020-02-03:
> Maybe we should work towards using something like `expect(condition)`, which would be like `assert()` but unaffected by NDEBUG.

Like this: https://github.com/RIOT-OS/RIOT/pull/13268

PR #13466 (2020-03-12): tests: fix compilation problems with NDEBUG
### Contribution description

This PR fixes compilation problems (hopefully all remaining) when `NDEBUG` is defined.

Now where PR #13268 is merged, test applications should use `expect` instead of `assert` for testing conditions and results. Otherwise, they are not compilable when `NDEBUG` is defined. In difference to `assert`, `expect`  is not compiled out when compiling with  `NDEBUG`.

**NOTE** commit https://github.com/RIOT-OS/RIOT/pull/13466/commits/0c1baf267eb57344e41f411e4a8fb4213eb1f46d is just for the compilation test with `NDEBUG` and has  [... cut]
closed 2020-03-12 by miri64