RIOT-OS/RIOT #13267: Failing tests on MSP430 (z1)

fjmolinas, 2020-02-03:
#### Description

Opening this issue for tracking the problems with tests found in 2020.01 release testing. Detailed issues can be found [here](https://github.com/fjmolinas/ci-results/tree/2020.01-RC1_results/results/z1).

#### Steps to reproduce the issue

```
./dist/tools/compile_and_test_for_board/compile_and_test_for_board.py . z1 --jobs=4
```

#### Expected results

No failure

#### Actual results

Failures during compilation:
- [ ] [tests/gnrc_ipv6_nib](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/gnrc_ipv6_nib/compilation.failed)
- [ ] [tests/gnrc_ipv6_nib_6ln](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/gnrc_ipv6_nib_6ln/compilation.failed)
- [ ] [tests/gnrc_sixlowpan_iphc_w_vrb](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/gnrc_sixlowpan_iphc_w_vrb/compilation.failed)
- [ ] [tests/gnrc_sock_async_event](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/gnrc_sock_async_event/compilation.failed)
- [ ] [tests/gnrc_sock_ip](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/gnrc_sock_ip/compilation.failed)
- [ ] [tests/gnrc_sock_neterr](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/gnrc_sock_neterr/compilation.failed)
- [ ] [tests/gnrc_sock_udp](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/gnrc_sock_udp/compilation.failed)
- [ ] [tests/pkg_ucglib](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/pkg_ucglib/compilation.failed)

Failures during test:
- [ ] [tests/bench_runtime_coreapis](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/bench_runtime_coreapis/test.failed)
- [ ] [tests/driver_grove_ledbar](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/driver_grove_ledbar/test.failed)
- [ ] [tests/driver_hd44780](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/driver_hd44780/test.failed)
- [ ] [tests/driver_my9221](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/driver_my9221/test.failed)
- [ ] [tests/event_wait_timeout](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/event_wait_timeout/test.failed)
- [ ] [tests/evtimer_msg](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/evtimer_msg/test.failed)
- [ ] [tests/periph_gpio](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/periph_gpio/test.failed)
- [ ] [tests/periph_timer_short_relative_set](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/periph_timer_short_relative_set/test.failed)
- [ ] [tests/pkg_c25519](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/pkg_c25519/test.failed)
- [ ] [tests/pkg_heatshrink](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/pkg_heatshrink/test.failed)
- [ ] [tests/pkg_lora-serialization](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/pkg_lora-serialization/test.failed)
- [ ] [tests/pkg_tiny-asn1](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/pkg_tiny-asn1/test.failed)
- [ ] [tests/pkg_tweetnacl](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/pkg_tweetnacl/test.failed)
- [ ] [tests/pthread_tls](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/pthread_tls/test.failed)
- [ ] [tests/shell](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/shell/test.failed)
- [ ] [tests/sys_crypto](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/sys_crypto/test.failed)
- [ ] [tests/thread_flags_xtimer](https://github.com/fjmolinas/ci-results/edit/2020.01-RC1_results/results/z1/tests/thread_flags_xtimer/test.failed)

#### Versions

2020.01-RC1, although issues are still present in master.

Would be interesting to see how many are fixed or changed with #12457

kaspar030, 2020-02-03:
I've been working on updating our toolchain build script, mainly to create a properly compiled gcc elf newlib toolchain for #12457. I't almost done, mainly needing a github repo with travis configured to build this automatically...

I suggest concentrating on that!

fjmolinas, 2020-08-06:
With #14693 in this one can be closed, only failing tests are:

```
- [tests/driver_grove_ledbar](tests/driver_grove_ledbar/test.failed)
- [tests/driver_hd44780](tests/driver_hd44780/test.failed)
- [tests/driver_my9221](tests/driver_my9221/test.failed)
- [tests/periph_timer_short_relative_set](tests/periph_timer_short_relative_set/test.failed)
```

which are all expected failures, thanks for the great work @kaspar030!

closed 2020-08-06 by fjmolinas