RIOT-OS/RIOT #13280: nrf52: Not able to add global or ULA address to interface

Citrullin, 2020-02-04:
#### Description
Not sure, if this is a bug or if I just configured it not correct. I already wrote an email, but I guess this setup is a bit too complicated for an email.
Fritzbox cable acts as a Modem (bridge mode) and is connected to an Archer C7 v2. The Archer acts as the main router to the internet. The router gets an /56 range via prefix delegation. 
I compiled a fresh OpenWRT (v19.07.1) with all necessary 6lowpan_bluetooth modules.
The Archer C7 uses a BLE USB Stick. After connecting the nrf52832 device to the router, I tried to add a global and ULA address to the interface.
```
ifconfig 9 add 2aaa:8aaa:baaa:b710::2/64
```
and
```
ifconfig 9 add fdc9:515d:129c::2/64
```
Both don't work and the shell is not responding. Instead I can just execute the next command. After checking the interface with ifconfig again, the interface still has only the link local address.
```
> ifconfig 9 add fdc9:515d:129c::2
ifconfig
2020-02-04 13:22:17,985 #  ifconfig
2020-02-04 13:22:17,988 # Iface  9  HWaddr: CC:C6:52:01:8E:8C 
2020-02-04 13:22:17,993 #           L2-PDU:1280 MTU:1280  HL:64  RTR  
2020-02-04 13:22:17,995 #           6LO  IPHC  
2020-02-04 13:22:18,000 #           Source address length: 6
2020-02-04 13:22:18,001 #           Link type: wireless
2020-02-04 13:22:18,011 #           inet6 addr: fe80::ccc6:52ff:fe01:8e8c  scope: link  VAL
2020-02-04 13:22:18,012 #           inet6 group: ff02::2
2020-02-04 13:22:18,013 #           inet6 group: ff02::1
2020-02-04 13:22:18,015 #           inet6 group: ff02::1:ff01:8e8c
2020-02-04 13:22:18,018 #           inet6 group: ff02::1a
2020-02-04 13:22:18,019 #           
2020-02-04 13:22:18,022 #           Statistics for Layer 2
2020-02-04 13:22:18,025 #             RX packets 0  bytes 0
2020-02-04 13:22:18,030 #             TX packets 0 (Multicast: 0)  bytes 852
2020-02-04 13:22:18,033 #             TX succeeded 0 errors 0
2020-02-04 13:22:18,035 #           Statistics for IPv6
2020-02-04 13:22:18,039 #             RX packets 45  bytes 5620
2020-02-04 13:22:18,043 #             TX packets 32 (Multicast: 18)  bytes 2018
2020-02-04 13:22:18,047 #             TX succeeded 32 errors 0
2020-02-04 13:22:18,047 # 
```
The configured interface in OpenWRT looks like [this](https://imgur.com/a/8ctUHG9) and [this](https://imgur.com/a/frYKx7N).

#### Steps to reproduce the issue
1. Modem to the internet with IPv6 (Fritz!Box 6490 in my case)
2. Configured prefix delegation of /56 (/48 should also work) at the modem
3. OpenWRT (v19.07.1) with compiled 6lowpan_bluetooth kernel modules. Described [here](https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/6lowpan-for-bluetooth-low-energy-on-openwrt).
4. Configured OpenWRT interfaces as described [here](https://devzone.nordicsemi.com/nordic/nordic-blog/b/blog/posts/openwrt-series-1-setting-up-tp-link-wdr4300-or-n75).
5. Flash gnrc_networking example on the nrf52832
6. Get BLE Mac address of the nrf52832 with executing `ble info` on the RIOT shell.
7. Connect the nrf52 with the OpenWRT router. Execute the following on the OpenWRT router:
MAC_ADDR for example CC:C5:51:01:8E:8C
```
echo "connect MAC_ADDR 2" >/sys/kernel/debug/bluetooth/6lowpan_control
```
8. Get the link local address of the nrf52 and ping this address on the OpenWRT router with:
Replace fe80::ccc6:52ff:fe01:8e8c with the address of your nrf52 device
```
ping6 fe80::ccc6:52ff:fe01:8e8c%bt0
```
You should be able to reach the device. If not, something went already wrong.
9. Try to add your global and ULA address to your nrf52. If your bt0 interface has the prefix:
`2aaa:8aaa:baaa:b700::1/60` or `2aaa:8aaa:baaa:b700::1/64`
Give the nrf52 the suffix ::2. In this case:
```
ifconfig 9 add 2aaa:8aaa:baaa:b700::2/64
```
The same doesn't work the configured ULA:
```
ifconfig 9 add fdc9:515d:129c::2/64
```
You need to get the ID of your interface first with `ifconfig`. In this case 9.

#### Expected results

#### Actual results

#### Versions

Citrullin, 2020-02-06:
I guess @haukepetersen could be familiar with this. Haven't tried it with other radios yet.

Citrullin, 2020-07-07:
@haukepetersen @miri64 The goal of it is basically to use BLE instead of slip or something else. I am not sure, if this is even possible. There is still no GHC in the linux kernel, I guess.

miri64, 2020-07-07:
> @haukepetersen @miri64 The goal of it is basically to use BLE instead of slip or something else. I am not sure, if this is even possible. I guess the header compression in linux is still missing and therefore it is not able to understand the header. But I am not that familiar with the status of it.

Have you the 6LoWPAN module loaded? Header compression should be possible with that. Most of what can be found in the `nordic_softdevice_ble` regarding enabling BLE should still be correct.

Citrullin, 2020-07-07:
> Have you the 6LoWPAN module loaded? Header compression should be possible with that. Most of what can be found in the `nordic_softdevice_ble` regarding enabling BLE should still be correct.

Yes. 
```
modprobe bluetooth_6lowpan
echo 1 > /sys/kernel/debug/bluetooth/6lowpan_enable
```
In openwrt it was a bit different, as a I remember. Yes, but the connection changed. It's not 1 anymore. The last time I checked you needed to connect it with 2. But I don't have an idea what the difference is. 
Okay, then there is only the adding the IP address issue in RIOT. Hopefully it works then. :)

miri64, 2020-07-07:
Which one is the router? For Linux as the router you need to install and run radvd to advertise the global prefix. See https://github.com/RIOT-Makers/wpan-raspbian/wiki/Setup-native-6LoWPAN-router-using-Raspbian-and-RADVD.

Citrullin, 2020-07-07:
> Which one is the router? For Linux as the router you need to install and run radvd to advertise the global prefix. See https://github.com/RIOT-Makers/wpan-raspbian/wiki/Setup-native-6LoWPAN-router-using-Raspbian-and-RADVD.

The Acher C7 (OpenWRT) is the router. It's [already in OpenWRT](https://openwrt.org/docs/guide-user/network/ipv6/start). I don't think they use radvd for router advertisements.

miri64, 2020-07-07:
> I don't think they use radvd for router advertisements.

OpenWRT uses the [`odhcpd`](https://openwrt.org/docs/techref/odhcpd) as the default router advertisement daemon, afaik. At least mine does.

haukepetersen, 2020-08-05:
@Citrullin I am not 100% but I have an assumption where this might be coming from: in the default configuration, one can only set I think 1 additional address per interface. So when you add the first ULA, that memory spot is taken and the netif will not accept any additional addresses. But whats strange is, that `ifconfig` should actually respond with an error message if that would be the case.

@miri64 could you quickly remind us what the config value is to increase the number of addresses per interface? Thx

Aside from that I don't have much experience connecting RIOT BLE nodes to Linux, thats something somewhere on my TODO list to explore this a little further...

miri64, 2020-08-05:
> @miri64 could you quickly remind us what the config value is to increase the number of addresses per interface? Thx

https://doc.riot-os.org/group__net__gnrc__netif__conf.html#ga366596e311b4a450887a1a180ae66fb3
Depending on the network and the address desired to be added, one might also need to increase the number of multicast groups https://doc.riot-os.org/group__net__gnrc__netif__conf.html#ga5dec745d3d45a5bdd2363120680b102e