RIOT-OS/RIOT #13285: Bug: debugging session Dies

kfessel, 2020-02-04:
#### Description
`$_TARGETNAME configure -rtos auto`

seemingly leads to maldetection of a different RTOS with Ubuntu upstream openocd-0.10.0-6

`gdb> attach <openocd pid>`

provides following backtrace

`
Thread 1 "openocd" received signal SIGSEGV, Segmentation fault.
uCOS_III_update_thread_offsets (rtos=0x564ae6dc72e0) at src/rtos/uCOS-III.c:193
193 src/rtos/uCOS-III.c: No such file or directory.
`
after that openocd dies

i patched the pm_set_lowest to do nothing to avoid missing clock beeing the reason

#### Steps to reproduce the issue

build and flash default example (does happen with other applications to)

then start debugging

    examples/default$ make BOARD=nucleo-f767zi debug
    <>/RIOT/dist/tools/openocd/openocd.sh debug <>/RIOT/examples/default/bin/nucleo-f767zi/default.elf
    ### Starting Debugging ###
    Open On-Chip Debugger 0.10.0
    Licensed under GNU GPL v2
    For bug reports, read
            http://openocd.org/doc/doxygen/bugs.html
    User : 13 3 command.c:544 command_print(): debug_level: 3
    Debug: 14 4 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_log_output /tmp/openocd.log
    Debug: 15 4 command.c:143 script_debug(): command - log_output ocd_log_output /tmp/openocd.log
    Reading symbols from <>/RIOT/examples/default/bin/nucleo-f767zi/default.elf...
    Remote debugging using :3333
    warning: Overlapping regions in memory map: ignoring
    0x080011fa in idle_thread (arg=<optimised out>) at <>/RIOT/core/init.c:61
    61              pm_set_lowest();
    (gdb) c
    Continuing.
    ^CRemote connection closed
    (gdb)
    
to debug this get openocd dbg symbols and do ein parrallel

    sudo gdb
    [sudo] password : 
    GNU gdb (Ubuntu 8.3-0ubuntu1) 8.3
    Copyright (C) 2019 Free Software Foundation, Inc.
    License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
    This is free software: you are free to change and redistribute it.
    There is NO WARRANTY, to the extent permitted by law.
    Type "show copying" and "show warranty" for details.
    This GDB was configured as "x86_64-linux-gnu".
    Type "show configuration" for configuration details.
    For bug reporting instructions, please see:
    <http://www.gnu.org/software/gdb/bugs/>.
    Find the GDB manual and other documentation resources online at:
        <http://www.gnu.org/software/gdb/documentation/>.

    For help, type "help".
    Type "apropos word" to search for commands related to "word".
    (gdb) attach 32425
    Attaching to process 32425
    [New LWP 32426]
    [Thread debugging using libthread_db enabled]
    Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
    0x00007fac7e5e729b in __GI___select (nfds=nfds@entry=15, readfds=readfds@entry=0x7ffc75292420, writefds=writefds@entry=0x0, exceptfds=exceptfds@entry=0x0, timeout=timeout@entry=0x7ffc75292410) at ../sysdeps/unix/sysv/linux/select.c:41
    41      ../sysdeps/unix/sysv/linux/select.c: No such file or directory.
    (gdb) c
    Continuing.

    Thread 1 "openocd" received signal SIGSEGV, Segmentation fault.
    uCOS_III_update_thread_offsets (rtos=0x564ae6dc72e0) at src/rtos/uCOS-III.c:193
    193     src/rtos/uCOS-III.c: No such file or directory.
    (gdb) Quit
    (gdb) quit
    A debugging session is active.

            Inferior 1 [process 32425] will be detached.

    Quit anyway? (y or n) y
    Detaching from program: /usr/bin/openocd, process 32425
    [Inferior 1 (process 32425) detached]
    
#### Expected results

not dieing openocd while debuging 

#### Actual results

openocd dies

#### Versions

Ubuntu openocd-0.10.0-6
riot master

#4058

kfessel, 2020-02-05:
i just saw i forgot the bt of the open ocd crash:

    #0  uCOS_III_update_thread_offsets (rtos=0x55587d6c0100) at src/rtos/uCOS-III.c:193
    #1  uCOS_III_update_threads (rtos=0x55587d6c0100) at src/rtos/uCOS-III.c:320
    #2  0x000055587b70617f in rtos_update_threads (target=target@entry=0x55587d6cde30) at src/rtos/rtos.c:546
    #3  0x000055587b6fe1a0 in gdb_signal_reply (connection=0x55587d6cf940, target=0x55587d6cde30) at src/server/gdb_server.c:884
    #4  gdb_frontend_halted (target=0x55587d6cde30, connection=0x55587d6cf940) at src/server/gdb_server.c:884
    #5  0x000055587b6fe965 in gdb_frontend_halted (connection=<optimised out>, target=<optimised out>) at src/server/gdb_server.c:900
    #6  gdb_target_callback_event_handler (event=TARGET_EVENT_GDB_HALT, priv=<optimised out>, target=<optimised out>) at src/server/gdb_server.c:900
    #7  gdb_target_callback_event_handler (target=<optimised out>, event=<optimised out>, priv=<optimised out>) at src/server/gdb_server.c:888
    #8  0x000055587b6f316a in target_call_event_callbacks (target=0x55587d6cde30, event=event@entry=TARGET_EVENT_GDB_HALT) at src/target/target.c:1523
    #9  0x000055587b6f31cf in target_call_event_callbacks (target=target@entry=0x55587d6cde30, event=event@entry=TARGET_EVENT_HALTED) at src/target/target.c:1513
    #10 0x000055587b7a220d in adapter_poll (target=0x55587d6cde30) at src/target/hla_target.c:467
    #11 0x000055587b6f31fe in target_poll (target=0x55587d6cde30) at src/target/target.c:531
    #12 0x000055587b6f43d8 in handle_target (priv=<optimised out>) at src/target/target.c:2579
    #13 handle_target (priv=<optimised out>) at src/target/target.c:2502
    #14 0x000055587b6eba79 in target_call_timer_callback (now=0x7ffcc49eca80, cb=0x55587d6d9bb0) at src/target/target.c:1612
    #15 target_call_timer_callbacks_check_time (checktime=checktime@entry=1) at src/target/target.c:1612
    #16 0x000055587b6f02ae in target_call_timer_callbacks_check_time (checktime=1) at src/target/target.c:1623
    #17 target_call_timer_callbacks () at src/target/target.c:1623
    #18 0x000055587b702828 in server_loop (command_context=command_context@entry=0x55587d66d2a0) at src/server/server.c:444
    #19 0x000055587b6ad88b in openocd_thread (cmd_ctx=0x55587d66d2a0, argv=<optimised out>, argc=23) at src/openocd.c:304
    #20 openocd_main (argc=23, argv=<optimised out>) at src/openocd.c:341
    #21 0x00007f38618811e3 in __libc_start_main (main=0x55587b6ad130 <main>, argc=23, argv=0x7ffcc49eccd8, init=<optimised out>, fini=<optimised out>, rtld_fini=<optimised out>, stack_end=0x7ffcc49eccc8) at ../csu/libc-start.c:308
    #22 0x000055587b6ad23e in _start () at src/rtos/rtos.c:245

kfessel, 2020-02-05:
This is an openocd bug:
within the RTOS detection it may happen that there is no RTOS detected but there is still a structure allocated for it so the other RTOS function of openocd may not detect the missing RTOS fuctionality 

	} else if (strncmp(packet, "qSymbol", 7) == 0) {
		if (rtos_qsymbol(connection, packet, packet_size) == 1) {
			if (target->rtos_auto_detect == true) {
				target->rtos_auto_detect = false;
				target->rtos->type->create(target);
			}
			target->rtos->type->update_threads(target->rtos);
		}
		return ERROR_OK;
	} else

	} else if (strncmp(packet, "qSymbol", 7) == 0) {
		if (rtos_qsymbol(connection, packet, packet_size) == 1) {
			if (target->rtos_auto_detect == true) {
				target->rtos_auto_detect = false;
				target->rtos->type->create(target);
			}
			target->rtos->type->update_threads(target->rtos);
		}else
                      os_free(target);
		return ERROR_OK;
	} else

kfessel, 2020-02-05:
wrong button

benemorius, 2020-02-06:
There's a branch of openocd [here](https://github.com/beduino-project/openocd) that has RTOS support for Riot cooked in. It doesn't get updated all the time but it's a fairly recent build.

kfessel, 2020-02-06:
I just also did a forward-port  of the daniel-k RIOT for openocd patches

kfessel, 2020-02-21:
The openocd is availible here https://github.com/kfessel/openocd
i also made the thread priority visible

aabadie, 2020-07-08:
> The openocd with the forward ported daniel-k RIOT RTOS patches is available here

Any plans to provide this work upstream ?

kfessel, 2020-09-09:
@aabadie: atm i have no intention to get this upstream

But something happened there: http://openocd.zylin.com/#/c/4256/ seems like @benpicco got something into openocds current master i think he will have tested with current RIOT

maribu, 2023-05-19:
The latest stable OpenOCD works fine and there is little we can do on the RIOT side anyway, so IMO this can be closed.

closed 2023-05-19 by maribu