RIOT-OS/RIOT #13286: Porblem compile z1 under riot

jabberimaher, 2020-02-04:
hi,
please i need to compile mote z1 under riot but when i launch BOARD z1=make it display error arch32, i try to upgrade msp430-gcc-4.6.3 to 4.7.3 but it's failed..please i need help
best regards

jabberimaher, 2020-02-05:
hi,
please i need to compile mote z1 under riot but when i launch BOARD z1=make it display error arch32, i try to upgrade msp430-gcc-4.6.3 to 4.7.3 but it's failed..please i need help
**### ubuntu@ubuntu-Lenovo-G50-80:~/RIOT/examples/dtls-sock$ make BOARD=z1
There are unsatisfied feature requirements: arch_32bit
/home/ubuntu/RIOT/examples/dtls-sock/../../Makefile.include:829: *** You can let the build continue on expected errors by setting CONTINUE_ON_EXPECTED_ERRORS=1 to the command line.  Stop.**
BEST REGARDS

muhammadadeelinfo, 2020-02-06:
Hi @jabberimaher,
Did you try?

RIOT comes with a version of `goodfet.bsl`, a flashing tool for MSP430 based
platforms. Just link or copy it somewhere into your `$PATH` and you are good to go.
e.g.
```
Bash
$> cp /path/to/RIOT/dist/tools/goodfet/goodfet.bsl /usr/local/bin

fjmolinas, 2020-03-18:
Hi @jabberimaher the application you are trying to compile lists as the print shows a requirement `arch_32bit`, `z1` has `arch_16bit` so what the print is saying is that the application and in this case `dtls` does not support achritectures that are not `32bits`, therefore `examples/sock_dtls` can't work with `z1`

Here is a list of `BOARD`s supported by the application you are testing:

```
make -C examples/dtls-sock/ info-boards-supported --no-print-directory 
```

```
acd52832 airfy-beacon arduino-due arduino-mkr1000 arduino-mkrfox1200 arduino-mkrwan1300 arduino-mkrzero arduino-zero avsextrem b-l072z-lrwan1 b-l475e-iot01a blackpill blackpill-128kib bluepill bluepill-128kib calliope-mini cc1312-launchpad cc1352-launchpad cc2538dk cc2650-launchpad cc2650stk ek-lm4f120xl esp32-mh-et-live-minikit esp32-olimex-evb esp32-ttgo-t-beam esp32-wemos-lolin-d32-pro esp32-wroom-32 esp32-wrover-kit esp8266-esp-12x esp8266-olimex-mod esp8266-sparkfun-thing f4vi1 feather-m0 feather-nrf52840 firefly fox frdm-k22f frdm-k64f frdm-kw41z hamilton hifive1 hifive1b i-nucleo-lrwan1 ikea-tradfri iotlab-a8-m3 iotlab-m3 limifrog-v1 lobaro-lorabox lsn50 maple-mini mbed_lpc1768 mcb2388 microbit msba2 msbiot mulle native nrf51dk nrf51dongle nrf52832-mdk nrf52840-mdk nrf52840dk nrf52dk nrf6310 nucleo-f030r8 nucleo-f031k6 nucleo-f042k6 nucleo-f070rb nucleo-f072rb nucleo-f091rc nucleo-f103rb nucleo-f207zg nucleo-f302r8 nucleo-f303k8 nucleo-f303re nucleo-f303ze nucleo-f334r8 nucleo-f401re nucleo-f410rb nucleo-f411re nucleo-f412zg nucleo-f413zh nucleo-f429zi nucleo-f446re nucleo-f446ze nucleo-f722ze nucleo-f746zg nucleo-f767zi nucleo-l031k6 nucleo-l053r8 nucleo-l073rz nucleo-l152re nucleo-l432kc nucleo-l433rc nucleo-l452re nucleo-l476rg nucleo-l496zg nucleo-l4r5zi nz32-sc151 opencm904 openmote-b openmote-cc2538 p-l496g-cell02 particle-argon particle-boron particle-xenon pba-d-01-kw2x phynode-kw41z pic32-clicker pic32-wifire pinetime pyboard reel remote-pa remote-reva remote-revb ruuvitag samd21-xpro same54-xpro saml10-xpro saml11-xpro saml21-xpro samr21-xpro samr30-xpro samr34-xpro seeeduino_arch-pro sensebox_samd21 slstk3401a slstk3402a sltb001a slwstk6000b-slwrb4150a slwstk6000b-slwrb4162a slwstk6220a sodaq-autonomo sodaq-explorer sodaq-one sodaq-sara-aff spark-core stk3600 stk3700 stm32f030f4-demo stm32f0discovery stm32f3discovery stm32f429i-disc1 stm32f4discovery stm32f723e-disco stm32f769i-disco stm32l0538-disco stm32l476g-disco teensy31 thingy52 ublox-c030-u201 udoo usb-kw41z yunjia-nrf51822
```

jabberimaher, 2020-03-18:
thanks a lot for your reply...but the problem cooja simulator don't
support the mentionned list, can you help me to find a simulator
wich support esp32 for example and can display energy consumption an time
execution...
best regards

fjmolinas, 2020-03-19:
> thanks a lot for your reply...but the problem cooja simulator don't
> support the mentionned list, can you help me to find a simulator
> wich support esp32 for example and can display energy consumption an time
> execution...

Hi @jabberimaher I have no experience with simulators, you might want to address this question to our mailing list, it will get more visibility and there are more chances of someone having an answer for you.

users@riot-os.org (https://lists.riot-os.org/mailman/listinfo/users)

jabberimaher, 2020-04-01:
Bonjour,
Je viens de signer la pétition "كل مواطن تونسي: عريضة من أجل رفع قضية ضد
برهان بلخيرية وكل من سيسفر عنه البحث بتهمة الشروع في القتل" et je
souhaitais savoir si vous voudriez nous aider en ajoutant votre signature.

Notre objectif est d'atteindre 7 500 signatures et nous avons besoin de
plus de soutiens. Pour en savoir plus et pour signer, c'est ici:

http://chng.it/NmDJP9JjD6

Merci!
ماهر

fjmolinas, 2020-04-01:
Closing this issue, since `z1` (`16bit`) is currently not supported by `sock_dtls` so compilation failure is expected and is what shows up in the error messaege. @jabberimaher is you mean this as a feature request I would suggest opening a new issue.

closed 2020-04-01 by fjmolinas