RIOT-OS/RIOT #13321: Sleep mode for Arduino

catcombo, 2020-02-09:
#### Description

`xtimer_sleep` doesn't wake up an Arduino from sleep mode. I think I do something wrong but can't find out what excactly. I'm trying to migrate from Arduino framework to RIOT OS but I face the problems every time when I trying to use new feature/connect a device/etc :)

#### Steps to reproduce the issue

```
#include <xtimer.h>
#include <periph/pm.h>
#include <avr/sleep.h>

void pm_set_lowest(void)
{
    // Can be any SLEEP_MODE except SLEEP_MODE_IDLE
    set_sleep_mode(SLEEP_MODE_PWR_SAVE);
    sleep_mode();
}

int main(void) {
    while (true) {
        puts("Hi");
        xtimer_sleep(10);
    }

    return 0;
}
```

#### Expected results

Wake every 10 seconds.

#### Actual results

Hangs after first call to `sleep_mode()`.

#### Versions

Hardware: Arduino Nano.

roberthartung, 2020-02-11:
Welcome to RIOT @catcombo!

Using xtimer here is a bit tricky, as it is not 100% compatible / designed for being used in sleep modes.

ATmega328p uses the 16bit timer (Timer 1), as it is required for the resolution of xtimer. It uses the `clk_I/O` clock domain by default, unless we have an external clock source here? I am not sure of that though. So in case you use the internal clock from `clk_I/O`, you can only enter idle mode at this point, unless you either use a different timer (e.g. timer 2). In any case, you need a wake up source, e.g. an interrupt wo wake-up from sleep.

Checkout #8207 for my PR on sleep modes for the atmega CPUs.

At this point I am afraid to say, that you cannot use `xtimer` with sleep as it is right now.

catcombo, 2020-02-11:
@roberthartung Thank you very much for your explanations!

I thought it will be easier and more common way (thanks to RIOT) to use a sleep mode :\ I planned to create a thread and put it into a sleep every 5 minutes. If I understand correctly if I want to use deep sleep modes I need to use interrupts (or Watchdog timer? I saw some people do so in Arduino) to wake up after a certain amount of time. Is there any convenient way to use interrupts in RIOT rather than direct use of AVR libraries?

I also thought that in SLEEP_MODE_IDLE MCU will sleep until the end of xtimer interval but it seems it wake up multiple times a second. If I understand correctly there are other interrupts in the system that cause MCU to wake up and do some job. Is there a way to get the list of active interrupts to understand their purpose?

roberthartung, 2020-02-11:
@catcombo You're welcome. Using sleep modes in RIOT is in theory very easy (once the PR is merged that introduces 'native' sleep modes using pm_layered). However, using sleep modes in a fully stacked operating system is a big challenge. Therefore we created the `pm_layered` module, that automatically uses the deepest sleep mode available, which depends on the peripherals you use. By default you will only able to enter IDLE mode, as the uart is always enabled, as well as xtimer.
To be able to sleep, you need to disable uart and xtimer for that and enable a wake-up source. In my PR, this is realised by using the `RTT`.
You should *never* use avr libraries directly in your application code, but rather use RIOT's drivers and interfaces for it.

RIOT features the [irq interface](https://doc.riot-os.org/group__core__irq.html). You can use it right away to enable interfaces. Additionally, you can use the [GPIO module](https://doc.riot-os.org/group__drivers__periph__gpio.html) to enable external interrupts.
Timers can be used to wake you up, e.g. the [RTT ](https://doc.riot-os.org/group__drivers__periph__rtt.html)or [RTC ](https://doc.riot-os.org/group__drivers__periph__rtc.html) modules. In any case, they have to be configured/enabled, and would then wake up your CPU.

You are mixing avr code (set_sleep_mode) and RIOT code here. In theory this should be hidden from you! A list of interrupts is not available.

For clarification: You would like to do something and then wake up every five minutes for the same task, correctly?

catcombo, 2020-02-11:
Thanks for links to documentation and for explanation! It's all a bit hard for me now but I will try to figure out and understand how everything works to not bother you too much with a newbie questions :)

> For clarification: You would like to do something and then wake up every five minutes for the same task, correctly?

Yes, simple periodic task: get temperature from DS18 sensor, display it on HD44780 display, then sleep for 5 minutes to wake up and repeat the cycle.

maribu, 2023-05-19:
We now have ztimer that is compatible with low power modes. However, the AVR port has no `periph_pm` implementation yet. So there is no way actually use that with AVR boards.

(But given that modern 32 bit controllers are faster, cheaper, have more RAM/flash and require less power all at the same time, one might just go for one of those MCUs instead.

Anyways, I think the issue can be closed. A tracking issue for which platforms are still missing `periph_pm` would however be quite useful.

closed 2023-05-19 by maribu