RIOT-OS/RIOT #13330: suit: remove duplicate CoAP code

bergzand, 2020-02-10:
#### Description

From [`sys/include/suit/coap.h`](https://github.com/RIOT-OS/RIOT/blob/2e32a11822f8a344e11421e34d41433281bcc36b/sys/include/suit/coap.h#L57-L69):
```
/*                                                                               
 * Dear Reviewer,                                                                
 *                                                                               
 * At the time of PR'ing this code, there was a pile of CoAP PR's waiting for    
 * reviews.  Some of that functionality is needed in one way or another for      
 * SUIT. In order to not block software updates with CoAP refactoring, some of   
 * the work-in-progress code has been copied here.  We expect this to be         
 * removed as soon as CoAP in master provides similar functionality.             
 *                                                                               
 * As this is internal code that will go soon, I exclude this from Doxygen.      
 *                                                                               
 * Kaspar (July 2019)                                                            
 */                                                                              
```

@kaspar I think it is time we start to clean this code up :wink:.

 - [ ] CoAP subtree handling (refactor to prefix handling #11098)
 - [ ] CoAP blockwise request callback
 - [ ] CoAP blocksize defines

### Useful links

None

aabadie, 2020-07-02:
@bergzand any plan soon to address the 2 remaining points ? ;)

miri64, 2021-02-09:
I guess this is still an issue, right?