RIOT-OS/RIOT #13345: native: segmentation fault: in function xtimer_msg_received_timeout

jan-mo, 2020-02-11:
#### Description
Bug in xtimer_msg_received_timeout(). Tested on native and saml21-xpro.

#### Steps to reproduce the issue
Test if callback can send msg to waiting thread. 

#### Expected results
Thread is waiting with timeout on some callback (e.g. interrupt). Thread continues after msg_received or after timeout.

#### Actual results
After some iterations Segmentation fault on native. In saml21 output stops.

#### Versions
```Operating System Environment
-----------------------------
       Operating System: "Ubuntu" "19.10 (Eoan Ermine)"
                 Kernel: Linux 5.3.0-29-generic x86_64 x86_64

Installed compiler toolchains
-----------------------------
             native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
      arm-none-eabi-gcc: arm-none-eabi-gcc (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
                avr-gcc: missing
       mips-mti-elf-gcc: missing
             msp430-gcc: missing
   riscv-none-embed-gcc: missing
   xtensa-esp32-elf-gcc: missing
   xtensa-lx106-elf-gcc: missing
                  clang: missing

Installed compiler libs
-----------------------
   arm-none-eabi-newlib: "3.1.0"
    mips-mti-elf-newlib: missing
riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-lx106-elf-newlib: missing
               avr-libc: missing (missing)

Installed development tools
---------------------------
                  cmake: missing
               cppcheck: missing
                doxygen: missing
                 flake8: missing
                    git: git version 2.20.1
                   make: GNU Make 4.2.1
                openocd: Open On-Chip Debugger 0.10.0+dev-00973-g80f1a92b (2019-12-03-14:11)
                 python: Python 2.7.17
                python2: Python 2.7.17
                python3: Python 3.7.5
             coccinelle: missing
```

#### Code
```c
#include <stdio.h>
#include <string.h>

#include "xtimer.h"
#include "msg.h"
#include "thread.h"

#define DELAY       (100UL * US_PER_MS)
xtimer_t timer;

static void timer_event(void *arg) {
    kernel_pid_t *pid = (kernel_pid_t*)arg;
    msg_t msg = {
        .content.value = 1
    };
    puts("Interrupt");
    msg_send_int(&msg, *pid);
    puts("msg send");

    xtimer_set(&timer, DELAY-20);
}

int main(void)
{
    static msg_t rcv_queue[2];

    printf("You are running RIOT on a(n) %s board.\n", RIOT_BOARD);
    printf("This board features a(n) %s MCU.\n", RIOT_MCU);

    msg_init_queue(rcv_queue, 2);
    timer.callback = timer_event;
    kernel_pid_t pid = thread_getpid();
    timer.arg = (void*)&pid;
    
    xtimer_set(&timer, DELAY-20);

    while (1) {
        puts("wait");
    	xtimer_msg_receive_timeout(rcv_queue, DELAY);
    	puts("received");
    	xtimer_sleep(1);
    }
    return 0;
}
```

#### Console output
```
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2019.07-devel-745-gfc957)
You are running RIOT on a(n) native board.
This board features a(n) native MCU.
wait
Interrupt
msg send
received
Interrupt
msg send
Interrupt
msg send
Interrupt
msg send
Interrupt
msg send
Interrupt
msg send
Interrupt
msg send
Interrupt
msg send
Interrupt
msg send
Interrupt
msg send
wait
received
make: *** [/home/jan/Documents/GIT/RIOT_Jan/examples/hello-world/../../Makefile.include:561: term] Segmentation fault (core dumped)
```

benpicco, 2020-02-16:
Looks like the crash can be provoked by the `xtimer_sleep(1);`

```C
#include <stdio.h>
#include <thread.h>
#include <xtimer.h>

#ifndef DO_CRASH
#define DO_CRASH (1)
#endif

#define LOCK_TIMEOUT_US (100 * US_PER_MS)
#define TIMER_PERIOD_US (LOCK_TIMEOUT_US - 20)

static xtimer_t timer;

static void timer_event(void *arg) {
    kernel_pid_t *pid = (kernel_pid_t*)arg;

    msg_t msg = {
        .content.value = 1
    };

    xtimer_set(&timer, TIMER_PERIOD_US);
    msg_send_int(&msg, *pid);
}

int main(void)
{
    static msg_t rcv_queue[2];
    msg_init_queue(rcv_queue, ARRAY_SIZE(rcv_queue));

    kernel_pid_t pid = thread_getpid();

    timer.callback = timer_event;
    timer.arg = &pid;
    xtimer_set(&timer, TIMER_PERIOD_US);

    while (1) {
        msg_t m;
        if (xtimer_msg_receive_timeout(&m, LOCK_TIMEOUT_US) == 0) {
            puts("got msg");
        } else {
            puts("timeout");
        }

        if (DO_CRASH) {
            xtimer_sleep(1);
        }
    }

    return 0;
}
```

[`sys/xtimer/xtimer_core.c:251`](https://github.com/RIOT-OS/RIOT/blob/master/sys/xtimer/xtimer_core.c#L251)` => 0x5656032a`
`*** RIOT kernel panic:`
`FAILED ASSERTION.`

benpicco, 2020-02-16:
The mutex version doesn't crash though

```C
#include <stdio.h>
#include <mutex.h>
#include <xtimer.h>

#ifndef DO_CRASH
#define DO_CRASH (1)
#endif

#define LOCK_TIMEOUT_US (100 * US_PER_MS)
#define TIMER_PERIOD_US (LOCK_TIMEOUT_US - 20)

static xtimer_t timer;

static void timer_event(void *arg) {
    xtimer_set(&timer, TIMER_PERIOD_US);
    mutex_unlock(arg);
}

int main(void)
{
    mutex_t m = MUTEX_INIT_LOCKED;

    timer.callback = timer_event;
    timer.arg = &m;
    xtimer_set(&timer, TIMER_PERIOD_US);

    while (1) {
        if (xtimer_mutex_lock_timeout(&m, LOCK_TIMEOUT_US) == 0) {
            puts("lock");
        } else {
            puts("timeout");
        }

        if (DO_CRASH) {
            xtimer_sleep(1);
        }
    }

    return 0;
}
```

kaspar030, 2020-02-19:
My gut feeling would be that the self-resetting timer (xtimer_set() within the timer ISR) messes with some assumptions of the new xtimer code.

MichelRottleuthner, 2020-02-20:
This line of your example: 
>xtimer_msg_receive_timeout(rcv_queue, DELAY);

Do you really want to receive the message by writing it to the memory used for the message queue?

Rregarding the problem: I think it could be related to how `_xtimer_msg_receive_timeout` is implemented.
One problem I see is that the message meant as notification for "timeout while waiting for message" (i.e. [tmsg](https://github.com/RIOT-OS/RIOT/blob/master/sys/xtimer/xtimer.c#L166)) can end up in the message queue and is then delivered instantly the next time `_xtimer_msg_receive_timeout` is used, wrongly indicating a timeout when there actually was none.

@JulianHolzwarth you did fixes for a similar thread-version of that problem. How about this one? ;)

jan-mo, 2020-02-21:
The mutex_timeout version went well.
In my case it is not necessary to send a message but it could be useful. :)

MichelRottleuthner, 2020-02-21:
> In my case it is not necessary to send a message but it could be useful. :)

Yes, absolutely! And it is a bug (or at least a quality defect) that should be fixed. Many thanks for reporting it!

kaspar030, 2020-02-21:
> One problem I see is that the message meant as notification for "timeout while waiting for message" (i.e. [tmsg](https://github.com/RIOT-OS/RIOT/blob/master/sys/xtimer/xtimer.c#L166)) can end up in [...]

Ah, yes, and that in turn makes the logic think that the timer has triggered, and the xtimer_remove() is skipped, which then has a stack allocated, then invalid timer in xtimer's list. I remember now, I converted this to use thread flags a while ago. Let me dig it out.

kaspar030, 2020-02-21:
Quickfix: https://gist.github.com/7cb18562a16eddb62fee69b6378691a2

thread flag version: #13429.

JulianHolzwarth, 2020-02-21:
I don't know if this helps but this does not crash:

```c
static int _msg_wait(msg_t *m, msg_t *tmsg, xtimer_t *t)
{
    msg_receive(m);
    if (m->type == MSG_XTIMER && m->content.ptr == tmsg) {
        /* we hit the timeout */
        xtimer_remove(t);
        return -1;
    }
    else {
        xtimer_remove(t);
        return 1;
    }
}
```

MichelRottleuthner, 2020-02-26:
@JulianHolzwarth the above version still has the problem that the xtimer callback may have already added the timeout message to the queue. The thread flag version also solves that.

JulianHolzwarth, 2020-02-26:
@MichelRottleuthner I just wanted to show the reason for the crash. To make sure there is no  problem in xtimer_core. 
To clarify this was not a fix.

JulianHolzwarth, 2020-02-26:
@MichelRottleuthner I think that is a fix:
```
static void _callback_msg(void* arg)
{
    msg_t *msg = (msg_t*)arg;
    thread_t* t = (thread_t*) sched_threads[msg->sender_pid];
    if (t->status == STATUS_RECEIVE_BLOCKED)
    {
        msg_send_int(msg, msg->sender_pid);
    }
    
}
```
Because it should only send a message if the thread is waiting for a message.

MichelRottleuthner, 2020-02-26:
This could do. Waht you posted before (always remove the timer) is also mandatory then because the `STATUS_RECEIVE_BLOCKED` could become active again after returning from the `_xtimer_msg_receive_timeout` context.

JulianHolzwarth, 2020-02-26:
But the timer should not run when going out of `_xtimer_msg_receive_timeout` context. Because the callback does not put anything in the queue when it should not. Meaning we don't have to remove the timer (the timer is not in the timer list)

MichelRottleuthner, 2020-02-26:
What I meant was that the same thread may call `msg_receive` again directly after returning from `_xtimer_msg_receive_timeout` (setting `STATUS_RECEIVE_BLOCKED` again) so it must be ensured that the timer is removed before returning. Thinking about it again - that should already be the case though.

JulianHolzwarth, 2020-02-26:
@MichelRottleuthner 

> Thinking about it again - that should already be the case though.

I agree.

kaspar030, 2020-02-28:
How about we don't set the timer at all if there's already a message waiting?
Lemme sketch that.

kaspar030, 2020-02-28:
This looks like it should be done anyways (no need to set up a timer if there's a message waiting in any case): https://github.com/RIOT-OS/RIOT/pull/13504

Also fixes the issue at hand.

closed 2021-04-22 by kaspar030