RIOT-OS/RIOT #13358: iotlab-m3: long HW address is in reversed order

aabadie, 2020-02-13:
#### Description

The computed hardware address returned on the iotlab-m3 is in reversed order. They all finished with `11:15` but they should start with `15:11`. It is not possible to send unicast packets with the `txtsnd` shell command of `examples/default`.

#### Steps to reproduce the issue

1. Start an experiment on IoT-LAB with a couple of M3 devices (but you can also use more):
```
$ iotlab-experiment submit -m buggy_m3 -d 42 -l saclay,m3,1-2
$ iotlab-experiment wait
```
2. Buid `examples/default` for the `iotlab-m3`:
```
$ make BOARD=iotlab-m3 -C examples/default
```
3. Flash the firmware on the boards:
```
$ iotlab-node --update examples/default/bin/iotlab-m3/default.elf
```
4. Use the [iotlab websocket serial tool](https://github.com/iot-lab/ws-cli-tools) and run `ifconfig`:
```
$ iotlab-ws
Connected to m3-1.saclay
Connected to m3-2.saclay
ifconfig
m3-1.saclay: ifconfig
m3-1.saclay: Iface  4  HWaddr: 11:15  Channel: 26  Page: 0  NID: 0x23
m3-2.saclay: ifconfig
m3-1.saclay:           Long HWaddr: 32:8F:F7:65:10:6B:11:15 
m3-1.saclay:            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
m3-2.saclay: Iface  4  HWaddr: 11:15  Channel: 26  Page: 0  NID: 0x23
m3-2.saclay:           Long HWaddr: 36:BD:FD:65:10:6B:11:15 
m3-2.saclay:            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
m3-1.saclay:           AUTOACK  ACK_REQ  CSMA  L2-PDU:102 Source address length: 2
m3-1.saclay:           
m3-2.saclay:           AUTOACK  ACK_REQ  CSMA  L2-PDU:102 Source address length: 2
m3-2.saclay:           
>
```

As you can see in the output, both HWaddr are in reversed order and end with `11:15`.

in 2019.10, the 2 last bytes in the HW address are different and both HW address start with `15:11`. Sending unicast packets works.

I'm wondering if this is affecting all at86rf2xx radios (haven't tried on samr21-xpro) or maybe this is more low level ?

`git bitsect` found that the culprit commit is f0317c50f2d7f928f0523f901637738c6f6aaf8b from PR #12606.

cc'ing @maribu, @benpicco and @miri64 who were involved in #12606.

I think the bug is quite serious and I would also propose to backport it to 2020.01. @fjmolinas, what do you think ?

#### Expected results

The HW addr is in the right order.

#### Actual results

The HW addr is in reversed order

#### Versions

```
Operating System Environment
-----------------------------
         Operating System: "Ubuntu" "19.10 (Eoan Ermine)"
                   Kernel: Linux 5.3.0-29-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
                  avr-gcc: avr-gcc (AVR_8_bit_GNU_Toolchain_3.6.2_1759) 5.4.0
         mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2016.05-03 for MIPS MTI Bare Metal) 4.9.2
               msp430-gcc: missing
     riscv-none-embed-gcc: riscv-none-embed-gcc (GNU MCU Eclipse RISC-V Embedded GCC, 64-bit) 8.2.0
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 9.0.0-2 (tags/RELEASE_900/final)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: "2.1.0"
  riscv-none-embed-newlib: "3.0.0"
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.7.3
                    cmake: cmake version 3.13.4
                 cppcheck: Cppcheck 1.88
                  doxygen: 1.8.13
                      git: git version 2.20.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01047-g09ac9ab1 (2020-02-05-08:53)
                   python: Python 2.7.17
                  python2: Python 2.7.17
                  python3: Python 3.7.5
                   flake8: 3.7.7 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.7.5 on Linux
               coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

miri64, 2020-02-13:
@aabadie what do you mean with "reverse" order in terms of a quasi-random generated address? The bytes provided by `luid_get()` don't have any meaning attached to them, so speaking of an "order" seems very strange to me. When merging #12606 we were fully aware that link-layer addresses would change, as the goal was to increase the entropy in the set of available addresses.

miri64, 2020-02-13:
That the short addresses are the same is unfortunate indeed, but they are also picked arbitrarily to be the last two bytes of the long address? Are you talking about this?

aabadie, 2020-02-13:
> That the short addresses are the same is unfortunate

It is not unfortunate, it is broken. But yes this is the problem I'm pointed out. That's why I'm wondering if the issue is in the driver or at a lower level (cpuid ?).

benpicco, 2020-02-13:
Wait does the node receive packets when you send them to `11:15`, so the issue is 'just' that the address changed, or does it not receive packets sent to the short address anymore at all?

BTW: samr21-xpro comes with a proper EUI-64 on the board that is not used currently, #12641 will enable that (and change the address once again)

aabadie, 2020-02-13:
Since all nodes have the same short address, it's not possible to send an unicast packet to a single node using txtsnd.

I tried on several samr21-xpro: there are also duplicates short addresses but not all boards have the same:
```
samr21-2.saclay:           Long HWaddr: 46:E6:75:EC:4E:0C:65:79 
samr21-9.saclay:           Long HWaddr: 26:9A:F8:DE:59:09:63:79 
samr21-4.saclay:           Long HWaddr: 22:9E:46:3E:57:37:64:79 
samr21-8.saclay:           Long HWaddr: 2E:60:4D:64:5C:3C:65:79 
samr21-1.saclay:           Long HWaddr: 1A:30:3B:84:5D:00:64:79 
samr21-7.saclay:           Long HWaddr: 9A:CD:86:F3:7C:0F:67:79 
samr21-3.saclay:           Long HWaddr: CE:BC:1A:14:7F:04:67:79 
samr21-5.saclay:           Long HWaddr: 2A:62:E2:F7:72:08:67:79 
samr21-6.saclay:           Long HWaddr: 96:B9:AF:02:60:33:67:79 
samr21-10.saclay:           Long HWaddr: E6:3D:A0:0E:64:33:62:79
```
If the long HW address was in reversed order, all boards would get a different short address.

fjmolinas, 2020-02-13:
I agree a fix for this needs to be backported, and an a test for this needs to be added to the Release-Specs. I'll be away for the next two weeks so unable to deal with it until then though.

maribu, 2020-02-13:
@aabadie: There seems to be a misunderstanding in that the short address must be deduced from the long address in a specific way (or at all) and that deducing the short address from a unique long address is guaranteed to result in a unique short address. (That is basically what @miri64 already said above.)

> Each device in an IEEE 802.15.4 network can have two types of address:
> - IEEE (MAC) address: This is a 64-bit address, allocated by the IEEE, which uniquely identifies the device - no two devices in the world can have the same IEEE address. It is also sometimes called the extended address.
> - Short address: This 16-bit address identifies the node in the network and is local to that network (thus, two nodes on separate networks may have the same short address). The short address may be allocated by a Coordinator when a node joins a network.

So the non-standard way of self-assigning the short address used before by luck did not result in collisions before. The change here (which fixed a byte order confusion) now results that you are not so fortunate as before. This is clearly a pity, but not a bug.

benpicco, 2020-02-13:
#13362 changes the short address generation to `luid_get_short()` which will generate a more random result than just taking the last two bytes of the long address.

aabadie, 2020-02-13:
Since I found a workaround by using the long address, I'm not sure anymore if #13362 needs a backport to 2020.01-branch.

PR #13362 (2020-02-13): drivers/at86rf2xx: use luid_get_eui64() / luid_get_short()
### Contribution description

Use dedicated helper functions to generate long and short address.
`luid_get_short()` will use the 'next' pseudo-random luid instead of re-using the previous one, so this might alleviate the address collisions observed in #13358

### Testing procedure

- send and receive should still work using both the long and the short address (*tested locally with two `samr21-xpro`*)
- repeat the steps outlined in #13358

### Issues/PRs references

Maybe fixes #13358
commit 63f6d6aa (2020-02-13): drivers/at86rf2xx: use luid_get_eui64() / luid_get_short()

Use dedicated helper functions to generate long and short address.

Maybe fixes #13358
closed 2020-02-13 by aabadie