RIOT-OS/RIOT #13390: Cannot use LLVM with Cortex-M boards

nmeum, 2020-02-17:
#### Description

Though LLVM is declared a supported toolchain for Cortex-M boards:

https://github.com/RIOT-OS/RIOT/blob/2c09d9bd5bcfb9c35473a86986e26d91a91a6a55/cpu/cortexm_common/Makefile.include#L24

Unfortunately, it does not seem to work for me with `clang 9.0.1-8` and `BOARD=nucleo-f401re`.
#### Steps to reproduce the issue

Try to compile any application with `BOARD=nucleo-f401re` and `TOOLCHAIN=llvm`, e.g. `examples/hello-world`:

```
$ make TOOLCHAIN=llvm BOARD=nucleo-f401re -C examples/hello-world/
make: Entering directory '/home/nmeum/RIOT/examples/hello-world'
Building application "hello-world" for "nucleo-f401re" with MCU "stm32f4".

"make" -C /home/nmeum/RIOT/boards/nucleo-f401re
"make" -C /home/nmeum/RIOT/core
"make" -C /home/nmeum/RIOT/cpu/stm32f4
"make" -C /home/nmeum/RIOT/drivers
"make" -C /home/nmeum/RIOT/sys
"make" -C /home/nmeum/RIOT/boards/common/nucleo
"make" -C /home/nmeum/RIOT/sys/auto_init
"make" -C /home/nmeum/RIOT/drivers/periph_common
"make" -C /home/nmeum/RIOT/cpu/cortexm_common
"make" -C /home/nmeum/RIOT/sys/newlib_syscalls_default
"make" -C /home/nmeum/RIOT/cpu/stm32_common
"make" -C /home/nmeum/RIOT/sys/pm_layered
"make" -C /home/nmeum/RIOT/cpu/stm32f4/periph
"make" -C /home/nmeum/RIOT/sys/stdio_uart
"make" -C /home/nmeum/RIOT/cpu/cortexm_common/periph
"make" -C /home/nmeum/RIOT/cpu/stm32_common/periph
/home/nmeum/RIOT/cpu/cortexm_common/thread_arch.c:309:6: error: instruction requires: fp registers
    "vstmdbeq r0!, {s16-s31}          \n" /* save FPU registers if FPU is used */
     ^
<inline asm>:5:1: note: instantiated into assembly here
vstmdbeq r0!, {s16-s31}          
^
/home/nmeum/RIOT/cpu/cortexm_common/thread_arch.c:311:6: error: incorrect condition in IT block; got 'al', but expected 'eq'
    "stmdb  r0!,{r4-r11}              \n" /* save regs */
     ^
<inline asm>:6:6: note: instantiated into assembly here
stmdb  r0!,{r4-r11}              
     ^
/home/nmeum/RIOT/cpu/cortexm_common/thread_arch.c:360:6: error: instruction requires: fp registers
    "vldmiaeq r1!, {s16-s31}          \n" /* load FPU registers if saved */
     ^
<inline asm>:12:1: note: instantiated into assembly here
vldmiaeq r1!, {s16-s31}          
^
/home/nmeum/RIOT/cpu/cortexm_common/thread_arch.c:362:6: error: incorrect condition in IT block; got 'al', but expected 'eq'
    "msr    psp, r1                   \n" /* restore user mode SP to PSP reg */
     ^
<inline asm>:13:4: note: instantiated into assembly here
msr    psp, r1                   
   ^
4 errors generated.
make[3]: *** [/home/nmeum/RIOT/Makefile.base:110: /home/nmeum/RIOT/examples/hello-world/bin/nucleo-f401re/cortexm_common/thread_arch.o] Error 1
make[2]: *** [/home/nmeum/RIOT/Makefile.base:30: ALL--/home/nmeum/RIOT/cpu/cortexm_common] Error 2
make[1]: *** [/home/nmeum/RIOT/Makefile.base:30: ALL--/home/nmeum/RIOT/cpu/stm32f4] Error 2
make: *** [/home/nmeum/RIOT/examples/hello-world/../../Makefile.include:539: /home/nmeum/RIOT/examples/hello-world/bin/nucleo-f401re/application_hello-world.a] Error 2
make: Leaving directory '/home/nmeum/RIOT/examples/hello-world'
```

#### Expected results

The application should compile successfully. 

#### Actual results

The build fails.

kaspar030, 2020-02-17:
Works for me:

```
[kaspar@ng riot.tmp/examples/hello-world (master)]$ TOOLCHAIN=llvm BOARD=nucleo-f401re make clean all -j8
Building application "hello-world" for "nucleo-f401re" with MCU "stm32f4".

"make" -C /home/kaspar/src/riot.tmp/boards/nucleo-f401re
"make" -C /home/kaspar/src/riot.tmp/core
"make" -C /home/kaspar/src/riot.tmp/cpu/stm32f4
"make" -C /home/kaspar/src/riot.tmp/drivers
"make" -C /home/kaspar/src/riot.tmp/sys
"make" -C /home/kaspar/src/riot.tmp/drivers/periph_common
"make" -C /home/kaspar/src/riot.tmp/cpu/cortexm_common
"make" -C /home/kaspar/src/riot.tmp/cpu/stm32_common
"make" -C /home/kaspar/src/riot.tmp/cpu/stm32f4/periph
"make" -C /home/kaspar/src/riot.tmp/boards/common/nucleo
"make" -C /home/kaspar/src/riot.tmp/sys/auto_init
"make" -C /home/kaspar/src/riot.tmp/sys/newlib_syscalls_default
"make" -C /home/kaspar/src/riot.tmp/cpu/stm32_common/periph
"make" -C /home/kaspar/src/riot.tmp/cpu/cortexm_common/periph
"make" -C /home/kaspar/src/riot.tmp/sys/pm_layered
"make" -C /home/kaspar/src/riot.tmp/sys/stdio_uart
   text    data     bss     dec     hex filename
   9224     112    2580   11916    2e8c /home/kaspar/src/riot.tmp/examples/hello-world/bin/nucleo-f401re/hello-world.elf
```

```
$ clang --version
clang version 9.0.1 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

Also compiles with `BUILD_IN_DOCKER=1`.

Full versions: https://gist.github.com/a6c502dfd5db5b20b11e672300e7bf21

Can you post the output of `dist/tools/ci/print_toolchain_versions.sh`?

nmeum, 2020-02-17:
> Can you post the output of `dist/tools/ci/print_toolchain_versions.sh`?

https://gist.github.com/nmeum/b0f88c4a26a616845065c095f399b04d

kaspar030, 2020-02-17:
@nmeum does the application build in Docker, on your system (with "BUILD_IN_DOCKER=1 make ...")?

miri64, 2020-07-06:
Am able to reproduce this on my Ubuntu machine on current master (4807a91f70770acdf0dbccccc1b2e0bc96a4fba5)

<details>

```
BOARD=nucleo-f401re TOOLCHAIN=llvm make -C examples/hello-world/ -j16
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world'
Building application "hello-world" for "nucleo-f401re" with MCU "stm32".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/nucleo-f401re
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/core
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers/periph_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/auto_init
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/newlib_syscalls_default
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/pm_layered
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/stdio_uart
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/common/nucleo
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/cortexm_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32/bootloader
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/cortexm_common/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32/stmclk
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32/vectors
/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/cortexm_common/thread_arch.c:325:6: error: instruction requires: fp registers
    "vstmdbeq r0!, {s16-s31}          \n" /* save FPU registers if FPU is used */
     ^
<inline asm>:9:1: note: instantiated into assembly here
vstmdbeq r0!, {s16-s31}          
^
/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/cortexm_common/thread_arch.c:327:6: error: incorrect condition in IT block; got 'al', but expected 'eq'
    "stmdb  r0!,{r4-r11}              \n" /* save regs */
     ^
<inline asm>:10:6: note: instantiated into assembly here
stmdb  r0!,{r4-r11}              
     ^
/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/cortexm_common/thread_arch.c:368:6: error: instruction requires: fp registers
    "vldmiaeq r1!, {s16-s31}          \n" /* load FPU registers if saved */
     ^
<inline asm>:22:1: note: instantiated into assembly here
vldmiaeq r1!, {s16-s31}          
^
/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/cortexm_common/thread_arch.c:370:6: error: incorrect condition in IT block; got 'al', but expected 'eq'
    "msr    psp, r1                   \n" /* restore user mode SP to PSP reg */
     ^
<inline asm>:23:4: note: instantiated into assembly here
msr    psp, r1                   
   ^
4 errors generated.
make[3]: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base:108: /home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/bin/nucleo-f401re/cortexm_common/thread_arch.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base:30: ALL--/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/cortexm_common] Error 2
make[1]: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base:30: ALL--/home/mlenders/Repositories/RIOT-OS/RIOT/cpu/stm32] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/../../Makefile.include:575: /home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world/bin/nucleo-f401re/application_hello-world.a] Error 2
make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/hello-world'
```

</details>

<details><summary><code>dist/tools/ci/print_toolchain_versions.sh</code></summary>

```
Operating System Environment
----------------------------
         Operating System: "Ubuntu" "19.10 (Eoan Ermine)"
                   Kernel: Linux 5.3.0-62-generic x86_64 x86_64
             System shell: /usr/bin/dash (probably dash)
             make's shell: /usr/bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 9.2.1-9ubuntu2) 9.2.1 20191008
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2018-q4-major) 8.2.1 20181213 (release) [gcc-8-branch revision 267074]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: mips-mti-elf-gcc (Codescape GNU Tools 2018.09-03 for MIPS MTI Bare Metal) 6.3.0
               msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: clang version 9.0.0-2 (tags/RELEASE_900/final)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.0.0"
      mips-mti-elf-newlib: "2.5.0"
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.7.3
                    cmake: cmake version 3.13.4
                 cppcheck: Cppcheck 1.88
                  doxygen: 1.8.13
                      git: git version 2.20.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-00910-g4dbcb1e7 (2019-06-17-16:24)
                   python: Python 2.7.17
                  python2: Python 2.7.17
                  python3: Python 3.7.5
                   flake8: 3.7.8 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.7.5 on Linux
               coccinelle: spatch version 1.0.4 with Python support and with PCRE support
```

</details>

miri64, 2020-07-06:
(building with docker works however)

miri64, 2020-07-06:
Should we book this as a `State: won't fix` since this seems to be a toolchain error fixed in later versions?

sunch20, 2022-06-09:
I also meet the same problems with you, could you tell me how to deal with this problem finally?

maribu, 2023-05-19:
I just tested again and fundamentally, the LLVM toolchain in Alpine is able to build apps for Cortex-M boards.

There are many exceptions, though. We're working on getting that fixed and hopefully getting LLVM based CI testes back. But given that there is no fundamental / systematic issue, I'd say this can be closed.

closed 2023-05-19 by maribu