RIOT-OS/RIOT #13398: Test/pkg_fatfs doesn't run on stm32 F446RE

ChristopherPERDRIX, 2020-02-18:
Hello everyone I have a problem with the test "pkg_fatfs", I will try to summarize it as best I can.

#### Description
I want to write/read on a SDcard, so I use the test code on (tests/pkg_fatfs)

Some time ago, I had similar issue on Arduino mega2560
I got around the problem in the following way:
In /boards/arduino-mega2560/include/[board.h]( https://github.com/RIOT-OS/RIOT/blob/8468fe1c948027cebc98c25e484eb4d957efc983/boards/arduino-mega2560/include/board.h) I add:

```
#define SDCARD_SPI_PARAM_SPI      SPI_DEV(0)
#define SDCARD_SPI_PARAM_CS       GPIO_PIN(PORT_B,0)
#define SDCARD_SPI_PARAM_CLK      GPIO_PIN(PORT_B,1)
#define SDCARD_SPI_PARAM_MOSI     GPIO_PIN(PORT_B,2)
#define SDCARD_SPI_PARAM_MISO     GPIO_PIN(PORT_B,3)
```

Now I bought a STM32 F446RE, and I try to compile again this test and it doesn't run.
I watch on board.h for the F446RE board. SPI Port is declared:
```
static const spi_conf_t spi_config[] = {
    {
        .dev      = SPI1,
        .mosi_pin = GPIO_PIN(PORT_A, 7),
        .miso_pin = GPIO_PIN(PORT_A, 6),
        .sclk_pin = GPIO_PIN(PORT_A, 5),
        .cs_pin   = GPIO_PIN(PORT_A, 4),
        .mosi_af  = GPIO_AF5,
        .miso_af  = GPIO_AF5,
        .sclk_af  = GPIO_AF5,
        .cs_af    = GPIO_AF5,
        .rccmask  = RCC_APB2ENR_SPI1EN,
        .apbbus   = APB2
    },
```

I plugged my cable in the following manner:

-  mosi --> A7
-  miso --> A6
-  Sclk --> A5
-  CS --> A4

But on pyterm, I haven't got anything.
I think I don't use the good pin for F446RE or maybe the test doesn't use the same port.

#### Steps to reproduce the issue

1. Plug your board with a sdcard (SPI )
2. cd RIOT/tests/pkg_fatfs
3. make BOARD=<Board> flash term (I use nucleo-f446re)

#### Expected results
The good result with Arduino mega2560 with editing the file (board.h)

> Building application "tests_pkg_fatfs" for "arduino-mega2560" with MCU "atmega2560".
> ....
> 2020-02-18 10:43:16,137 # Connect to serial port /dev/tty.usbmodem14101
> Welcome to pyterm!
> Type '/exit' to exit.
> 2020-02-18 10:43:17,145 # 
> 
> 2020-02-18 10:43:17,147 # main(): This is RIOT! (Version: 2020.04-devel-524-g9866)
> 2020-02-18 10:43:17,160 # init sdcard_mtd 0 [OK]
> > /exit

#### Actual results
Result on stm32 F446RE:

> Building application "tests_pkg_fatfs" for "nucleo-f446re" with MCU "stm32f4".
> ...
> 2020-02-18 10:48:14,232 # Connect to serial port /dev/tty.usbmodem14103
> Welcome to pyterm!
> Type '/exit' to exit.
> 2020-02-18 10:54:37,691 # [auto_init_storage] error initializing sdcard_spi #0
> 2020-02-18 10:54:37,696 # main(): This is RIOT! (Version: 2020.04-devel-524-g9866)
> 2020-02-18 10:54:37,698 # Initializing the RTC driver
> 2020-02-18 10:54:37,701 # Setting RTC to 2000-02-01 00:00:00
> /exit

I have the same result when I test an I2C driver and I'm not connected with it.

Maybe I am not using the right SPI port (I also tried the second SPI port). Or is the program using a default spi port which prevents me from using that of my card?

#### Versions
Operating system: Mac OSX Catalina : 10.15.3
Build environment: 

> GNU Make 4.3
> Built for x86_64-apple-darwin19.3.0

MichelRottleuthner, 2020-02-20:
Thanks for taking the time providing all the detailed information for opening an issue, very much appreciated.
So to me it looks like you wired it up correctly (given your information above is right), but you didn't tell the `sdcard_spi` driver which pins it should use.
Did you check the configuration in `drivers/sdcard_spi/include/sdcard_spi_params.h`? 
Try changing the pin configuration (i.e. `SDCARD_SPI_PARAM_CS`, `SDCARD_SPI_PARAM_CLK`,
`SDCARD_SPI_PARAM_MOSI`, and `SDCARD_SPI_PARAM_MISO` according to your board configuration that you already posted above. 
You can also overwrite these settings from your application Makefile if you add the following:
```
CFLAGS += -DSDCARD_SPI_PARAM_CS=GPIO_PIN\(0,4\)
CFLAGS += -DSDCARD_SPI_PARAM_CLK=GPIO_PIN\(0,5\)
CFLAGS += -DSDCARD_SPI_PARAM_MOSI=GPIO_PIN\(0,7\)
CFLAGS += -DSDCARD_SPI_PARAM_MISO=GPIO_PIN\(0,6\)
```

Let me know if you need any further help.

ChristopherPERDRIX, 2020-02-21:
Thank you for your answer.

I think the port SPI 1 doesn't work or I don't use it well.
When I am plugging on SPI 1, the LED 2 is green, but i don't know why. (config Arduino? )
The result is always the same: 

> 2020-02-21 08:14:04,096 - INFO # �[auto_init_storage] initializing sdcard_spi #0
> 2020-02-21 08:20:30,139 - INFO # [auto_init_storage] error initializing sdcard_spi #0
> 2020-02-21 08:20:30,144 - INFO # main(): This is RIOT! (Version: 2019.07-devel-314-gc439)
> 2020-02-21 08:20:30,147 - INFO # SD-card spi driver test application
> 2020-02-21 08:20:30,152 - INFO # insert SD-card and use 'init' command to set card to spi mode
> 2020-02-21 08:20:30,159 - INFO # WARNING: using 'write' or 'copy' commands WILL overwrite data on your sd-card and
> 2020-02-21 08:20:30,166 - INFO # almost for sure corrupt existing filesystems, partitions and contained data!

I try on port SPI 2 and it's ok: (LED 2 is off)
```
CFLAGS += -DSDCARD_SPI_PARAM_CS=GPIO_PIN\(1,12\)
CFLAGS += -DSDCARD_SPI_PARAM_CLK=GPIO_PIN\(1,13\)
CFLAGS += -DSDCARD_SPI_PARAM_MOSI=GPIO_PIN\(1,15\)
CFLAGS += -DSDCARD_SPI_PARAM_MISO=GPIO_PIN\(1,14\)
```
And the result:

> 2020-02-21 09:01:17,904 - INFO # �[auto_init_storage] initializing sdcard_spi #0
> 2020-02-21 09:01:17,922 - INFO # main(): This is RIOT! (Version: 2019.07-devel-314-gc439)
> 2020-02-21 09:01:17,932 - INFO # SD-card spi driver test application
> 2020-02-21 09:01:17,933 - INFO # insert SD-card and use 'init' command to set card to spi mode
> 2020-02-21 09:01:17,938 - INFO # WARNING: using 'write' or 'copy' commands WILL overwrite data on your sd-card and
> 2020-02-21 09:01:17,944 - INFO # almost for sure corrupt existing filesystems, partitions and contained data!
> > 

I don't understand why port SPI 1 doesn't work, but i will use SPI 2 for now.
Thank you so much!

MichelRottleuthner, 2020-02-21:
Could be that one of the SPI pins is connected to something else (the LED?) via a solder bridge. The [Nucleo-64 User Manual](https://www.st.com/content/ccc/resource/technical/document/user_manual/98/2e/fa/4b/e0/82/43/b7/DM00105823.pdf/files/DM00105823.pdf/jcr:content/translations/en.DM00105823.pdf) would be the right place to look for that. 
Anyway, out of curiosity, may I ask what you are using the sdcard / filesystem for? I wanted to do some improvements of these modules at some point and it would be nice to collect some feedback on how people use it.

ChristopherPERDRIX, 2020-02-21:
Yes I think it can be the reason. ( I think PA5 (or D13 on Arduino) is connected to the LED)

I need to write down the data of the temperature and times on an SD card.
Later, I will also want to read it on the card to initialize my board

I'm a beginner, so I need time to understand the code. But after a few hours, I managed to adapt the code to my application.

MichelRottleuthner, 2020-02-21:
Ok, then I wish happy temperature sensing and data collection! I'll close the Issue as it is a configuration problem and not a bug in one of the modules.

closed 2020-02-21 by MichelRottleuthner