RIOT-OS/RIOT #13406: cpu/saml21: How to implement proper power saving?

jue89, 2020-02-19:
#### Description

I am current working on a project that utilizes the SAMR30 CPU. Since it is a battery-powered device, I need to get the CPU and peripheral to use as little energy as possible.

When using RIOT master branch, I unblock the power state *STANDBY* in the layered power management. This brings the board (I use the samr30-xpro board) down to 650uA. Too much for the device.

So I digged into the relevant `cpu` folders and made two little modifications. All in all I changed two things:

1. Run the *External Interrupt Controller (EIC)* with the *CLK_ULP32K* (the 32kHz clock) instead of the *GCLK* (the 16MHz clock).
2. Turn off *GLCK* if it is not required by the CPU or any peripheral.

This brings the power consumption down to 2uA! In this state the RTC is still running, the RAM content is retained and external interrupts can also be caught.

I would like to contribute this little change to RIOT but I don't know how to implement this without affecting other projects. I thought of some ifdefs here and there. Does anyone has a good idea how to implement this? I would like to come up with a PR. I thought about a CFLAG (or Kconfig in the future) that prepares the CPU for low-power operation.

For those interested:

```diff
diff --git a/cpu/sam0_common/periph/gpio.c b/cpu/sam0_common/periph/gpio.c
index a5f76d946..7991f03a9 100644
--- a/cpu/sam0_common/periph/gpio.c
+++ b/cpu/sam0_common/periph/gpio.c
@@ -220,7 +220,7 @@ int gpio_init_int(gpio_t pin, gpio_mode_t mode, gpio_flank_t flank,
     while (_EIC->STATUS.reg & EIC_STATUS_SYNCBUSY) {}
 #else /* CPU_FAM_SAML21 */
     /* enable the EIC module*/
-    _EIC->CTRLA.reg = EIC_CTRLA_ENABLE;
+    _EIC->CTRLA.reg = EIC_CTRLA_ENABLE | EIC_CTRLA_CKSEL;
     while (_EIC->SYNCBUSY.reg & EIC_SYNCBUSY_ENABLE) {}
 #endif
     return 0;
diff --git a/cpu/saml21/cpu.c b/cpu/saml21/cpu.c
index 65783ae97..03e3b1dd1 100644
--- a/cpu/saml21/cpu.c
+++ b/cpu/saml21/cpu.c
@@ -119,7 +119,7 @@ void cpu_init(void)
 
     /* set OSC16M to 16MHz */
     OSCCTRL->OSC16MCTRL.bit.FSEL = 3;
-    OSCCTRL->OSC16MCTRL.bit.ONDEMAND = 0;
+    OSCCTRL->OSC16MCTRL.bit.ONDEMAND = 1;
     OSCCTRL->OSC16MCTRL.bit.RUNSTDBY = 0;
 
     _osc32k_setup();
```

### Useful links

https://www.mouser.com/datasheet/2/268/70005303b-1108106.pdf

benpicco, 2020-02-19:
Setting the `ONDEMAND` bit should have no ill effects AFAIK. Maybe @dylad can correct me here, but I think we should just set that bit.

Running the EIC *always* at 32kHz can cause issues with if you get interrupts faster than that.
`samd21` does it and I lose interrupts with the `at86rf215` driver unless I run EIC off the main clock.

*However* and this is something on my TODO list at least for samd5x, we can switch the clock source when we enter power save mode.
This is actually *needed* to wake up from deep sleep through EXTI (currently not possible on samd5x and probably saml21 too as it always uses the main clock for the interrupt controller).

So my suggestion for that would be to switch to the slower clock in `periph/pm.c` for `SLEEPMODE_BACKUP` and `SLEEPMODE_STANDBY` and switch back to the fast clock after wake-up.
I'd be happy to pull that (and then do the same for the other SAM0s).

As for longer sleep, I just set an RTC alarm and then call `pm_set(0)` directly.

jue89, 2020-02-19:
> Running the EIC _always_ at 32kHz can cause issues with if you get interrupts faster than that.
> `samd21` does it and I lose interrupts with the `at86rf215` driver unless I run EIC off the main clock.

Good to know! Thank you.

> So my suggestion for that would be to switch to the slower clock in `periph/pm.c` for `SLEEPMODE_BACKUP` and `SLEEPMODE_STANDBY` and switch back to the fast clock after wake-up.
> I'd be happy to pull that (and then do the same for the other SAM0s).
> 
> As for longer sleep, I just set an RTC alarm and then call `pm_set(0)` directly (and accept the MCU reset on wake-up).

I like the idea! AFAIK the BACKUP mode switches the EIC off - so I think it should be sufficient add a few lines to `cpu/saml21/periph/pm.c` and switch the EIC clock to CLK_ULP32K before calling `coretexm_sleep(0);` and switch back to GLCK after the sleep call returned.

I'm going to come up with a PR implementing this.

jue89, 2020-03-02:
I think aspect I addressed in this issue has been implemented and merged by follow-up PRs. Thank you all for your great support!

closed 2020-03-02 by jue89