RIOT-OS/RIOT #13408: Bug: PWM test crashes on Arduino Mega 2560

kmikaz51, 2020-02-19:
#### Description
The test periph_pwm crashes on Arduino Mega 2560 using the command oscillate.

#### Steps to reproduce the issue
```
$ cd RIOT/tests/periph_pwm
$ make BOARD=arduino-mega2560
$ make flash BOARD=arduino-mega2560
$ make term BOARD=arduino-mega2560
>osci
```

#### Expected results
After reading the code, the test should make each usable PWM oscillate for a time with these messages:

```
RIOT PWM test
Connect an LED or scope to PWM pins to see something.

Available PWM device between 0 and n
Initialized PWM_i @ freq Hz.
...
Initialized PWM_n @ freq Hz.
Letting the PWM pins oscillate now...
```

#### Actual results
```
RIOT PWM test
Connect an LED or scope to PWM pins to see something.

Available PWM device between 0 and 1
Stack Pointer: 0x0a5c
*** RIOT kernel panic:
FAILED ASSERTION.

*** halted.
```

#### Versions
```
Operating System Environment
-----------------------------
         Operating System: "Ubuntu" "18.04.4 LTS (Bionic Beaver)"
                   Kernel: Linux 4.15.0-76-generic x86_64 x86_64
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: cmake version 3.10.2
                 cppcheck: missing
                  doxygen: missing
                      git: git version 2.17.1
                     make: GNU Make 4.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01058-g853a0528 (2020-02-11-10:48)
                   python: Python 2.7.17
                  python2: Python 2.7.17
                  python3: Python 3.6.9
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

maribu, 2020-02-29:
This is actually not really a bug, more a missing feature: The PWM code currently only implements `PWM_LEFT` as alignment, and not `PWM_CENTER` or `PWM_RIGHT`. It will throw an assertion when a mode is requested that is not currently supported.

maribu, 2021-02-13:
There is still no `PWM_CENTER` or `PWM_RIGHT` implemented. Nor did we extend the feature system to allow applications (and also the test with the blowing assertion) to test for this. The best would likely be to provide both. Both is not too difficult, but requires some spare time. I put this on my TODO list, but I don't think I'll find time for this soonish.

maribu, 2022-03-02:
Oh, yes. That is sadly still on my todo list. (But if anyone else wants to fix it, please don't wait for me.)