RIOT-OS/RIOT #13419: Can't use nimble as prefix for modules

HendrikVE, 2020-02-20:
#### Description
It's not possible to use the prefix `nimble_` for custom modules in `sys/`

#### Steps to reproduce the issue
Add a new dummy module with the name `nimble_prefix_test` and add `USEMODULE += nimble_prefix_test` to the Makefile, e.g. of the `default` example. 
#### Expected results
It should be build. There should be no technical restriction for choosing module names, except duplicates.

#### Actual results
Make will fail with the following message:
```
make[1]: *** No rule to make target 'nimble_test', needed by 'all'. Stop.
make: *** [/home/hendrik/RIOT/examples/default/../../Makefile.include:639: pkg-build-nimble] Error 2
```
But if you rename the module to `my_nimble_prefix_test` everything works as expected.

kaspar030, 2020-02-20:
Yeah, nimble takes that prefix. Not sure where exactly.

Can you try adding "nimble_test" here:

https://github.com/RIOT-OS/RIOT/blob/ea3296df62d7a54b08dcc5292bbbe53ce6b7a4bc/pkg/nimble/Makefile#L26

HendrikVE, 2020-02-20:
Thank you very much, this fixes the issue. But it feels quite hacky. Is it intended that a whole prefix is "reserved" for a specific purpose?

miri64, 2020-07-06:
> […] this fixes the issue.

I take this as a reason for this issue to be closed.

closed 2020-07-06 by miri64