RIOT-OS/RIOT #13434: drivers/bmx280: CI should fail, but apparently doesn't?

wosym, 2020-02-21:
#### Description
When I compile tests/driver_bmx280, I get several compiler errors notifying me of arrays going out of bounds in the _read_calibration_data function in drivers/bmx280/bmx280.c
When I check this file, as far as I can tell, these errors are justified:
BMX280_DIG_P9_LSB_REG = 0x9E
OFFSET_TP(x) is (x - CALIB_T_P_BASE)
CALIB_T_P_BASE = BMX280_DIG_T1_LSB_REG = 0x88
--> maximum offset = 0x9E - 0x88 = 0x16 = 0d22

The buffer which gets read has a size of CALIB_T_P_LEN, which is defined as 17. 
17 > 22, so we definitely read out of the defined array size. (Even more so, because the array is read by offset+1.

So the errors my compiler spews out seem justified to me. However... The CI does not appear to complain about this. One might say that this is a CI issue, but... When I mentioned this issue on IRC, @benpicco said he did not experience this behavior when he tried to compile the exact same program.

Maybe this is because of a gcc version mismatch? In any case, to me it feels like it's important that the compiler (or CI, because we don't have control over what compiler people use) at least warns us of these types of errors. 
 
#### Steps to reproduce the issue
compile /tests/driver_bmx280 with the default board.

#### Expected results
The CI should notify us of the errors.

#### Actual results
On my machine? Exactly what I would expect.
On the CI (and some other machines?): no errors.

#### Versions
```
Operating System Environment
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 5.4.3-arch1-1 x86_64 unknown
             System shell: GNU bash, version 5.0.11(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.11(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (GCC) 9.2.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (GCC) 9.2.0
         mips-mti-elf-gcc: missing
               msp430-gcc: missing
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
   xtensa-esp8266-elf-gcc: missing
                    clang: missing

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
  riscv-none-embed-newlib: missing
  xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
                 avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: missing
                    cmake: missing
                 cppcheck: missing
                  doxygen: 1.8.16
                      git: git version 2.24.1
                     make: GNU Make 4.2.1
                  openocd: Open On-Chip Debugger 0.10.0
                   python: Python 3.8.0
                  python2: Python 2.7.17
                  python3: Python 3.8.0
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing
```

wosym, 2020-02-21:
For reference, these are the errors I get:
```
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c: In function 'bmx280_init':
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:131:29: error: array subscript 17 is outside array bounds of 'uint8_t[17]' {aka 'unsigned char[17]'} [-Werror=array-bounds]
  131 |     return (((int16_t)buffer[offset + 1]) << 8) | buffer[offset];
      |                       ~~~~~~^~~~~~~~~~~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:143:13: note: while referencing 'buf'
  143 |     uint8_t buf[CALIB_T_P_LEN];
      |             ^~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:131:29: error: array subscript 19 is outside array bounds of 'uint8_t[17]' {aka 'unsigned char[17]'} [-Werror=array-bounds]
  131 |     return (((int16_t)buffer[offset + 1]) << 8) | buffer[offset];
      |                       ~~~~~~^~~~~~~~~~~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:143:13: note: while referencing 'buf'
  143 |     uint8_t buf[CALIB_T_P_LEN];
      |             ^~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:131:57: error: array subscript 18 is outside array bounds of 'uint8_t[17]' {aka 'unsigned char[17]'} [-Werror=array-bounds]
  131 |     return (((int16_t)buffer[offset + 1]) << 8) | buffer[offset];
      |                                                   ~~~~~~^~~~~~~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:143:13: note: while referencing 'buf'
  143 |     uint8_t buf[CALIB_T_P_LEN];
      |             ^~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:131:29: error: array subscript 21 is outside array bounds of 'uint8_t[17]' {aka 'unsigned char[17]'} [-Werror=array-bounds]
  131 |     return (((int16_t)buffer[offset + 1]) << 8) | buffer[offset];
      |                       ~~~~~~^~~~~~~~~~~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:143:13: note: while referencing 'buf'
  143 |     uint8_t buf[CALIB_T_P_LEN];
      |             ^~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:131:57: error: array subscript 20 is outside array bounds of 'uint8_t[17]' {aka 'unsigned char[17]'} [-Werror=array-bounds]
  131 |     return (((int16_t)buffer[offset + 1]) << 8) | buffer[offset];
      |                                                   ~~~~~~^~~~~~~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:143:13: note: while referencing 'buf'
  143 |     uint8_t buf[CALIB_T_P_LEN];
      |             ^~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:131:29: error: array subscript 23 is outside array bounds of 'uint8_t[17]' {aka 'unsigned char[17]'} [-Werror=array-bounds]
  131 |     return (((int16_t)buffer[offset + 1]) << 8) | buffer[offset];
      |                       ~~~~~~^~~~~~~~~~~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:143:13: note: while referencing 'buf'
  143 |     uint8_t buf[CALIB_T_P_LEN];
      |             ^~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:131:57: error: array subscript 22 is outside array bounds of 'uint8_t[17]' {aka 'unsigned char[17]'} [-Werror=array-bounds]
  131 |     return (((int16_t)buffer[offset + 1]) << 8) | buffer[offset];
      |                                                   ~~~~~~^~~~~~~~
/home/wsymons/repo/RIOT/drivers/bmx280/bmx280.c:143:13: note: while referencing 'buf'
  143 |     uint8_t buf[CALIB_T_P_LEN];
      |             ^~~
cc1: all warnings being treated as errors
```

benpicco, 2020-02-21:
Hm looks like you are using a newer toolchain and the one used by Murdock didn't catch these errors.
Then https://github.com/RIOT-OS/riotdocker/pull/78 should also trigger those errors in CI.

PR #13456 (2020-02-24): cpu/arm: fixes necessary for toolchain upgrade

### Contribution description

The ARM toolchain upgrade (https://github.com/RIOT-OS/riotdocker/pull/78) slipped through some new compiler warnings (which turn into errors now).

This PR tries to fix them.

### Testing procedure

### Issues/PRs references

Fixes #13434.
PR #13536 (2020-03-03): cpu/arm: fixes necessary for toolchain upgrade [backport 2020.01]
# Backport of #13456

### Contribution description

The ARM toolchain upgrade (https://github.com/RIOT-OS/riotdocker/pull/78) slipped through some new compiler warnings (which turn into errors now).

This PR tries to fix them.

### Testing procedure

### Issues/PRs references

Fixes #13434.
closed 2020-02-24 by kaspar030