RIOT-OS/RIOT #13442: gnrc_networking not sending packets when hardware has multiple interfaces

Nishchay-sopho, 2020-02-21:
#### Description
I am working under @roberthartung at IBR on riot as a part of my thesis and encountered this bug.

While trying to run example/gnrc_networking, I get the error :
`
ipv6: no link-layer address or interface for next hop to 
`

This error comes up when trying to send a packet with the nucleo-l476rg board and at86rf215 as the radio. This radio has 2 network interfaces (one is a subGhz interface and one 2.4Ghz interface).

This is happening because the udp.c file in examples/gnrc_networking isn't designed to extract the interface (through which the data has to be sent) from the ip address when there are multiple interfaces. 
For reference, this is the code(in udp.c of gnrc_networking example) which should be able to extract the interface:

```
iface = ipv6_addr_split_iface(addr_str);
    if ((!iface) && (gnrc_netif_numof() == 1)) {
        netif = gnrc_netif_iter(NULL);
    }
    else if (iface) {
        netif = gnrc_netif_get_by_pid(atoi(iface));
    }

```

@haukepetersen @miri64  Could you please look into this?

I suggest allowing ipv6 address to be assigned as `<ip-address>%<interface-number>` as a quick fix. This would require only the ` ipv6_addr_from_str(&addr, addr_str); ` method to add the functionality of handling ip address in the above specified format.

#### Steps to reproduce the issue
Use the RIOT code from the PR #12128 with at86rf215 (which has 2 network interfaces).
Compile and flash the examples/gnrc_networking onto the boards. 

Try and send the packet to another node in the network. The shell will show that the packet was successfully sent but the packet isn't actually sent which can be confirmed by enabling debug in the gnrc_ipv6.c file and seeing the debug statement (as mentioned above): no link layer address found.

#### Expected results

The code should be able to extract the interface along which the data is to be sent and send the data accordingly.

#### Actual results

The iface variable in the send function of udp.c is empty which causes the network interface to remain NULL and finally the data not sent.

#### Versions
```

Operating System Environment
-----------------------------
         Operating System: "Ubuntu" "18.04.4 LTS (Bionic Beaver)"
                   Kernel: Linux 4.15.0-88-generic x86_64 x86_64
             System shell: /bin/dash (probably dash)
             make's shell: /bin/dash (probably dash)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Ubuntu 7.4.0-1ubuntu1~18.04.1) 7.4.0
        arm-none-eabi-gcc: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
                  avr-gcc: avr-gcc (GCC) 5.4.0
         mips-mti-elf-gcc: missing
               msp430-gcc: msp430-gcc (GCC) 4.6.3 20120301 (mspgcc LTS 20120406 unpatched)
     riscv-none-embed-gcc: missing
     xtensa-esp32-elf-gcc: missing
     xtensa-esp8266-elf-gcc: missing
                    clang: clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)

Installed compiler libs
-----------------------
     arm-none-eabi-newlib: "3.1.0"
      mips-mti-elf-newlib: missing
     riscv-none-embed-newlib: missing
xtensa-esp32-elf-newlib: missing
xtensa-esp8266-elf-newlib: missing
avr-libc: "2.0.0" ("20150208")

Installed development tools
---------------------------
                   ccache: ccache version 3.4.1
                    cmake: cmake version 3.10.2
                 cppcheck: missing
                  doxygen: 1.8.13
                      git: git version 2.17.1
                     make: GNU Make 4.1
                  openocd: Open On-Chip Debugger 0.10.0+dev-01000-gdb23c13d (2020-01-06-15:02)
                   python: Python 2.7.17
                  python2: Python 2.7.17
                  python3: Python 3.6.9
                   flake8: error: /usr/bin/python3: No module named flake8
               coccinelle: missing

```

miri64, 2020-02-21:
Will have a look into it after the weekend!

miri64, 2020-02-24:
> I suggest allowing ipv6 address to be assigned as `<ip-address>%<interface-number>` as a quick fix. This would require only the `ipv6_addr_from_str(&addr, addr_str);` method to add the functionality of handling ip address in the above specified format.

This is exactly what *should* be happening in the lines you quoted... Not sure what is going wrong. I will try to find some radios to test this.

roberthartung, 2020-02-24:
> This is exactly what _should_ be happening in the lines you quoted... Not sure what is going wrong. I will try to find some radios to test this.

What we have seen is that there is no code to parse the interface from it. `udp.c` from the example does correctly extract the netif, but `ipv6_addr_from_str` does not handle the `%` at the end and fails parsing.

miri64, 2020-02-24:
Ah, but that's why you call `ipv6_addr_split_iface()` before `ipv6_addr_from_str()`. Adding the functionality to `ipv6_addr_from_str()` wouldn't be a quick fix actually, as it would require additional return values from this function.

roberthartung, 2020-02-24:
> Ah, but that's why you call `ipv6_addr_split_iface()` before `ipv6_addr_from_str()`. Adding the functionality to `ipv6_addr_from_str()` wouldn't be a quick fix actually, as it would require additional return values from this function.

Then it still doesn't work in `examples/gnrc_networking/`. `udp.c` *does* call `ipv6_addr_split_iface()` before `ipv6_addr_from_str()` and it fails parsing the address.

miri64, 2020-02-24:
I saw that. But I'm still trying to understand, why the failure is happening.

roberthartung, 2020-02-24:
@Nishchay-sopho can you please check again and see if the pointer is changed correctly? You said that you tried specifying the interface, but can you do it again, please? Maybe you mixed up HW Address with the interface or something?

Nishchay-sopho, 2020-02-24:
Yeah sure, will check it again and update with the console log.

roberthartung, 2020-02-24:
@miri64  `ipv6_addr_split_iface` splits the interface using `ipv6_addr_split_str`. However `ipv6_addr_split_str` returns a *new* pointer, which is the interface, and it does *not* change the parameter.

miri64, 2020-02-24:
I just tried the following.

- I added an output to `udp.c`:
```diff
diff --git a/examples/gnrc_networking/udp.c b/examples/gnrc_networking/udp.c
index e8a559846e..811fce586a 100644
--- a/examples/gnrc_networking/udp.c
+++ b/examples/gnrc_networking/udp.c
@@ -52,6 +52,7 @@ static void send(char *addr_str, char *port_str, char *data, unsigned int num,
         netif = gnrc_netif_get_by_pid(atoi(iface));
     }
 
+    printf("%s\n",addr_str);
     /* parse destination address */
     if (ipv6_addr_from_str(&addr, addr_str) == NULL) {
         puts("Error: unable to parse destination address");
```
- I flashed that to a `samr21-xpro` (only one radio, but not really important for this to have more than one)
- I started a sniffer on channel 26
- On the `samr21-xpro` I did:
```
2020-02-24 14:59:48,744 # ifconfig
2020-02-24 14:59:48,749 # Iface  7  HWaddr: 2D:4B  Channel: 26  Page: 0  NID: 0x23
2020-02-24 14:59:48,754 #           Long HWaddr: 5A:9D:93:86:22:08:65:78 
2020-02-24 14:59:48,761 #            TX-Power: 0dBm  State: IDLE  max. Retrans.: 3  CSMA Retries: 4 
2020-02-24 14:59:48,768 #           AUTOACK  ACK_REQ  CSMA  L2-PDU:102 MTU:1280  HL:64  RTR  
2020-02-24 14:59:48,770 #           6LO  IPHC  
2020-02-24 14:59:48,773 #           Source address length: 8
2020-02-24 14:59:48,776 #           Link type: wireless
2020-02-24 14:59:48,781 #           inet6 addr: fe80::589d:9386:2208:6578  scope: link  VAL
2020-02-24 14:59:48,784 #           inet6 group: ff02::2
2020-02-24 14:59:48,787 #           inet6 group: ff02::1
2020-02-24 14:59:48,791 #           inet6 group: ff02::1:ff08:6578
2020-02-24 14:59:48,794 #           inet6 group: ff02::1a
2020-02-24 14:59:48,795 #           
2020-02-24 14:59:48,798 #           Statistics for Layer 2
2020-02-24 14:59:48,801 #             RX packets 4  bytes 172
2020-02-24 14:59:48,806 #             TX packets 14 (Multicast: 10)  bytes 558
2020-02-24 14:59:48,809 #             TX succeeded 10 errors 0
2020-02-24 14:59:48,812 #           Statistics for IPv6
2020-02-24 14:59:48,815 #             RX packets 4  bytes 256
2020-02-24 14:59:48,819 #             TX packets 14 (Multicast: 10)  bytes 834
2020-02-24 14:59:48,823 #             TX succeeded 14 errors 0
2020-02-24 14:59:48,823 # 
udp send fe80::589d:9386:2208:6579%7 1337 test
2020-02-24 14:59:55,772 #  udp send fe80::589d:9386:2208:6579%7 1337 test
2020-02-24 14:59:55,774 # fe80::589d:9386:2208:6579
2020-02-24 14:59:55,781 # Success: sent 4 byte(s) to [fe80::589d:9386:2208:6579]:1337
udp send fe80::589d:9386:2208:6579%8 1337 test
2020-02-24 15:00:00,804 #  udp send fe80::589d:9386:2208:6579%8 1337 test
2020-02-24 15:00:00,806 # fe80::589d:9386:2208:6579
2020-02-24 15:00:00,812 # Success: sent 4 byte(s) to [fe80::589d:9386:2208:6579]:1337
> 
```
- the sniffer showed
[image]

As you can see,the IPv6 address is split of correctly by `ipv6_addr_split_iface()`, the first UDP packet is sent (with link-layer retransmissions) the second one (see timestamp) is not (as expected).

roberthartung, 2020-02-24:
Let's wait for @Nishchay-sopho and see what he tells us and why it failed in the first place. Thanks @miri64!

miri64, 2020-02-24:
> However, the adress parsing shouldn't fail, right? Because the `\0` should replace separator and therefore parsing the address should work.

Exactly!

Nishchay-sopho, 2020-02-24:
> Let's wait for @Nishchay-sopho and see what he tells us and why it failed in the first place. Thanks @miri64!

I called the udp send function with the ipv6 address as ``` fe80::2014:3f03:3173:6862%8```.

Snippet of udp.c :
```
printf("Address before parsing iface: %s.\n",addr_str);
    iface = ipv6_addr_split_iface(addr_str);
    if ((!iface) && (gnrc_netif_numof() == 1)) {
      printf("Interface is null.\n");
        netif = gnrc_netif_iter(NULL);
    }
    else if (iface) {
      printf("Interface is not null with value %d.\n", atoi(iface));
        netif = gnrc_netif_get_by_pid(atoi(iface));
    } else {
      printf("Falling back to 8.\n");
        netif = gnrc_netif_get_by_pid(8);
    }
    printf("iface is %d.\n",*iface);
    printf("netif is %s.\n", !netif?"not there":"there");
    printf("Address after parsing iface: %s.\n",addr_str);
    /* parse destination address */
    if (ipv6_addr_from_str(&addr, addr_str) == NULL) {
        puts("Error: unable to parse destination address");
        return;
    }
```

Console log of it:
```
2020-02-24 16:09:07,681 # Address before parsing iface: fe80::2014:3f03:3173:6862%8.
2020-02-24 16:09:07,684 # Interface is not null with value 8.
2020-02-24 16:09:07,687 # iface is 56.
2020-02-24 16:09:07,687 # netif is there.
2020-02-24 16:09:07,692 # Address after parsing iface: fe80::2014:3f03:3173:6862%8.
2020-02-24 16:09:07,699 # Error: unable to parse destination address
```

miri64, 2020-02-24:
> 2020-02-24 16:09:07,681 # Address before parsing iface: fe80::2014:3f03:3173:6862%8.
> 2020-02-24 16:09:07,684 # Interface is not null with value 8.

Up until here everything makes sense

> 2020-02-24 16:09:07,687 # iface is 56.

56 is the ASCII value of `'a'`, so given that you used `%d` (not `%c`) this also makes sense.

> 2020-02-24 16:09:07,687 # netif is there.

:heavy_check_mark: (with `netif->pid` you can check if it is actually interface 8.)

> 2020-02-24 16:09:07,692 # Address after parsing iface: fe80::2014:3f03:3173:6862%8.

This is exactly opposite to what I tried above. Which version of RIOT are you using? I used f0c68a7433b1ca80f67631a02b22d (somewhat current with master).

Nishchay-sopho, 2020-02-24:
It is the one forked from the PR #12128 .

janschlicht, 2020-02-24:
@Nishchay-sopho I just checked which commit we use and it is  c8461a4bef61925bbfaece1b94ed7f26871d5f87 with the radio driver from #12128 applied on top. We can do a rebase tomorrow to a more recent version.

Nishchay-sopho, 2020-02-24:
ok sure

miri64, 2020-02-25:
> @Nishchay-sopho I just checked which commit we use and it is [c8461a4](https://github.com/RIOT-OS/RIOT/commit/c8461a4bef61925bbfaece1b94ed7f26871d5f87) with the radio driver from #12128 applied on top. We can do a rebase tomorrow to a more recent version.

Mh there were no changes to the code in question between c8461a4 (`git log --oneline c8461a4..f0c68a7 -- examples/gnrc_networking sys/net/network_layer/ipv6/addr/`), so I don't really understand how your code is not working but mine is. :-/.

Nishchay-sopho, 2020-02-25:
We have still no idea WHY this is happening, but we know that the original address is not changed:

```
char *ipv6_addr_split_str(char *addr_str, char separator)
{
    char *sep = addr_str;
    while (*(++sep)) {
      printf("comparing %c == %c\n", *sep, separator);
        if (*sep == separator) {
           printf("separator changed????\n");
            *sep++ = '\0';
            break;
        }
    }
    printf("addr in split function: '%s' at %p with '%c'.\n", sep, addr_str, addr_str[25]);
    return *sep ? sep : NULL;
}
```

```
2020-02-25 10:51:45,025 # addr pointer: 0x80162b2
2020-02-25 10:51:45,026 # Address before parsing iface: 'fe80::4417:5f03:3b73:686d%8' with character '%'.
2020-02-25 10:51:45,026 # comparing e == %
2020-02-25 10:51:45,026 # comparing 8 == %
2020-02-25 10:51:45,026 # comparing 0 == %
2020-02-25 10:51:45,026 # comparing : == %
2020-02-25 10:51:45,027 # comparing : == %
2020-02-25 10:51:45,027 # comparing 4 == %
2020-02-25 10:51:45,027 # comparing 4 == %
2020-02-25 10:51:45,027 # comparing 1 == %
2020-02-25 10:51:45,027 # comparing 7 == %
2020-02-25 10:51:45,027 # comparing : == %
2020-02-25 10:51:45,027 # comparing 5 == %
2020-02-25 10:51:45,028 # comparing f == %
2020-02-25 10:51:45,028 # comparing 0 == %
2020-02-25 10:51:45,028 # comparing 3 == %
2020-02-25 10:51:45,028 # comparing : == %
2020-02-25 10:51:45,028 # comparing 3 == %
2020-02-25 10:51:45,028 # comparing b == %
2020-02-25 10:51:45,029 # comparing 7 == %
2020-02-25 10:51:45,029 # comparing 3 == %
2020-02-25 10:51:45,029 # comparing : == %
2020-02-25 10:51:45,029 # comparing 6 == %
2020-02-25 10:51:45,029 # comparing 8 == %
2020-02-25 10:51:45,029 # comparing 6 == %
2020-02-25 10:51:45,029 # comparing d == %
2020-02-25 10:51:45,030 # comparing % == %
2020-02-25 10:51:45,030 # separator changed????
2020-02-25 10:51:45,030 # addr in split function: '8' at 0x80162b2 with '%'.
2020-02-25 10:51:45,030 # Interface is not null with value 8.
2020-02-25 10:51:45,030 # iface is 56.
2020-02-25 10:51:45,031 # netif is there.
2020-02-25 10:51:45,031 # Address after parsing iface: 'fe80::4417:5f03:3b73:686d%8' at 0x80162b2.
2020-02-25 10:51:45,032 # Error: unable to parse destination address
```

janschlicht, 2020-02-25:
Just a quick note: while the issue persist with that version of riot we were able to get it working with the most recent master version (d044800300b0392280791d802ef7fc0e5cdf52cb).

miri64, 2020-02-25:
Then can we assume the issue as fixed? (even though we don't understand what happened?)

janschlicht, 2020-02-28:
Yes it can be closed. We are not sure what causes the issue but since it works with the newer version of riot we won't invest more time in debugging.

closed 2020-03-03 by miri64