RIOT-OS/RIOT #13444: Potential security and safety race conditions on attached devices

p2041, 2020-02-22:
#### Description
Dear, RIOT developers

I checked your codes for some reason.
During that time, I found some potential race conditions which can lead to security and safety issues.

For example, SD cards can be accessed by multiple threads or processes. 
In the current system, SD card APIs (e.g., sdcard_spi_read_sds) are accessed by user-level applications.

In each function, multiple device accesses are involved.

Then, if there are multiple reading or writing requests from the multiple user-level applications, the internal state in the SD card controller (or SD card) may be unpredictable (e.g., reading mode can be suddenly changed to writing mode); hence, it can cause data corruption IMO.

For that reason, I checked the patch history and I found you applied some patches.

**P1.** https://github.com/RIOT-OS/RIOT/pull/2326/commits   -> “Transaction”-level I2C race condition device protection. But, this lock protection is not enforced to all the devices (attachable to I2C)

**P2.** https://github.com/RIOT-OS/RIOT/pull/2323/commits   -> “Access”-level I2c race condition - not at a “Transaction”-level

**P3.** https://github.com/RIOT-OS/RIOT/pull/2317/commits   -> “Access”-level SPI race condition - not at a “Transaction”-level (Only partially enforced)

Based on my patch inspection, I found that **P1** is consistent with my opinion. 
However, this patch’s policy is not applied to many of the devices attachable to SPI and I2C.

I think they must be protected with a single lock at one transaction level (e.g., SD card’s “sdcard_spi_read_sds”) especially if there are multiple threads or processes.

Here is the list of the devices which are vulnerable (and their patches are not consistent with P1) as follows:
 - sht3x, SD card controller, hih6130, ad7746, ds75lx, fxos8700, hdc1000, lis3dh, shell_run.

There could be more. But, I didn’t analyze all of them.

Based on the analysis, 

**1. sdcard_spi:** I already described as an example : )

**2. hih6130:** Simplified procedure is as follows: command request, wait, read https://github.com/RIOT-OS/RIOT/blob/26c689ffbb2967e87ad5b2971be37551f4742426/drivers/hih6130/hih6130.c#L118
In function “hih6130_get_humidity_temperature_float”

        if (hih6130_measurement_request(dev) != 0) { // command request
             return -1; 
        }       

        xtimer_usleep(MEASUREMENT_DELAY); // wait
        status = hih6130_get_humidity_temperature_raw(dev, &hum_raw, &temp_raw); // read

   First of all, there is no lock covering all. Then, after there is a request, host firmware may wait for its requested job completion. But, if there is some thread to do the samething, it can corrupt read data or cause failure (for “hih6130_get_humidity_temperature_raw”) . So, I think there should be a lock spanning these three steps.

**3. ad7746:** It is a case similar to hih6130 
    This is related to https://github.com/RIOT-OS/RIOT/blob/master/drivers/ad7746/ad7746.c#L371
    “ad7746_read_temperature_int” and “_read_voltage_temp” are called sequentially. 

       if (dev->params.vt_mode != mode) {
            if (ad7746_set_vt_ch_mode(dev, mode)) { // “ad7746_set_vt_ch_mode”
                return AD7746_NOI2C;
            }

            xtimer_usleep(_vt_sr_times[dev->params.vt_sample_rate] * US_PER_MS); //sleep
            _read_raw_ch(dev, AD7746_READ_VT_CH, &raw); // _read_raw_ch
       }
       res = _read_raw_ch(dev, AD7746_READ_VT_CH, &raw); // _read_raw_ch

    Then, “ad7746_set_vt_ch_mode” can be called. 

    If so, there is sleep time. Before sleeping is done, if another request comes in, reading operation by “_read_raw_ch” can be unpredictable with a different thread.

**4. ds75lx:** This is also similar to Case 2 and 3.
Based on tests/driver_ds75lx/main.c 
https://github.com/RIOT-OS/RIOT/blob/eafa381a06fed27bedc938c870567c68ceda84d3/tests/driver_ds75lx/main.c#L47 
I think “ds75lx_wakeup” repeatedly happens.

    Then, inside of ds75lx_wakeup function, it also has sleep after reading some value w/o lock.

    Here is the related code. https://github.com/RIOT-OS/RIOT/blob/9f58ad6e3e8f3216b37fbb9575fb61d7371bc1ca/drivers/ds75lx/ds75lx.c

**5. fxos8700:** https://github.com/RIOT-OS/RIOT/blob/26c689ffbb2967e87ad5b2971be37551f4742426/drivers/fxos8700/fxos8700.c#L136
  In function “fxos8700_read”, 

       if (fxos8700_set_active(dev)) // active 
        …..
        fxos8700_read_regs(dev, FXOS8700_REG_STATUS, &ready, 1); // read something
        ….
       if (fxos8700_set_idle(dev)) { // idle

    “set_active” after “set_idle”  before “fxos8700_read_regs” can happen since there is no lock spanning them.

**6. hdc1000:** It is similar to Case 2,3,4, 
    In function “hdc1000_read” https://github.com/RIOT-OS/RIOT/blob/2fde8335c950f2e1279a56b6f7495cf428ab19cd/drivers/hdc1000/hdc1000.c#L128 

        if (hdc1000_trigger_conversion(dev) != HDC1000_OK) { // Command send = conversion
            return HDC1000_BUSERR;
        }
        xtimer_usleep(HDC1000_CONVERSION_TIME); // wait
        return hdc1000_get_results(dev, temp, hum); // Read data

   One lock should span all of them to prevent from conversion operation failure.

**7. lis3dh** (couting for loop by lis3dh_get_fifo_level, then, reading, no lock for # of loops) https://github.com/RIOT-OS/RIOT/blob/1ffca2b963234dcc8d4d3ce67632b268546c8988/tests/driver_lis3dh/main.c#L127
  I think this is a typical way to read the LIS3DH sensor values. 
  That means sensor reading using the same loop will be used by other users.

   Unlike the others, it interestingly gets the fifo level first. Then, codes read the value according to “fifo_level”. But, concurrently access may dequeue the data faster.  I think it may need to be locked by one lock.

        fifo_level = lis3dh_get_fifo_level(&dev); // fifo level
        … 
        while (fifo_level > 0) { // # of repeation = fifo_level
            if (lis3dh_read_xyz(&dev, &acc_data) != 0) { // read 
            ...
            if (lis3dh_read_aux_adc3(&dev, &temperature) != 0) { // read
            ...
            int1 = gpio_read(lis3dh_params[0].int1);
            ...

**8. sht3x**
   To read sensor values, “sht3x_read” is called. Then, “_get_raw_data” is called as you can see in the link. https://github.com/RIOT-OS/RIOT/blob/61a78cf81790b9785a109d6a792f8cfb92ecb846/drivers/sht3x/sht3x.c#L207

   Then, you can see there are multiple steps.

        if (!dev->meas_started && (res = _start_measurement (dev)) != SHT3X_OK) { // measure
            …
        if (elapsed < dev->meas_duration) {
            xtimer_usleep(dev->meas_duration - elapsed); // sleep 
            ….
        if (dev->mode != SHT3X_SINGLE_SHOT &&
           _send_command(dev, SHT3X_CMD_FETCH_DATA) != SHT3X_OK) { // send command
           ....
        if (_read_data(dev, raw_data, SHT3X_RAW_DATA_SIZE) != SHT3X_OK) { // read
           ….

    If there are multiple threads, it can cause failure or result corruption.

**9. shell_run**
    When I check “AT30TSE75x”, I could not find a lock protecting “shell_run”. I think this can also cause weird console output IMO (due to race condition).

#### Steps to reproduce the issue
If I2C or SPI are concurrently accessed, this problem can happen.

#### Expected results
Peripheral device operation failure.
Potentially, it can cause various security and safety issues.
e.g., The data (in SD card controlled via Sd card controller attached to SPI) can be corrupted due to race condition.
It is applicable to various boards supported by RIOT.

#### Versions
Applicable to recent versions.

gschorcht, 2020-02-25:
@maribu Could you publish your answer to the initial email here as input for further discussion?

maribu, 2020-02-25:
On Wed, 12 Feb 2020 10:02:10 +0100
Marian Buschsieweke <marian.buschsieweke@ovgu.de> wrote:

> Hi,
> 
> this seems not to be security related. An attacker who would be able to call C
> functions on a device without MMU already has full access to the device. I
> doubt that any attacker who already has gained full code execution cares about
> potential race conditions in drivers and would just scramble the memory, if the
> goal is to corrupt the state of drivers...
> 
> Also: Is it really worth the overhead to protect e.g. a sensor like the SHT30
> using a mutex? I don't see a use case where multiple threads need to access the
> sensor. I suggest to instead point this out in the documentation, so that
> anyone really wanting to have multiple threads concurrently accessing the
> sensor to handle the locking internally.
> 
> (I wouldn't mind to abuse the `i2c_acquire()` and `i2c_release()` to also
> synchronize access to the driver state, when carefully combined with I2C
> transactions. That wouldn't add ROM size (it would only move some lines of code
> inside already existing calls to `i2c_acquire()` and `i2c_release()`) and would
> have negligible impact on the duration the I2C bus is blocked. But any byte of
> additional ROM required to synchronize access to device that in practice are
> never accessed concurrently seem to be a bad trade off to me.)
> 
> Kind regards,
> Marian

maribu, 2020-02-25:
Please note: Due to my parental leave I won't be able to work on this for a while.

However, I suggest something like the following todo:

1. Identify a list of modules / drivers that are not thread safe.
2. Identify for which of them thread safety is actually beneficial
3. Add thread safety to those identified in 2., extend the documentation of the other modules with an `@warning This module is not thread safe.` to the remaining modules.

Maybe we could also document this on a higher level. E.g. we could document that all actuator and sensor drivers are not thread safe, unless explicitly stated in the documentation.

maribu, 2023-05-18:
Won't fix. Sensor and actuator device drivers are not written with thread safety in mind. Applications need to be written in a way that at most one thread interacts with every given sensor / actuator.

This could be more clearly documented which APIs are thread-safe and which are not. But I think a tracking issue would be better suited to tackle this.

closed 2023-05-18 by maribu