RIOT-OS/RIOT #13459: cpu/cortex-m23: compiler bug when compiling c11 atomics

kaspar030, 2020-02-24:
#### Description

```
/home/kaspar/src/riot/tests/unittests/tests-core/tests-core-atomic.c: In function 'test_atomic_inc_negative':
/home/kaspar/src/riot/tests/unittests/tests-core/tests-core-atomic.c:53:1: error: insn does not satisfy its constraints:
   53 | }                                                        
      | ^                                                        
(insn 145 144 146 4 (set (reg:SI 3 r3 [139])          
        (plus:SI (reg:SI 12 ip [orig:110 _1 ] [110])
            (const_int 1 [0x1]))) "/home/kaspar/src/riot/tests/unittests/tests-core/tests-core-atomic.c":50:9 811 {*thumb1_addsi3}
     (nil))                                                      
during RTL pass: cprop_hardreg                                   
/home/kaspar/src/riot/tests/unittests/tests-core/tests-core-atomic.c:53:1: internal compiler error: in extract_constrain_insn, at r
ecog.c:2211                                                      
Please submit a full bug report,        
with preprocessed source if appropriate.
See <https://bugs.archlinux.org/> for instructions.
make[2]: *** [/home/kaspar/src/riot/Makefile.base:110: /home/kaspar/src/riot/tests/unittests/bin/saml11-xpro/tests-core/tests-core-
atomic.o] Error 1                                                                                                                  
```

#### Steps to reproduce the issue

Both Cortex-M23 boards (saml10-xpro and saml11-xpro) are blacklisted now, but try compiling unittests for one of them.

#### Expected results

#### Actual results

#### Versions

dylad, 2020-02-24:
@kaspar030 Could you specify which compiler version you were using ?

kaspar030, 2020-02-24:
> @kaspar030 Could you specify which compiler version you were using ?

Sorry,

The most recent ARM toolchain from here: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads (9-2019-q4), which is now the one used by our Docker container.

Same my arch linux, using gcc 9.2.0: https://gist.github.com/7d9712b1576ead201b4af93635a3ccc9

dylad, 2020-02-24:
Huh my version is old now I guess...
No problem so far with 7-2018-q2 from ARM.
I'll update and have a look.

dylad, 2020-02-26:
Closing this as #13462 fixes it.
The patch can be reverted if future GCC version fix it on its side.

closed 2020-02-26 by dylad

dylad, 2021-02-18:
@kaspar030 FYI I just tested with `gcc-arm-none-eabi-10-2020-q4` and the GCC bug was fixed.
We could revert #13462 at some point later.