RIOT-OS/RIOT #13460: Makefile.dep: DEFAULT_MODULE += stdio_rtt will always load dependencies - even if other stdio is selected

krasoa, 2020-02-24:
#### Description

Affected boards:
```
feather-nrf52840
hamilton
pinetime
ruuvitag
thingy52
```

`DEFAULT_MODULE += stdio_rtt` in the `/boards/affected_board/Makefile.dep` will force the dependencies of `stdio_rtt` (e.g. xtimer) to load, even if another stdio module is selected. 

#### Steps to reproduce the issue/Actual results

`USEMODULE=stdio_null BOARD=ruuvitag make info-modules` will print
```
auto_init                                         
-->  auto_init_xtimer                               
board
boards_common_nrf52xxdk
core
core_init
core_msg
core_panic
cortexm_common
cortexm_common_periph
cortexm_fpu
cpu
cpu_common
-->  div                                                    
newlib
newlib_nano
newlib_syscalls_default
nrf5x_common_periph
periph
periph_common
periph_gpio
periph_pm
-->  periph_timer                                           
stdio_null
sys
-->  xtimer                                                  
```
Whereas `BOARD=ruuvitag` can obviously be replaced with any other affected board.

#### Expected results

Output above without lines beginning with  `-->`.

#### Versions

Master

haukepetersen, 2020-02-24:
@aabadie would you mind to take a look? Thanks!

miri64, 2020-03-18:
Not completely solved yet, right? See https://github.com/RIOT-OS/RIOT/pull/13650#pullrequestreview-376900604 and https://github.com/RIOT-OS/RIOT/pull/13650#issuecomment-600669167.

miri64, 2020-03-18:
Summon @fjmolinas ;-) (was not part of the conversation so far)

fjmolinas, 2020-03-20:
> Not completely solved yet, right? See #13650 (review) and #13650 (comment).

Hmm yes and no, for `stdio_rtt` `DEFAULT_MODULE` it works, what I mentioned in #13650 is a different issue since it relates to `bootloader_arduino`. The issue is fixed for `stdio_%` modules IMO.

miri64, 2020-03-20:
Fine by me.

closed 2020-03-20 by miri64

fjmolinas, 2020-03-20:
BTW opened #13664 to addres the mentioned issue

PR #13650 (2020-03-18): Makefile.dep: disable stdio_% modules before they are included
### Contribution description

This PR moves all `stdio_%` dependency resolution to is own file to cleanup inclusion order and better isolate the dependencies. Maybe #13469 will lead to a parent module and the dependencies can be moved to that `Makefile.dep`?

This PR also fixes disabling of `stdio` modules when this is done not by using `DISABLE_MODULE` directly but by the use of a conflicting `USEMODULE`.

Before this PR because of recursive declaration, on the first pass through `Makefile.dep` `stdio_rtt` would be added and then  [... cut]