RIOT-OS/RIOT #13471: 6ctx: incorrectly removes 6ctx context when the non-zero part of the address matches

bergzand, 2020-02-25:
I'm not really sure how to describe this

#### Description

The 6ctx module seems to incorrectly remove 6ctx context when the non-zero part of the address matches (I think).

#### Steps to reproduce the issue

1. Deploy the gnrc_border_router example as described in #8576. 
2. Start the terminal with: `make -C examples/gnrc_border_router -j BOARD=samr21-xpro term IPV6_PREFIX="2001:db8::/64"`
3. Wait for the node to configure the prefix as 6ctx context.
4. Stop the shell (don't reboot the node).
5. Start the terminal with: `make -C examples/gnrc_border_router -j BOARD=samr21-xpro term IPV6_PREFIX="2001:db8:1::/64"` (different prefix)
6. The node should still have the old prefix
7. Wait for the node to pick up the new prefix.
8. The node only has 2001:db8:1::/64" as 6ctx prefix.
9. Repeat with `IPV6_PREFIX="2001:db8:2::/64"`
10. The node will have both `2001:db8:1::/64` and 2001:db8:2::/64` as 6ctx contexts.

#### Expected results

The node had kept the `2001:db8::/64` prefix.

#### Actual results

The node removed the `2001:db8::/64` prefix.

miri64, 2020-02-25:
Up until now, I wasn't able to reproduce this in a unittest :-/

```C
diff --git a/tests/unittests/tests-sixlowpan_ctx/tests-sixlowpan_ctx.c b/tests/unittests/tests-sixlowpan_ctx/tests-sixlowpan_ctx.c
index e4dfb9d40f..3da821ecb0 100644
--- a/tests/unittests/tests-sixlowpan_ctx/tests-sixlowpan_ctx.c
+++ b/tests/unittests/tests-sixlowpan_ctx/tests-sixlowpan_ctx.c
@@ -103,6 +103,34 @@ static void test_sixlowpan_ctx_update__wrong_id2(void)
     TEST_ASSERT_NOT_NULL(gnrc_sixlowpan_ctx_lookup_id(DEFAULT_TEST_ID));
 }
 
+static void test_sixlowpan_ctx_update__i13471(void)
+{
+    /* regression test for https://github.com/RIOT-OS/RIOT/issues/13471 */
+    ipv6_addr_t addr = {
+            .u8 = { 0x20, 0x01, 0x0d, 0xb8, 0x00, 0x00, 0x00, 0x00,
+                    0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00 }
+        };
+    gnrc_sixlowpan_ctx_t *ctx;
+
+    TEST_ASSERT_NOT_NULL(gnrc_sixlowpan_ctx_update(0, &addr,
+                                                   64U, 60U, true));
+    TEST_ASSERT_NOT_NULL((ctx = gnrc_sixlowpan_ctx_lookup_id(0)));
+    TEST_ASSERT_EQUAL_INT(GNRC_SIXLOWPAN_CTX_FLAGS_COMP, ctx->flags_id);
+    TEST_ASSERT_EQUAL_INT(64U, ctx->prefix_len);
+    TEST_ASSERT(60U >= ctx->ltime);
+    TEST_ASSERT(64U <= ipv6_addr_match_prefix(&addr, &ctx->prefix));
+    addr.u8[6] = 0x01;  /* set addr's prefix from 2001:db8:: to 2001:db8:1:: */
+    TEST_ASSERT_NOT_NULL(gnrc_sixlowpan_ctx_update(1, &addr,
+                                                   64U, 60U, true));
+    addr.u8[6] = 0x00;  /* re-set to 2001:db8:: for comparison */
+    TEST_ASSERT_NOT_NULL((ctx = gnrc_sixlowpan_ctx_lookup_id(0)));
+    TEST_ASSERT_EQUAL_INT(GNRC_SIXLOWPAN_CTX_FLAGS_COMP, ctx->flags_id);
+    TEST_ASSERT_EQUAL_INT(64U, ctx->prefix_len);
+    TEST_ASSERT(60U >= ctx->ltime);
+    TEST_ASSERT(64U <= ipv6_addr_match_prefix(&addr, &ctx->prefix));
+    TEST_ASSERT_NOT_NULL(gnrc_sixlowpan_ctx_lookup_id(1));
+}
+
 static void test_sixlowpan_ctx_update__wrong_prefix_len(void)
 {
     ipv6_addr_t addr = DEFAULT_TEST_PREFIX;
@@ -203,6 +231,7 @@ Test *tests_sixlowpan_ctx_tests(void)
         new_TestFixture(test_sixlowpan_ctx_update__wrong_prefix_len),
         new_TestFixture(test_sixlowpan_ctx_update__success),
         new_TestFixture(test_sixlowpan_ctx_update__ltime0),
+        new_TestFixture(test_sixlowpan_ctx_update__i13471),
         new_TestFixture(test_sixlowpan_ctx_lookup_addr__empty),
         new_TestFixture(test_sixlowpan_ctx_lookup_addr__same_addr),
         new_TestFixture(test_sixlowpan_ctx_lookup_addr__other_addr_same_prefix),
```

Will investigate further.

miri64, 2020-02-25:
This is a bug introduced by #8576. I fixed it there.

closed 2020-02-25 by miri64