RIOT-OS/RIOT #13473: SCHC: LoRa Header Compression + Fragmentation for RIOT?

cgundogan, 2020-02-25:
#### Description
Just a pointer that there is a minimal [SCHC](https://datatracker.ietf.org/doc/draft-ietf-lpwan-ipv6-static-context-hc/) implementation [here](https://github.com/imec-idlab/libschc). It includes a static context header compression and fragmentation for LoRa. Maybe integrating this package is a *good first issue*? (:

### Useful links
* https://github.com/imec-idlab/libschc
* https://datatracker.ietf.org/doc/draft-ietf-lpwan-ipv6-static-context-hc/

aabadie, 2020-02-25:
The library is licensed under GPLv3. Is it compatible with RIOT license ? (LGPV2.1)

cgundogan, 2020-09-14:
@bartmoons maybe you can answer the question of @aabadie ?

bartmoons, 2020-09-14:
I'm not a lawyer, but from what I can read, I think they are! [Here](https://www.gnu.org/licenses/gpl-faq.en.html#AllCompatibility) you can see a compatibility matrix, where GPLv3 matches LGPLv2.1.

cgundogan, 2020-09-17:
see #15040

closed 2021-07-08 by stale[bot]