RIOT-OS/RIOT #13490: cpu/stm32/eth, board/nucleo-f767z1: ethernet initalisation fails sometimes

kfessel, 2020-02-26:
#### Description

Starting RIOT on the nucleo-f767z1 Ethernet sometimes fails to work often after flashing a new firmware sometimes after a reset most the time the green Ethernet-connector-LED stays dark when this happens.

#### Steps to reproduce the issue

make and flash a IPv6 example to the nucleo-f767z1 and run a PING to its IPv6 address keep the ping running  (linux does it endlessly by default)

do resets (button) short and long presses
do reflashes

#### Expected results

ping works every time after RIOT boot  (real network issues aside)

#### Actual results

ping does not work after some resets or flashes

#### Versions

current master

PeterKietzmann, 2020-02-26:
Discovered testing #13383, #13487

kfessel, 2020-02-26:
there might be a hardware issue part in this Problem (the manufacturer of the LAN8742A advise against RC-networks at the reset pin) the nucleo-f767z1 has an RC Network there (R are the combination of all parallel reset-pin pullups in the ICs and the C is accross the Reset switch)

closed 2020-10-23 by benpicco