RIOT-OS/RIOT #13492: make -j flash broken on esp* (will always flash the previous binary)

benpicco, 2020-02-26:
#### Description
I'm a habitual user of `make -j`. It looks like the build system has developed a very reliable race condition where a `make -j flash` will always flash the previous binary on esp*.

#### Steps to reproduce the issue

- In `examples/hello-world` run `make BOARD=esp32-wroom-32 -j flash term`.
- change the text in `main.c`, better yet print it in a `while(1)` loop
- run `make BOARD=esp32-wroom-32 -j flash term` again

#### Expected results
- the new text is printed

#### Actual results
- the old text is printed.
- run `make BOARD=esp32-wroom-32 -j flash term` again
- the new text is printed.

#### Workaround

- run `make BOARD=esp32-wroom-32 -j && make BOARD=esp32-wroom-32 flash term`

#### Versions
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0
GNU Make 4.2.1

gschorcht, 2020-05-08:
Hm, strange an not really clear why. `main.c` is definitly recompiled and the change is in ELF file `tests_shell.elf`. I have checked this by renaming a function in `main.c`.

gschorcht, 2020-05-08:
It seems that it starts `preflash` target before the `tests_shell.elf` has been updated. At least I can observe with `QUIET=0` that the following preparation of the flash file is executed before the compilation of the ELF file `tests_shell.elf` has been finished:
https://github.com/RIOT-OS/RIOT/blob/2d1d13d45276e238e7ac65fe80d4c54ddd8c65a8/cpu/esp_common/Makefile.include#L96-L106

This part is executed by the `preflash` target and `flash` depends on `preflash` https://github.com/RIOT-OS/RIOT/blob/2d1d13d45276e238e7ac65fe80d4c54ddd8c65a8/cpu/esp_common/Makefile.include#L110

So it seems to be a missing dependency of the `preflash` target on the ELF file. Maybe I misinterpreted the `preflash` target when I wrote the makefiles. I thought it is a target that is executed before an ELF file is flashed. But obviously it only depends on the `BUILD_BEFORE_FLASH` variable which isn't set anywhere https://github.com/RIOT-OS/RIOT/blob/2d1d13d45276e238e7ac65fe80d4c54ddd8c65a8/Makefile.include#L702

There are two options to solve the problem. Either we move the code for the preparation of the flash binary to the `flash` target or we just set the variable
```
BUILD_BEFORE_FLASH += $(FLASHFILE)
```

gschorcht, 2020-05-12:
The fix in PR #14041 had to be reverted for the moment.

jenswet, 2021-01-08:
I can reproduce the same problem on the current master.

PR #14041 (2020-05-08): cpu/esp_common: fix the dependency of the flash image on the ELF file
### Contribution description

This PR fixes the dependency of the flash image on the ELF file for the ESP* make system.

Flashing an ESP board first requires the creation of a flash image from the ELF file.  This is realized in the `preflash` target. However, the `preflash` target only depends on the variable `BUILD_BEFORE_FLASH` but on the ELF file. Therefore, the variable `BUILD_BEFORE_FLASH` must be set to the ELF file to ensure that when using multiple make processes, the compilation of the ELF file is completed before t [... cut]
closed 2021-04-27 by fjmolinas