RIOT-OS/RIOT #13496: cpu/stm32/eth, board/nucleo-f767z1: ethernet buffering issue (ping >1s)

kfessel, 2020-02-27:
#### Description

while debugging a nucleo-f767zi during break Ethernet packets are seem to get buffered this buffer is not processed to completion after continuation.

#### Steps to reproduce the issue

push something like /examples/gcoap to your board 

    cd <riot>/example/gcoap
    make BOARD=nucleo-f767zi -j 8 clean all flash

second terminal > ping the board:
    
    ping -6 fe80::...

    see pings that are ok 

keep pinging

start debugging
    
    make BOARD=nucleo-f767zi debug
    
    ....
    (gdb) c
    Continuing.

second terminal see the pings:
    
    64 bytes from fe80::...: icmp_seq=202 ttl=64 time=2007 ms

    

#### Expected results

lose some packets while in break but then get back to normal

ping <= 1 digit ms 

#### Actual results

ping stays bad most the time ~1 s 
while preparing this issue report it was worse

#### Versions

ubunut 19...

RIOT current master

PeterKietzmann, 2020-02-28:
> lose some packets while in break but then get back to normal

So I assume you quit the debug session at one point while the other node still pinged. And the RTT never went down to its original value before debugging.

kfessel, 2020-02-28:
since when quitting the debug session the mcu is left hated i never tested this ( i will in a couple of minutes)

kfessel, 2020-02-28:
the pings return to normal if `run` is issued -> reinitalisation back to normal

    (gdb) run
    The program being debugged has been started already.
    Start it from the beginning? (y or n) y

closed 2020-08-17 by benpicco