RIOT-OS/RIOT #13501: SDL2 does not work due to missing getpid

ccoors, 2020-02-28:
#### Description
Trying to initialize SDL2 on the native board leads to a segmentation fault (maybe) because `getpid()` is not implemented.

#### Steps to reproduce the issue
```
$ cd tests/pkg_u8g2/
$ make TEST_OUTPUT=2
$ bin/native/tests_pkg_u8g2.elf 
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

Help: Press s to start test, r to print it is ready
s
START
main(): This is RIOT! (Version: 2020.04-devel-821-g23c5d)
Initializing to SDL.
bin/native/tests_pkg_u8g2.elf: not implemented
bin/native/tests_pkg_u8g2.elf: not implemented
[1]    25234 segmentation fault (core dumped)  bin/native/tests_pkg_u8g2.elf
```

Running the same binary in a Wayland-session with `SDL_VIDEODRIVER=wayland` prints one `bin/native/tests_pkg_u8g2.elf: not implemented` more, apparently `SDL_Init` does not crash but the later commands do.

Running the binary in gdb leads to this backtrace:
```
Program received signal SIGSEGV, Segmentation fault.
0xf79dcf5c in _XlcResolveLocaleName () from /usr/lib32/libX11.so.6
(gdb) bt
#0  0xf79dcf5c in _XlcResolveLocaleName () from /usr/lib32/libX11.so.6
#1  0xf79e07e6 in ?? () from /usr/lib32/libX11.so.6
#2  0xf79dfdac in ?? () from /usr/lib32/libX11.so.6
#3  0xf79e060d in _XlcCreateLC () from /usr/lib32/libX11.so.6
#4  0xf7a02402 in _XlcUtf8Loader () from /usr/lib32/libX11.so.6
#5  0xf79e806d in _XOpenLC () from /usr/lib32/libX11.so.6
#6  0xf79e81b8 in _XlcCurrentLC () from /usr/lib32/libX11.so.6
#7  0xf79e820b in XSetLocaleModifiers () from /usr/lib32/libX11.so.6
#8  0xf7dd1dd4 in ?? () from /usr/lib32/libSDL2-2.0.so.0
#9  0xf7dd8df8 in ?? () from /usr/lib32/libSDL2-2.0.so.0
#10 0xf7d9e738 in ?? () from /usr/lib32/libSDL2-2.0.so.0
#11 0xf7d093db in ?? () from /usr/lib32/libSDL2-2.0.so.0
#12 0x5655c9f8 in u8g_sdl_init (width=128, height=64) at /home/ccoors/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:100
#13 0x5655ce23 in u8x8_d_sdl_128x64 (u8g2=0x56567e70 <main_stack+11632>, msg=9 '\t', arg_int=0 '\000', arg_ptr=0x0)
    at /home/ccoors/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:274
#14 0x5655c2a3 in u8x8_SetupMemory (u8x8=0x56567e70 <main_stack+11632>) at /home/ccoors/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/csrc/u8x8_display.c:95
#15 0x5655cf3c in u8x8_Setup_SDL_128x64 (u8x8=0x56567e70 <main_stack+11632>) at /home/ccoors/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:372
#16 0x5655cf8a in u8g2_SetupBuffer_SDL_128x64_4 (u8g2=0x56567e70 <main_stack+11632>, u8g2_cb=0x56564eb0 <u8g2_cb_r0>)
    at /home/ccoors/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:390
#17 0x56556cce in main () at /home/ccoors/dev/RIOT/tests/pkg_u8g2/main.c:122
```
Doing the same thing in Wayland leads to completely different errors from `libdrm.so.2` - sorry, I can not really copy and paste from Wayland yet.
[image]

#### Expected results
SDL initializes correctly and shows the test.

#### Actual results
The `not implemented` comes from `cpu/native/syscalls.c:416`:
```
int getpid(void)
{
    warnx("not implemented");
    return -1;
}
```
I can not confirm that this actually is the problem, but it may have something to do with it.

#### Versions
```
$ make print-versions                                                                                                                                                                                        Operating System Environment
----------------------------
         Operating System: "Arch Linux" 
                   Kernel: Linux 5.5.6-arch1-1 x86_64 unknown
             System shell: GNU bash, version 5.0.16(1)-release (x86_64-pc-linux-gnu)
             make's shell: GNU bash, version 5.0.16(1)-release (x86_64-pc-linux-gnu)

Installed compiler toolchains
-----------------------------
               native gcc: gcc (Arch Linux 9.2.1+20200130-2) 9.2.1 20200130
                    clang: clang version 9.0.1 
[...]

Installed development tools
---------------------------
                   ccache: ccache version 3.7.7
                    cmake: cmake version 3.16.4
                 cppcheck: missing
                  doxygen: 1.8.17
                      git: git version 2.25.1
                     make: GNU Make 4.3
                  openocd: missing
                   python: Python 3.8.1
                  python2: Python 2.7.17
                  python3: Python 3.8.1
                   flake8: 3.7.9 (mccabe: 0.6.1, pycodestyle: 2.5.0, pyflakes: 2.1.1) CPython 3.8.1 on Linux
               coccinelle: missing
```
I also have `lib32-sdl2 2.0.10-1` installed, which it links against.

benpicco, 2020-02-28:
I encountered the same issue when I tried to do a native implementation for `drivers/ws281x`.

Unfortunately the obvious fix

```patch
--- a/cpu/native/syscalls.c
+++ b/cpu/native/syscalls.c
@@ -415,8 +415,7 @@ void errx(int eval, const char *fmt, ...)
 
 int getpid(void)
 {
-    warnx("not implemented");
-    return -1;
+    return real_getpid();
 }
```

is not correct/enough

```
main(): This is RIOT! (Version: 2020.04-devel-706-g801a7-at86rf215)
Initializing to SDL.

Program received signal SIGSEGV, Segmentation fault.
0xf78f211c in _XlcResolveLocaleName () from /lib/i386-linux-gnu/libX11.so.6
(gdb) bt
#0  0xf78f211c in _XlcResolveLocaleName () from /lib/i386-linux-gnu/libX11.so.6
#1  0xf78f57a3 in ?? () from /lib/i386-linux-gnu/libX11.so.6
#2  0xf78f4d7c in ?? () from /lib/i386-linux-gnu/libX11.so.6
#3  0xf78f55ce in _XlcCreateLC () from /lib/i386-linux-gnu/libX11.so.6
#4  0xf7917281 in _XlcUtf8Loader () from /lib/i386-linux-gnu/libX11.so.6
#5  0xf78fd00e in _XOpenLC () from /lib/i386-linux-gnu/libX11.so.6
#6  0xf78fd167 in _XlcCurrentLC () from /lib/i386-linux-gnu/libX11.so.6
#7  0xf78fd1aa in XSetLocaleModifiers () from /lib/i386-linux-gnu/libX11.so.6
#8  0xf7f16505 in ?? () from /lib/i386-linux-gnu/libSDL2-2.0.so.0
#9  0xf7f1d4f4 in ?? () from /lib/i386-linux-gnu/libSDL2-2.0.so.0
#10 0xf7f03906 in ?? () from /lib/i386-linux-gnu/libSDL2-2.0.so.0
#11 0xf7e70e72 in ?? () from /lib/i386-linux-gnu/libSDL2-2.0.so.0
#12 0x5655cd1e in u8g_sdl_init (width=128, height=64)
    at /home/benpicco/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:100
#13 0x5655d123 in u8x8_d_sdl_128x64 (u8g2=0x56566dd0 <main_stack+11632>, msg=9 '\t', arg_int=0 '\000', arg_ptr=0x0)
    at /home/benpicco/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:274
#14 0x5655c366 in u8x8_SetupMemory (u8x8=0x56566dd0 <main_stack+11632>)
    at /home/benpicco/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/csrc/u8x8_display.c:95
#15 0x5655d23c in u8x8_Setup_SDL_128x64 (u8x8=0x56566dd0 <main_stack+11632>)
    at /home/benpicco/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:372
#16 0x5655d28a in u8g2_SetupBuffer_SDL_128x64_4 (u8g2=0x56566dd0 <main_stack+11632>, u8g2_cb=0x56563e10 <u8g2_cb_r0>)
    at /home/benpicco/dev/RIOT/tests/pkg_u8g2/bin/pkg/native/u8g2/sys/sdl/common/u8x8_d_sdl_128x64.c:390
#17 0x56556cf7 in main () at /home/benpicco/dev/RIOT/tests/pkg_u8g2/main.c:122
```

j-devel, 2020-08-14:
Hi, I have found that the culprit is not `getpid()` but some `X11` string-related functions.  As a workaround, when using SDL2 with mods not calling into those functions, the demo works fine without segmentation faults.

kfessel, 2021-10-07:
Fixed the main issue with #16954