RIOT-OS/RIOT #13555: Use both UART and USB CDC ACM at same time

toonst, 2020-03-04:
Since the STM32 USB driver is merged, I would like to start using it for stdio logging, while at the same time keeping UART for the shell. We use the vfs abstraction for writing logs in our application.
Currently both USB and UART fall under the stdio_base abstraction, see image to get an idea.

[image]

I tried modifying the shell implementation to optionally use periph/uart.h directly, but I don't think this is a clean solution. Do you have some ideas on how to accomplish this in a clean way?
Maybe both USB and UART stdio could be made different files under VFS, but I don't directly see how this would work.

Thanks for the help!

toonst, 2020-03-06:
apparently there's already a discussion going on here https://github.com/RIOT-OS/RIOT/issues/13469. I'll join the discussion there.

miri64, 2020-03-06:
:+1: I'll close this as a duplicate then.

closed 2020-03-06 by miri64