RIOT-OS/RIOT #13566: CC2538-bsl submodule

JelmerT, 2020-03-05:
I wanted to check if the position on sub-modules is still the same. See https://github.com/RIOT-OS/RIOT/pull/5784 and https://github.com/RIOT-OS/RIOT/pull/4237 and https://github.com/JelmerT/cc2538-bsl/pull/92

In the mean while there's been improvements on the script on the RIOT side that I would love to have upstream and the other way round. Submodule or something that tracks upstream would be very convenient for this.

(@smlng )

benpicco, 2020-03-06:
We don't do submodules, but tools like [edbg](https://github.com/RIOT-OS/RIOT/tree/master/dist/tools/edbg) are just downloaded on-demand.

I didn't even realize cc2538-bsl was an external tool - we should definitely replace it with e pkg-based approach then instead of maintaining a fork.

closed 2020-03-06 by benpicco

JelmerT, 2020-03-06:
👍 thanks for this! 

Welcoming all improvements upstream

PR #13568 (2020-03-06): dist/tools/cc2538-bsl: use upstream version

### Contribution description

Automatically fetch the upstream version of cc2538-bsl.py instead of maintaining our own fork of the tool.

### Testing procedure

Flash one of the supported boards, e.g `openmote-b`.
`cc2538-bsl.py` should automatically be downloaded. 

### Issues/PRs references

fixes #13566
commit ed4d8718 (2020-03-06): dist/tools/cc2538-bsl: use upstream version

Automatically fetch the upstream version of cc2538-bsl.py instead
of maintaining our own fork of the tool.

fixes #13566