RIOT-OS/RIOT #13580: ztimer and pm_layered

jue89, 2020-03-06:
#### Description

I would like to use ztimer with a `periph_timer` and a `periph_rtt` in a low power application. The application wakes up in a fixed interval. To save energy, I have to disable `periph_timer` when the board is sleeping. How do achieve this with ztimer?

My basic idea is to block the power state that puts the device into deep sleep if at least one `XTIMER_USEC` clock is running.

Does ztimer know, how many timers are currently running on a specific clock?

Would it be a good idea to extend `ztimer_clock_t` with callbacks that are called if the number of the clock's user changes from 0 to 1 or from 1 to 0?

kaspar030, 2020-03-06:
> Does ztimer know, how many timers are currently running on a specific clock?

Well, every clock knows if there are any timers (clock->list.next==0).

> Would it be a good idea to extend `ztimer_clock_t` with callbacks that are called if the number of the clock's user changes from 0 to 1 or from 1 to 0?

That would be one way to do it. I intended to initially just add one more field to the periph/timer backend to specify a power mode to block/unblock, and do that depending on set / not set timers.

Adding callbacks to every clock would make this more generic, at some cost.

Do you think unblocking a powermode for ztimer_periph_timer would be sufficient, if pm_layered can then switch to deep sleep? Or does the periph_timer need to be expllicitily powered down?

kaspar030, 2020-03-06:
Another option would be to explore callbacks on sleep / wakeup, and make ztimer register itself there. It could then power-up/down hardware at that point.

kaspar030, 2020-03-06:
> Or does the periph_timer need to be expllicitily powered down?

If yes, there's some semantical challenge on when ztimer_now() is supposed to work.

jue89, 2020-03-06:
> Do you think unblocking a powermode for ztimer_periph_timer would be sufficient, if pm_layered can then switch to deep sleep? Or does the periph_timer need to be expllicitily powered down?

I think the cleanest solution would be to explicitly power down the timer and let the timer handle the whole power management. Maybe we could add some configuration booleans to ztimer like `CONFIG_ZTIMER_PERIPH_TIMER_POWER_DOWN_ON_IDLE` and document the side effect of `timer_now(clock)` not being incremented while the underlying timer is powered down. (That is the challenge you are talking about?)

jue89, 2020-03-06:
Maybe @benpicco also has an opinion on that. He was really active on pm-related PRs recently ;)

An addition to just disabling the timers:
I have the Microchip/Atmel SAM series in mind. Its timers can be configured to *block* the deep sleep in hardware (i.e. disabling the main clock). So RIOT and pm_layered doesn't have to keep track of the timers actually being used or not.

benpicco, 2020-03-06:
> I think the cleanest solution would be to explicitly power down the timer and let the timer handle the whole power management.

+1

I haven't looked into zTimer yet, but it would sound reasonable to expect something to request a timer to keep running if you want to use `timer_now(clock)` on it.

>  Its timers can be configured to *block* the deep sleep in hardware

Uh didn't know about that yet - do you mean the `RUNSTDBY` flag?

jue89, 2020-03-06:
> I haven't looked into zTimer yet, but it would sound reasonable to expect something to request a timer to keep running if you want to use `timer_now(clock)` on it.

To be clear: `timer_now(ZTIMER_USEC)` will always stuck if you disable the underlying timer. And this is always necessary for power saving.

But I can think of use cases where you need high precision time measurement without any high precision timer being running. This problem isn't trivial at all ...

> Uh didn't know about that yet - do you mean the `RUNSTDBY` flag?

Yip. That's why I used the emphasis markup for the word *block*.

jue89, 2020-03-06:
FYI ... I tinkered around with the ztimer to have some kind of occupancy detection. It should work as long as no clock ist extended.

If you are curious, you can have a look into my sandbox:
https://github.com/jue89/RIOT/tree/sandbox/ztimer_deep-sleep

jue89, 2020-04-02:
Since we found a way to go, I'll close this issue and move the discussion to #13722.

PR #13722 (2020-05-01): sys/ztimer: add power management for ztimer clocks
### Contribution description

This is a follow-up of #13585. After some discussion with @kaspar030 I came up with an implementation of his approach.

The basic idea / assumptions behind this PR:

* On some platforms, the power consumption can be reduced dramatically by disabling high frequency hardware timers. Those timers are used by the `ZTIMER_USEC` clock.
* The high frequency timers require a certain pm mode to be active. Lower pm modes deactivate those timers.
* ztimer knows if any `ztimer_t` is running on the `ZTIMER_USEC` clock.
* ztime [... cut]
closed 2020-04-02 by jue89