RIOT-OS/RIOT #13587: drivers/cc110x: Lost IRQs can lead to driver deadlock

maribu, 2020-03-08:
#### Description

During `cc110x_send()` the CC110x driver blocks until the transmission is completed to prevent upper layers from sending during an ongoing transmission. This is done by using a locked mutex to signal the driver thread from IRQ when the transmission is completed (or the TX FIFO needs to be filled with remaining bytes for the outgoing frame). However, when the corresponding IRQ is lost, `cc110x_send()` keeps waiting until another event on the device causes an interrupt, which might never happen.

#### Steps to reproduce the issue

In general, very difficult. I have a local setup which reliably looses one IRQ during startup.

#### Expected results

The driver will at least recover and give up sending the frame, or better try to complete sending the frame before the TX FIFO has underflown.

#### Actual results

`cc110x_send()` remains blocked, the driver is in a deadlock state.

#### Versions

Current master

wosym, 2020-03-09:
I'm not sure if I understand your issue entirely, but I encountered something similar last week.
When the ADC is sampling, it locks a mutex. If an (unrelated) interrupt occurs during the sampling proces, you'd expect it to return to where it left off so it can unlock the mutex again. However, it appeared that after the interrupt occurred, it never returned to the sampling function. 

I'm not sure if it is related at all to this issue, but it might be worth mentioning.

maribu, 2020-03-09:
> However, it appeared that after the interrupt occurred, it never returned to the sampling function.

Maybe the issue is that a higher priority thread whats to be unblocked by the second (unrelated) IRQ and then never lets go of the CPU again? Note that only higher priority threads will preempt low priority threads in RIOT. This means if a high priority thread spins endlessly, all other threads will starve. (However, if the higher priority thread wants to lock a mutex currently held by a lower priority thread, the higher priority thead will block, allowing the lower priority thread to complete until it unlocks the mutex.)

Maybe you also have a cycle in the "wait for" graph? (E.g. thread t1 is waiting to get a mutex held by t2, t2 is waiting for t3 and t3 is waiting for t1.)

In any case: I'm pretty sure these are unrelated issues. However, feel free to give more info on your issue so that you can get help. (If the issue is more likely in your code rather in RIOTs, maybe you should ask in the RIOT user email list?)

wosym, 2020-03-09:
> Maybe the issue is that a higher priority thread whats to be unblocked by the second (unrelated) IRQ and then never lets go of the CPU again? 

No, the sampling got called from main. After the IRQ happened, it ended up in main again, calling the same sampling function. This caused the deadlock. 

I solved it in the meantime by removing the mutexes and adding a timeout to the sample-function. The ADC-was not super critical for my application, so it doesn't matter if one reading is wrong from time to time (even though I don't think it will happen, because the ADC is only accessed from this single thread).
Anyways, if they are unrelated I won't pollute this Issue-posting with this spam :)

maribu, 2020-03-09:
Can you open an issue for this? This sounds like a bug in RIOT. Which board are you using?

wosym, 2020-03-09:
@maribu I experienced the behavior on an AVR board. 
I'll try to make a reproduction of the issue in a minimal working sample and open an Issue for it.

maribu, 2020-03-09:
I just checked the code of `adc_sample()` in `cpu/atmega_common/periph/adc.c`. There seems to be no way this function can exit without the mutex being unlocked again. If I had to guess, I'd say that the context switching from IRQ context doesn't work correctly. That code was recently touched (by me) to fix an unrelated issue, so chances are good that I fucked up something there. Having in mind that a `RETI` is used to return to the last instruction the thread to switch to was previously left, it would make sense that if the stack is off by 16 bit that the context switch returns the caller of the function previously running in that threat, rather than to the correct location.

The minimum working sample would be super helpful to debug the issue. Feel free to assign me (or mention me) in the new issue.

wosym, 2020-03-09:
@maribu : Indeed, I also spent quite some time investigating how it would be possible to exit the sample function without unlocking the mutex, but could not find anything. 
Not sure about the other stuff you mention, I haven't investigated that yet. But it did indeed seem like the code never returned to where it left off after the IRQ happened. (I added prints "locking", "unlocking" and "IRQ happened". Occasionally (let's say approx 10% chance of it happening) it showed locking - IRQ - locking, after which it deadlocked. (Except for the IRQ's which could still happen ofcourse)

I added it to my TODO list and will try to open an issue asap :)

closed 2020-03-12 by benpicco

maribu, 2020-03-12:
@wosym: I just did some testing and context switching triggered from ISR seems to just work as expected.

Maybe your issue simply is a stack overflow or another memory corruption? Can you check what happens if you increase the stack?

wosym, 2020-03-13:
@maribu I will do it asap, but am currently a bit preoccupied with other stuff. I hope to have it tested by next week.