RIOT-OS/RIOT #13602: OpenMote-B + Enable MAC + Enable CSMA/CA

pietrotedeschi, 2020-03-10:
#### Description
- It is possible to know how to enable the MAC Layer on Openmote-b? It is possible to adopt it? For example I saw lwmac, but I'm not sure that I can adopt it.

- Further, with the new version of RIOT, I have some problems related to the PHY addresses when I attach multiple motes to my PC.

- How can it is possible send and receive messages only at MAC Layer without using UDP? Please if you have a good example, provide me it.

- With the previous versions of RIOT, it is not possible enable the duplicate packets recognition and so on. It is a problem related only to Openmote-b?

- When I send several messages, it seems that I have a problem with the Buffer in RX. It seems that it is not enough.

- How can it is possible enable the synchronization like in OpenWSN?

fjmolinas, 2021-01-14:
> #### Description
> 
>     * It is possible to know how to enable the MAC Layer on Openmote-b? It is possible to adopt it? For example I saw lwmac, but I'm not sure that I can adopt it.

You should be able to use lwmac, or gomach for this.

>     * Further, with the new version of RIOT, I have some problems related to the PHY addresses when I attach multiple motes to my PC.

What kind of problems?

>     * How can it is possible send and receive messages only at MAC Layer without using UDP? Please if you have a good example, provide me it.

You mighe be able to use the `ieee802154_submac` for this, what do you think @jia200x?

>     * With the previous versions of RIOT, it is not possible enable the duplicate packets recognition and so on. It is a problem related only to Openmote-b?

Can you give more details on this?

>     * When I send several messages, it seems that I have a problem with the Buffer in RX. It seems that it is not enough.

There have been many changes to the driver since, could you test if this is still the case?

>     * How can it is possible enable the synchronization like in OpenWSN?

You can use OpenWSN + openmote-b to do this

jia200x, 2021-01-14:
Hi @pietrotedeschi 

> How can it is possible send and receive messages only at MAC Layer without using UDP? Please if you have a good example, provide me it.

The at86rf215 is supported using the old interface (`netdev_ieee802154`). You can send and receive frames directly using the netdev interface. You can check e.g `tests/driver_at86rf215`.

> With the previous versions of RIOT, it is not possible enable the duplicate packets recognition and so on. It is a problem related only to Openmote-b?

Unfortunately this is implemented only for GNRC. You would need to manually de-duplicate packets for now if you want to use the radio directly. We are currently working that with the new interface (`ieee802154_radio_hal`).

> When I send several messages, it seems that I have a problem with the Buffer in RX. It seems that it is not enough.

Could you give more details? Do you have logs or some experiment to replicate this results?

closed 2022-04-18 by stale[bot]