RIOT-OS/RIOT #13633: SAMR30 with LWMAC

samken600, 2020-03-13:
I have been working on trying to get LWMAC working with the SAMR30-XPRO boards for a little bit now and have noticed there seem to be problems with the boards receiving things. Debug outputs suggest that both LWMAC and the at86rf2xx radio drivers are fine sending things, with TX_SUCCESS states being entered correctly, but after an RX_START (NETDEV_EVENT_RX_START) is sent an RX_COMPLETE (NETDEV_EVENT_RX_COMPLETE) is never received!

I was wondering if anyone had any experience either with the at86rf2xx drivers or LWMAC and the SAMR30s and could offer me some advice on this. I have confirmed that it works fine with the SAMR21s, which use the AT86RF233 instead of the AT86RF212B used in the SAMR30 chips.

Any help is appreciated! 
Thanks

jue89, 2020-03-14:
There were a few pull requests related to the samr30-xpro board around, which have been merged recently. Does your problem still persist with the current master?

samken600, 2020-03-14:
> There were a few pull requests related to the samr30-xpro board around, which have been merged recently. Does your problem still persist with the current master?

I should clarify that, without using the LWMAC protocol, wireless communications works fine. I am already using the SPI clock rate redefine you committed a bit ago and know that the antenna is transmitting properly from a separate receiver I have been using for testing. Seems to be something a bit wrong with the internals when receiving, although can't figure out what.

I will try updating to master anyways and seeing if that fixes anything! Will reply again if it does.

samken600, 2020-07-06:
Sorry for forgetting about this! I realised the main source of the issue is that I had forgotten to recalibrate the timings of the MAC layer to something more reasonable, since the timings were for the faster 2.4GHz on the SAMR21s. I have done this now and it works, but is quite unreliable. I was wondering if anyone else had some timings for the layer that they found to be more reliable? Thanks

closed 2021-02-13 by stale[bot]