RIOT-OS/RIOT #13651: Documentating usage and correct usage of `DEFAULT_MODULE`

fjmolinas, 2020-03-18:
### Description

There is currently no documentation and usage guide on how to use `DEFAULT_MODULE` and `DISABLE_MODULE`. Both tools also have some limitations but these may not be apparent.

This issue wants to open up the discussion on the them and see where we want to allow or limit its use.

### Current usage `DEFAULT_MODULE` and `DISABLE_MODULE`

##### `makefiles/defaultmodules.inc.mk`

This file declares two types of `DEFAULT_MODULE`s

- `periph_init` and `auto_init`: modules that trigger dependency resolution (they include other `periph_init_%` and `auto_init_%` modules)

- `board cpu core core_init core_msg core_panic sys` modules that mostly will never be disabled (and where disabling some of them does nothing, they are still compiled in e.g. `sys`) 

##### `DEFAULT_MODULE` declared at application level

The only case so far is `tests_utils_interactive_sync` included in all test applications

##### `DEFAULT_MODULE` by other `MODULE`s or `PKG`s

- `stdio_rtt`
- `oneway_malloc`
- `auto_init_%`

##### `DISABLE_MODULE` at application level

Used mainly in tests to disable some functions and `auto_init_%`

##### `DISABLE_MODULE` in dependency resolution

There is only one case for this `DISABLE_MODULE += stdio_rtt`.

#### Converting `DEFAULT_MODULES` -> `USEMODULE`

`DEFAULT_MODULES` are converted into `USEMODULES` in two places:

1. Before parsing `Makefile.dep`: https://github.com/RIOT-OS/RIOT/blob/efb113696de6ec193b78b20f8f8b9a2a5ca78c8a/Makefile.include#L363-L364

1. After the recursive catch only for `periph_init_%` and `auto_init_%` modules.

### Limitations

`Makefile.dep` uses a recursive resolution method, it will parse `Makefile.dep` many times, at the end of each parse it will evaluate `USEMODULE` and re-parse `Makefile.dep`. This also means that its designed as an incremental algorithm and is prone to issues when removing `MODULE`s.

https://github.com/RIOT-OS/RIOT/blob/efb113696de6ec193b78b20f8f8b9a2a5ca78c8a/Makefile.dep#L1099-L1102

Since we use `USEMODULE := $(sort $(USEMODULE))` the order of inclusion becomes relevant and can change build outcome.

To summarize the impact on `DEFAULT_MODULE`/`DISABLE_MODULE`

- order of inclusion matters, e.g. `DISABLE_MODULE += stdio_rtt` in `Makefile.dep`, this is a case where order of inclusion matters, #13650 fixes the issue in #13460 where module dependencies would be included even if the module was disabled.

- Currently only modules added to `DEFAULT_MODULE` in the first pass will be converted to `USEMODULE`s

- `DEFAULT_MODULE` must be disabled in the same iteration they are converted to `USEMODULE`. (currently this can only happen in the first iteration and for some `MODULE`s on the last iteration)

- `DEFAULT_MODULE`/`DISABLE_MODULE` can not be added to the recursive catch, the recursive catch only works for incrementally adding `MODULE`s it can not work backwards.

### Usage proposal

I've summarized the all possible I could think of here. Some are pretty straight forward, but there are some complicated cases, currently the main one is `stdio_rtt` and `stdio_cdc_acm` **(c)**

**(a)** Modules that do not trigger dependency resolution and do not depend on another module (e.g.: `tests_utils_interactive_sync`, `oneway_malloc`)

These modules can be handled either at the end or at the begging of the iteration through `Makefile.dep`.

Modules could have a `default.inc.mk` parsed at the same time as the common `default.inc.mk` 

**(b)** Modules that  trigger dependency resolution and do not depend on another module (e.g.: `periph_init, `auto_init`)

These modules must be resolved before there dependencies or in the same iteration

**(c)** Modules that  trigger dependency resolution and  depend on another module (e.g.: `stdi_rtt`, `stdio_cdc_acm`)

These modules most be resolved in the first iteration through `Makefile.dep` and care must be taken sot that if they are disabled, this must be handled before there dependencies are included  #13650.

**(d)**  Modules that do not trigger dependency resolution but depend on another module(e.g.: `periph_init_%`, `auto_init_%`)

These modules after all its parent modules are resolved, therefore they must be resolved in the last iteration.

fjmolinas, 2020-03-20:
Also if looking at the current definition `DISABLE_MODULE`:

```
export DISABLE_MODULE        # Used in the application's Makefile to suppress DEFAULT_MODULEs.
```

maribu, 2020-03-20:
OK. So the original intent was that `DEFAULT_MODULE` contained a static list of modules to always pull in, unless blacklisted. Right now that use expends to mimic something like `FEATURES_OPTIONAL`, but for modules.

The problem with using `DEFAULT_MODULE` as lets say `MODULES_OPTIONAL` is, that there is no efficient way to check if a module can be supported with the existing dependency resolution approach. (One could of course build a list of all possible subsets of the optional modules, including the empty and the full set, and run a dependency resolution for them. The largest set that does not fail would be the one to choose. But this approach does not really scale.) So instead a manual way to tackle these by using some heuristics for each module and the blacklist.

The "right" way to solve this would be to change approach fundamentally. E.g. each module could contain a JSON/YAML/... that lists is hard module and feature requirements and its optional module and feature requirements. That way the individual requirements of a module (and its transitive dependencies) could be tracked, allowing to more efficiently decide if an optional module can be pulled in. Buuuut: That would be hell lot of work.

An easier way would be to just decide that `DEFAULT_MODULE` and `DISABLE_MODULE` must be static and may not be touched in `Makefile.dep`. That would btw. work fine for e.g. the `auto_init` stuff. We could say that `auto_init_all` becomes a default module, which pulls in all parts of `auto_init`. If someone wants to have all but one part of `auto_init`, that person would have to blacklist `auto_init_all` and manually add all `auto_init_<foo>` modules that would have been automatically been pulled in except the one she/he wants to skip. Not super convenient, but as this use case seems to not effect the regular RIOT users, I would personally prefer just that.

fjmolinas, 2020-03-26:
> Buuuut: That would be hell lot of work.

Yes although it would probably be quite useful for any migration (whether it is Kconfig, or another tool).

> An easier way would be to just decide that DEFAULT_MODULE and DISABLE_MODULE must be static and may not be touched in Makefile.dep. That would btw. work fine for e.g. the auto_init stuff. We could [...]

Maybe a way to mitigate both is to add a a new variable name for the non static  **(d)** case, so a `DEFAULT_MODULE_OPTIONAL` or another name. `DISABLE_MODULE` would still apply to them, but `DEFAULT_MODULE` would be kept as a static list. **(a)** and **(b)** will keep the current `DEFAULT_MODULE` behavior and **(c)** would just be unauthorized (since we don't have a way of dealing with this right now we might have to keep it for `stdio_%`)

We could add static checks as well.

miri64, 2020-07-06:
I think it might also be a good help to rename `DISABLE_MODULE` to `DISABLE_DEFAULT_MODULE`.

maribu, 2020-07-06:
> I think it might also be a good help to rename `DISABLE_MODULE` to `DISABLE_DEFAULT_MODULE`.

I'm not sure. To me, `DISABLE_DEFAULT_MODULE` would sound like it would not fail the build if it ends up being used due to a hard dependency. But if I recall correctly, the build will fail in that case. (If my memory is incorrect, I would be in favor of the renaming.)

However, I like the idea of being able "un-default" modules in a way builds don't fail if they end up being used for some good reason.

miri64, 2020-07-06:
> I'm not sure. To me, `DISABLE_DEFAULT_MODULE` would sound like it would not fail the build if it ends up being used due to a hard dependency. But if I recall correctly, the build will fail in that case. (If my memory is incorrect, I would be in favor of the renaming.)

Maybe my knowledge about `DEFAULT_MODULE`s (as one of the people [who originated the idea](https://github.com/RIOT-OS/RIOT/issues/926)) is a bit out dated, but IMHO, a module that when deactivated leads to build errors is not an optional module but a mandatory component:

```py
if RIOT:
    component required
```

maribu, 2020-07-06:
It seems `DISABLE_MODULE` is not limited to default modules in its use:

```
$ DISABLE_MODULE=xtimer make -C examples/gnrc_networking
make: Entering directory '/home/maribu/Repos/software/RIOT/examples/gnrc_networking'
Required modules were disabled using DISABLE_MODULE: xtimer
/home/maribu/Repos/software/RIOT/examples/gnrc_networking/../../Makefile.include:886: *** You can let the build continue on expected errors by setting CONTINUE_ON_EXPECTED_ERRORS=1 to the command line.  Stop.
make: Leaving directory '/home/maribu/Repos/software/RIOT/examples/gnrc_networking'
```

To me, renaming `DISABLE_MODULE` to `DISABLE_DEFAULT_MODULE` would seem out of place in the above example.

miri64, 2020-07-06:
AFAIK `xtimer` is not a `DEFAULT_MODULE`,

```
$ git grep DEFAULT_MODULE.*xtimer
Makefile.dep:    DEFAULT_MODULE += auto_init_xtimer
```

so that confusion might be lifted by the rename ;-)

miri64, 2020-07-06:
(the grep'd dependency comes from `xtimer`.

miri64, 2020-07-06:
```
$ git grep -B 3 DEFAULT_MODULE.*xtimer
Makefile.dep-ifneq (,$(filter xtimer,$(USEMODULE)))
Makefile.dep-  ifeq (,$(filter ztimer_xtimer_compat,$(USEMODULE)))
Makefile.dep-    # xtimer is used, ztimer xtimer wrapper is not
Makefile.dep:    DEFAULT_MODULE += auto_init_xtimer
```

miri64, 2020-07-06:
(and maybe remove that error message)

maribu, 2020-07-06:
> AFAIK `xtimer` is not a `DEFAULT_MODULE`

True. But that is not the point. My point is that `DISABLE_MODULE` will enforce that a module is not used, not just "un-default" them.

E.g. let's say `foo` is a default module and compilation without `foo` is at least for some applications possible. But if I use module `bar`, `foo` gets pulled in as hard dependency. If a run now `DISABLE_MODULE=foo USEMODULE=bar make -C some/app`, it is obvious that I asked for something that is impossible and I would expect an error. And that is what the build system does.

If we rename to `DISABLE_DEFAULT_MODULE`, I would expect a different behavior: I would expect the build system to just use `foo` anyway and a friendly warning that `foo` is not used "by default", but as dependency.

I'm not against the rename. But if so, please lets also change the build system behavior to match the expectations associated with the new name.

Btw: Why are `board`, `cpu`, `core`, `core_init`, and `core_panic` default modules? Should I be allowed to disable them?

miri64, 2020-07-06:
> True. But that is not the point. My point is that `DISABLE_MODULE` will enforce that a module is not used, not just "un-default" them.

But that was what `DISABLE_MODULE` was originally introduced for... See #926. That's why I argue a rename would clear-up this misunderstanding (and, after I realized it `DISABLE_MODULE` leads to an error, not a warning, removing the error / transforming it to a warning).

> Btw: Why are board, cpu, core, core_init, and core_panic default modules? Should I be allowed to disable them?

I'd say yes. With the caveat, that they are a different category of default, as they are non-deactivatible and required.

maribu, 2020-07-06:
> as they are non-deactivatible and required.

Wasn't this the distinctive feature dividing hard dependencies from default modules?

Why don't we go for `USEMODULE += board cpu core core_init core_panic` instead? That would make sure that they cannot be deactivated.

miri64, 2020-07-06:
> Why don't we go for `USEMODULE += board cpu core core_init core_panic` instead? That would make sure that they cannot be deactivated.

True! So :+1:!

aabadie, 2021-01-10:
Still not addressed, unstale.

closed 2022-04-18 by stale[bot]